Koenkk/zigbee2mqtt #22966: Moes BHT-002-GCLZB - Floods logs - The requested state (933) exceeds the maximum allowed length (255)

richsimpson, 2024-06-07:
### What happened?

Getting warning in homeassistant logs. One state is too long, includes all the days time and temp settings.

If it is not an issue how can I stop this particular warning coming through

### What did you expect to happen?

No warnings

### How to reproduce it (minimal and precise)

Always happening now

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20221226

### Adapter

/dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0

### Setup

proxmox running HASS is VM. Core 2024.6.0 Supervisor 2024.06.0 Operating System 12.3 Frontend 20240605.0

### Debug log

Logger: homeassistant.components.mqtt.sensor
Source: components/mqtt/util.py:268
integration: MQTT (documentation, issues)
First occurred: 6 June 2024 at 10:33:01 (1909 occurrences)
Last logged: 08:33:28

Cannot update state for entity sensor.underfloor_heating_program after processing payload on topic zigbee2mqtt/underfloor_heating. The requested state (933) exceeds the maximum allowed length (255). Fall back to unknown, failed state: {'saturday_p1_hour': 1, 'saturday_p1_minute': 0, 'saturday_p1_temperature': 17, 'saturday_p2_hour': 5, 'saturday_p2_minute': 0, 'saturday_p2_temperature': 17, 'saturday_p3_hour': 22, 'saturday_p3_minute': 0, 'saturday_p3_temperature': 17, 'saturday_p4_hour': 23, 'saturday_p4_minute': 31, 'saturday_p4_temperature': 24, 'sunday_p1_hour': 1, 'sunday_p1_minute': 0, 'sunday_p1_temperature': 17, 'sunday_p2_hour': 5, 'sunday_p2_minute': 0, 'sunday_p2_temperature': 17, 'sunday_p3_hour': 22, 'sunday_p3_minute': 0, 'sunday_p3_temperature': 17, 'sunday_p4_hour': 23, 'sunday_p4_minute': 31, 'sunday_p4_temperature': 24, 'weekdays_p1_hour': 1, 'weekdays_p1_minute': 0, 'weekdays_p1_temperature': 17, 'weekdays_p2_hour': 5, 'weekdays_p2_minute': 0, 'weekdays_p2_temperature': 17, 'weekdays_p3_hour': 22, 'weekdays_p3_minute': 0, 'weekdays_p3_temperature': 17, 'weekdays_p4_hour': 23, 'weekdays_p4_minute': 31, 'weekdays_p4_temperature': 24}

richsimpson, 2024-06-07:
Is only opton (preferred option?) to filter the 'program' settings/attributes of the device?
[image]

stalakerob, 2024-06-09:
I'm getting the same error for the Sonoff TRVZB device (firmware version 1.1.5):

`Logger: homeassistant.components.mqtt.sensor
Quelle: components/mqtt/util.py:268
Integration: MQTT (Dokumentation, Probleme)
Erstmals aufgetreten: 11:49:17 (6 Vorkommnisse)
Zuletzt protokolliert: 11:49:21

Cannot update state for entity sensor.sonoff_thermostat_buro_weekly_schedule after processing payload on topic zigbee2mqtt/Sonoff Thermostat Büro. The requested state (477) exceeds the maximum allowed length (255). Fall back to unknown, failed state: {'friday': '00:00/19 00:00/19 00:00/19 00:00/19 00:00/19 00:00/19', 'monday': '00:00/19 00:00/19 00:00/19 00:00/19 00:00/19 00:00/19', 'saturday': '00:00/19 00:00/19 00:00/19 00:00/19 00:00/19 00:00/19', 'sunday': '00:00/19 00:00/19 00:00/19 00:00/19 00:00/19 00:00/19', 'thursday': '00:00/19 00:00/19 00:00/19 00:00/19 00:00/19 00:00/19', 'tuesday': '00:00/19 00:00/19 00:00/19 00:00/19 00:00/19 00:00/19', 'wednesday': '00:00/19 00:00/19 00:00/19 00:00/19 00:00/19 00:00/19'}`

steve-prentice, 2024-06-19:
I came here to say exactly the same thing... would love an official workaround, the length thing seems to be a fixed thing in home assistant, so i'm not sure how it was working ok previously.

Either way, as Rich has suggested above I've filtered on the word "schedule" and we'll see what happens.

h3llrais3r, 2024-07-31:
> I'm getting the same error for the Sonoff TRVZB device (firmware version 1.1.5):
>
> `Logger: homeassistant.components.mqtt.sensor Quelle: components/mqtt/util.py:268 Integration: MQTT (Dokumentation, [...]

Hi, having the same issue with my sonoff TRVZB (firmware 1.1.1).
Any solution for this yet, as my logs are crowded with these warnings:
```
024-07-31 14:40:01.058 WARNING (MainThread) [homeassistant.components.mqtt.sensor] Cannot update state for entity sensor.smart_trv_livingroom_1_weekly_schedule after processing payload on topic zigbee2mqtt/Smart trv livingroom 1. The requested state (477) exceeds the maximum allowed length (255). Fall back to unknown, failed state: {'friday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'monday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'saturday': '00:00/16 07:00/19 10:00/19 10:00/19 10:00/19 23:00/16', 'sunday': '00:00/16 07:00/19 10:00/19 10:00/19 10:00/19 23:00/16', 'thursday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'tuesday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'wednesday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16'}
2024-07-31 14:40:01.058 WARNING (MainThread) [homeassistant.components.mqtt.sensor] Invalid state message '{'friday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'monday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'saturday': '00:00/16 07:00/19 10:00/19 10:00/19 10:00/19 23:00/16', 'sunday': '00:00/16 07:00/19 10:00/19 10:00/19 10:00/19 23:00/16', 'thursday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'tuesday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'wednesday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16'}' from 'zigbee2mqtt/Smart trv livingroom 1'

```

steve-prentice, 2024-07-31:
It'd be nice to see something on the zigbee2mqtt side to fix it if possible, but for now I've just done the following to exclude it:

[image]

(in the Settings tab for the TRVZB)

closed 2024-12-29 by github-actions[bot]