Koenkk/zigbee2mqtt #22967: New giex gx02 water valve

Akstef, 2024-06-07:
### Link

https://fr.aliexpress.com/item/1005006547328147.html?spm=a2g0o.order_list.order_list_main.23.69455e5bduwHMO&gatewayAdapt=glo2fra

### Database entry

{"id":28,"type":"EndDevice","ieeeAddr":"0xa4c138f7b13e9aac","nwkAddr":24816,"manufId":4417,"manufName":"_TZE200_7ytb3h8u","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1717763874465}

### Comments

Hi 
I have ordered a new giex gx02 water valve. I have already installed 3 of them and they are fully recognized in zigbee2mqtt. The new one is discovered but not supported. The manufacturer name changed from _TZE204_7ytb3h8u to _TZE200_7ytb3h8u.

I have used this manual configuration but it still not recognized:

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const MINUTES_IN_A_DAY = 1440;
const SECONDS_IN_12_HOURS = 43200;

module.exports = [
    {
        fingerprint: [
            {modelID: 'TS0601', manufacturerName: '_TZE200_7ytb3h8u'},
        ],
        model: 'GX02',
        vendor: 'GiEX',
        description: 'GX02 GiEX Water Valve',
        extend: [],
        meta: {},
        onEvent: tuya.onEventSetLocalTime,
        fromZigbee: [legacy.fromZigbee.giexWaterValve],
        toZigbee: [legacy.toZigbee.giexWaterValve],
        exposes: [
            e.battery(),
            e.binary(legacy.giexWaterValve.state, ea.STATE_SET, 'ON', 'OFF')
                .withDescription('State'),
            e.enum(legacy.giexWaterValve.mode, ea.STATE_SET, ['duration', 'capacity'])
                .withDescription('Irrigation mode'),
            e.numeric(legacy.giexWaterValve.cycleIrrigationNumTimes, ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(100)
                .withDescription('Number of cycle irrigation times, set to 0 for single cycle'),
            e.numeric(legacy.giexWaterValve.irrigationStartTime, ea.STATE)
                .withDescription('Last irrigation start time'),
            e.numeric(legacy.giexWaterValve.irrigationEndTime, ea.STATE)
                .withDescription('Last irrigation end time'),
            e.numeric(legacy.giexWaterValve.lastIrrigationDuration, ea.STATE)
                .withDescription('Last irrigation duration'),
            e.numeric(legacy.giexWaterValve.waterConsumed, ea.STATE)
                .withUnit('L')
                .withDescription('Last irrigation water consumption'),
            e.numeric(legacy.giexWaterValve.irrigationTarget, ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(SECONDS_IN_12_HOURS)
                .withUnit('seconds or litres')
                .withDescription('Irrigation target, duration in seconds or capacity in litres (depending on mode), ' +
                    'set to 0 to leave the valve on indefinitely, ' +
                    'for safety reasons the target will be forced to a minimum of 10 seconds in duration mode'),
            e.numeric(legacy.giexWaterValve.cycleIrrigationInterval, ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(SECONDS_IN_12_HOURS)
                .withUnit('sec')
                .withDescription('Cycle irrigation interval'),
        ],
    },
];

How can I add the new valve?

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_7ytb3h8u',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

SnouF, 2024-06-08:
I have the same manufacturer ID. After a few tests, it works with this external definition

``` javascript
const exposes = require('zigbee-herdsman-converters/lib/exposes');
//const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const MINUTES_IN_A_DAY = 1440;
const SECONDS_IN_12_HOURS = 43200;

module.exports = [
    {
        fingerprint: [
            {modelID: 'TS0601', manufacturerName: '_TZE200_7ytb3h8u'},
        ],
        model: 'GX02',
        vendor: 'GiEX',
        description: 'GX02 GiEX Water Valve',
        extend: [],
        meta: {},
        onEvent: tuya.onEventSetLocalTime,
        fromZigbee: [legacy.fromZigbee.giexWaterValve],
        toZigbee: [legacy.toZigbee.giexWaterValve],
        exposes: [
            e.battery(),
            e.binary(legacy.giexWaterValve.state, ea.STATE_SET, 'ON', 'OFF')
                .withDescription('State'),
            e.enum(legacy.giexWaterValve.mode, ea.STATE_SET, ['duration', 'capacity'])
                .withDescription('Irrigation mode'),
            e.numeric(legacy.giexWaterValve.cycleIrrigationNumTimes, ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(100)
                .withDescription('Number of cycle irrigation times, set to 0 for single cycle'),
            e.numeric(legacy.giexWaterValve.irrigationStartTime, ea.STATE)
                .withDescription('Last irrigation start time'),
            e.numeric(legacy.giexWaterValve.irrigationEndTime, ea.STATE)
                .withDescription('Last irrigation end time'),
            e.numeric(legacy.giexWaterValve.lastIrrigationDuration, ea.STATE)
                .withDescription('Last irrigation duration'),
            e.numeric(legacy.giexWaterValve.waterConsumed, ea.STATE)
                .withUnit('L')
                .withDescription('Last irrigation water consumption'),
            e.numeric(legacy.giexWaterValve.irrigationTarget, ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(SECONDS_IN_12_HOURS)
                .withUnit('seconds or litres')
                .withDescription('Irrigation target, duration in seconds or capacity in litres (depending on mode), ' +
                    'set to 0 to leave the valve on indefinitely, ' +
                    'for safety reasons the target will be forced to a minimum of 10 seconds in duration mode'),
            e.numeric(legacy.giexWaterValve.cycleIrrigationInterval, ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(SECONDS_IN_12_HOURS)
                .withUnit('sec')
                .withDescription('Cycle irrigation interval'),
        ],
    },
];
```

Akstef, 2024-06-08:
Thank you, it works perfect. I don't know what I did wrong with my external definition. Maybe a syntax error somewhere. 

Can someone add the new device as supported device in zigbee2mqtt or should we always use an external definition?

Koenkk, 2024-06-08:
Added!

closed 2024-06-08 by Koenkk

andre991, 2024-06-11:
@Koenkk This adjustment will be in which release?

Koenkk, 2024-06-11:
1 July release.

Wright55, 2024-06-12:
I have followed the solution but get this error when adding to the configuration.yaml

can not read a block mapping entry; a multiline key may not be an implicit key (62:20)

 59 | 
 60 | module.exports = [
 61 |     {
 62 |         fingerprint: [
-------------------------^
 63 |             {modelID: 'TS0601', manufacturerName:  ...

I have a new device after successfully using a previous device fox several weeks, and wondered why is stated ‘unsupported’.

robertobonilla1, 2024-06-27:
Hello,

I'm new here and I have the same issue with the new manufacter valve.
Could you explain how to use "external definition"? Where I should paste the code provided? I'm using ZHA (I don't know if this could be a problem).

Thanks in advance, regards.