Koenkk/zigbee2mqtt #22969: Aqara H1 binding

bendeve, 2024-06-07:
### What happened?

I'm trying to bing Aqara H1 (WS-EUK04) to a bulb, without success. 

Can't bind to a single device : 
[image]

I can bind to a group but nothing append when I press the button : 

[image]

Moreover, is 1 the left button and 2 the right the one ? 

[image]

Thank you for your help

### What did you expect to happen?

I would like to be able to bind left button and right button directly to others devices

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ZBDongle-E

### Setup

NUC with Proxpox

bendeve, 2024-06-17:
Any news on this?
Thank you.

bendeve, 2024-06-19:
genOnOff is missing in the output, is that the issue ?

[image]

l3zha, 2024-06-20:
did you manage to do it?

bendeve, 2024-06-26:
No, situation is still the same

vlacour97, 2024-09-18:
Hi !
Nothing new about this topic ?

lerra, 2024-10-25:
I'm also interested if anybody managed to solve this, I'm running the slzb-06m coordinator

chinery, 2024-12-08:
I'm also facing this issue, if anyone here had any luck. Does it simply not support binding?

lerra, 2024-12-08:
I returned mine, from my understanding their "binding" works when you have their hub so I would not expect this to get fixed

chinery, 2024-12-08:
That's a shame. It's only for very occasional use for me, so I can make do with having them setup as automations instead. But definitely a knock on the device.

koeberlue, 2025-01-03:
Possibly related to https://github.com/Koenkk/zigbee2mqtt/issues/24069 (TL;DR: This is not supported by Aqara)

adynis, 2025-01-03:
I'm not sure if it's related or helpful, but recentely I got an update on Aqara H1 and at least I found something which was resolved (check this https://github.com/Koenkk/zigbee2mqtt/issues/20545#issuecomment-2563179410 ).
I got the update through an Aqara Hub.

davidsarkany, 2025-02-06:
Anybody can check with the new firmware?

chinery, 2025-03-05:
I ended up having to get an Aqara hub to update my no-neutral H1 firmware due to the same bug linked above. 

Sadly no, the firmware update didn't add any new output clusters for binding. There was no firmware update for the neutral wire version of the H1 at all.

> Z2M does not support it because Aqara is not using Zeigbee's standards for this feature (I might be wrong, because as I said I don't have Aqara HUB to check if it works)

I don't think that's the reason... I have already returned my Aqara Hub so I can't say for sure, but I think Aqara carefully only advertise the fact that their [battery powered wireless H1 model](https://www.aqara-shop.co.uk/product/aqara-wireless-remote-switch-h1-double-rocker/) can be used to bind to a normal H1, this is from the product description:
> **2-Way Switch Setup Support with Wall Switch H1**: By combining the Wireless Switch H1 with the Smart Wall Switch H1, you can easily control the same lights from two or more locations without having to add additional wiring.

[on the wired H1](https://www.aqara-shop.co.uk/product/aqara-smart-wall-switch-h1-no-neutral-double-rocker/), it says this:
> **2-Way Setup Support with Wireless Switch H1**: An existing 2-way setup can be upgraded to the smart one using the [Wall Switch H1](https://www.aqara-shop.co.uk/product-tag/aqara-smart-wall-switch-h1/) EU and the [Wireless Switch H1](https://www.aqara-shop.co.uk/product/aqara-wireless-remote-switch-h1-double-rocker/), which can be placed both on top of the existing wall box used for a secondary 2-way switch or on any flat surface such as a wall using the provided sticky pad.

So my interpretation is that the wireless H1 has the `genOnOff` *output cluster*, and the wired version only has a `genOnOff` *input cluster*. Meaning you can bind the wireless H1 to control the wired one, but you cannot bind a wired one to control another wired one, even using the Aqara app.

FWIW using the "single press" action to trigger an automation in Home Assistant works with just a slight lag, maybe 0.5 seconds. I spent a little while trying to debug this to get it faster, but the entire chain from the z2m action to the z2m triggering the 2-way light is only a fraction of this time. 

I eventually realised that because the H1 supports a "double press" action, this is causing a big delay. It's waiting to see if you press again before it triggers the "single press" action via zigbee. That's built into the switch and not configurable. So in short, even if you were able to bind the devices at the zigbee layer, you'd still have the majority of the delay.

zak01011996, 2025-03-05:
@chinery thanks for clarification. Yeah it's sad, that it's not possible...

For me the input lag is not a problem actually. Currently I use automation in HA and H2 in decoupled mode (because of smart ceiling I have), so buttons are only triggering on/off state of that ceiling. Rely is always ON. 
It does it's job and I'm satisfied. 

But the thing I don't like is that if HA stops responding there won't be a way to control the ceiling... 
So I was hoping that it's possible to bind switch buttons and ceiling directly.

BTW, maybe you could suggest a zigbee switch that could work this way?

closed 2025-05-13 by github-actions[bot]

MateuszKukiela, 2025-08-25:
@zak01011996  SONOFF ZBM5 switch does it all, but only comes in white: https://sonoff.tech/products/sonoff-switchman-zigbee-smart-wall-switch-zbm5-80-86-type/54?variant=46546052579569

lerra, 2025-08-27:
From sonoff support: "ZBM5 doesn't support the direct binding function"