Koenkk/zigbee2mqtt #22971: Aurora AU-A1ZB110 1-10v inline Dimmer

Puntoboy, 2024-06-07:
### Link

https://auroralighting.com/gb/trade/productdetail/au-a1zb110

### Database entry

{"id":44,"type":"Router","ieeeAddr":"0x60a423fffe9918c6","nwkAddr":45407,"manufId":4636,"manufName":"Aurora","powerSource":"Mains (single phase)","modelId":"NPD3032","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c8db83","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c8db83","endpointID":1}],"configuredReportings":[{"cluster":8,"attrId":0,"minRepIntval":60,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"","zclVersion":2,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1717781967743}

### Comments

I had a quick look at the adding support for the new device but I couldn't really understand what I needed to do. 

It pairs and I can see the link quality and control the power (on/off) but I cannot set brightness. Pressing identify does make the light pulse (dim/bright) but that's all I can control.

If I go to the dev console, I can see there is a "LevelCtrl" cluster and I can read the current value. 

Read result of 'genLevelCtrl': {"currentLevel":254}

### External definition

```shell
const {identify, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['NPD3032'],
    model: 'NPD3032',
    vendor: 'Aurora',
    description: 'Automatically generated definition',
    extend: [identify(), onOff({"powerOnBehavior":false})],
    meta: {},
};

module.exports = definition;
```

Koenkk, 2024-06-08:
Added (including brightness support)

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Could you also submit a picture for the docs? This can be done by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/upload/master/public/images/devices). 

Make sure that:
- The filename is `MODEL.png` (update `model` accordingly) 
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2024-06-08 by Koenkk

Puntoboy, 2024-06-08:
> Added (including brightness support)
>
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.
>
> Could you also [...]

Thank you :)

Puntoboy, 2024-06-08:
> Added (including brightness support)
>
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.
>
> Could you also [...]

Thank you :)

I couldn't upload the image as uploads are disabled? 

[image]

Here is the image though. 

[image]

If I change the Edge version of Zigbee2MQTT, is it possible to change back to the release branch? What's the best way migrating to the Edge version from the release (as that's what I use now).

Puntoboy, 2024-06-10:
OK, just migrated to the Edge branch, device is now working for brightness, thank you. I did however notice this error when reading or writing the power on behaviour.

`[2024-06-10 11:12:24] error: 	z2m: Publish 'get' 'power_on_behavior' to 'Office LEDs' failed: 'Error: ZCL command 0x60a423fffe9918c6/1 genOnOff.read(["startUpOnOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

Koenkk, 2024-06-10:
It's not supported, removed it!