Koenkk/zigbee2mqtt #22973: Problem managing Tuya TS011F_plug_1 under ember

garry0garry, 2024-06-08:
### What happened?

Sonoff Zigbee 3.0 USB Dongle Plus V2” (model "ZBDongle-E"), coordinator v7.4.3.0 build 0. Zigbee2MQTT v1.38.0

ember (probability 50%):
`Publish 'set' 'state' to '0xa4c138b0ae6701b2' failed: 'Error: ZCL command 0xa4c138b0ae6701b2/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":28728,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":217},"zclSequence":19,"commandIdentifier":11} timed out after 10000ms)'
`

ezsp - no problem.

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

7.4.3.0 build 0

### Adapter

ZBDongle-E

### Setup

Docker

Nerivec, 2024-06-09:
I have one of these in a corner that worked with `ezsp` and works after `ember` migration without problem (been running `ember` since March on that network, haven't had a single timeout).
Any chance the route to the coordinator changed and a router in the middle is making trouble? Check the network map, see how it is linked, and if possible, try to pair it to another router, or directly to the coordinator (if not too far).

garry0garry, 2024-06-11:
The network configuration has not changed. The location of the sensors has not changed. Nothing has changed except for `ember`.

Nerivec, 2024-06-11:
The network may not have changed from your actions, but it will change on its own if/when needed. The `ember` driver changes a lot of internal things, which can result in the network rebuilding itself, at least partially, in some cases.

garry0garry, 2024-06-11:
Visually, nothing changed when switching to `ember`. But when I try to turn the smart plug on or off, it only works half the time. I can't believe this is a network issue, but you know better. There is no such problem on the `ezsp`.

rvalitov, 2024-10-21:
I have this device working for a day. My signal strength is around 170 LQI and no issues with timeout. I also tested it with firmware version 78 that was originally installed and now I'm testing the version 192 after OTA update.

My setup:
```
Zigbee2MQTT version 1.40.2
Coordinator type EmberZNet
Coordinator revision 7.4.4 [GA]
zigbee-herdsman-converters version 20.21.0
zigbee-herdsman version 2.1.3
```

closed 2024-12-29 by github-actions[bot]