Koenkk/zigbee2mqtt #22974: [New device support]: Support for CTM Lyng mKomfy Tak

mixedbreed, 2024-06-08:
### Link

https://www.zigbee2mqtt.io/devices/mKomfy_Sensor.html

### Database entry

{"id":33,"type":"EndDevice","ieeeAddr":"0x6c5cb1fffe1a3338","nwkAddr":11476,"manufId":4919,"manufName":"CTM Lyng","powerSource":"Battery","modelId":"mKomfy Tak","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,3,1026,65261,65481],"outClusterList":[3,25,4096],"clusters":{"genBasic":{"attributes":{"modelId":"mKomfy Tak","manufacturerName":"CTM Lyng","powerSource":3,"zclVersion":8,"appVersion":10,"stackVersion":106,"swBuildId":"1.0"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":198}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00257ccc68","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b00257ccc68","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10,"manufacturerCode":null},{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100,"manufacturerCode":null}],"meta":{}}},"appVersion":10,"stackVersion":106,"swBuildId":"1.0","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1717830579280}

### Comments

Hello, would like to add device CTM Lyng mKomfy Tak to Z2M.
CTM Lyng mKomfy Tak is the roof sensor integrated to the all ready supported stove guard.
See link: https://www.zigbee2mqtt.io/devices/mKomfy_Sensor.html

Please see the attached external definition.

BR
Mixedbreed

### External definition

```shell
const {battery, identify, temperature} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['mKomfy Tak'],
    model: 'mKomfy Tak',
    vendor: 'CTM Lyng',
    description: 'Automatically generated definition',
    extend: [battery(), identify(), temperature()],
    meta: {},
};

module.exports = definition;
```

Koenkk, 2024-06-08:
Added!

Could you also submit a picture for the docs? This can be done by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/upload/master/public/images/devices). 

Make sure that:
- The filename is `MODEL.png` (update `model` accordingly) 
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2024-06-08 by Koenkk

mixedbreed, 2024-06-08:
Hello and thanks again.
The picture is already uploaded as part of the documentation for CTM Lyng Stove Guard where you can see both the stove guard and the roof sensor.
Please see link of image.

https://www.zigbee2mqtt.io/images/devices/mKomfy_Sensor.png

Have a nice weekend!

overskeid, 2024-07-04:
A better way to do this would be to simply add 'mKomfy Tak' to the 'mKomfy_Sensor' converter.
Like this:
    zigbeeModel: ['mKomfy', 'mKomfy Tak'],
    model: 'mKomfy_Sensor',
    vendor: 'CTM Lyng',
    description: 'mKomfy, stove guard',

Then you will get more functionality as well.

Stian, CTM Lyng

mixedbreed, 2024-08-03:
> A better way to do this would be to simply add 'mKomfy Tak' to the 'mKomfy_Sensor' converter. Like this: zigbeeModel: ['mKomfy', 'mKomfy Tak'], model: 'mKomfy_Sensor', vendor: 'CTM Lyng', description: 'mKomfy, stove guard',
> 
> Then you will get more functionality as well.
> 
> Stian, CTM Lyng

Thanks for the suggestion Stian, but wouldn't it be better to add it to correct supported device instead of a "hack" with converters?

BR