Koenkk/zigbee2mqtt #22977: [Feature request]: Allow HA discvery refresh

ghoz, 2024-06-08:
### Is your feature request related to a problem? Please describe

Since https://github.com/Koenkk/zigbee2mqtt/pull/22701, changes to the homeassistant discovery overrides are no longer picked up 
https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html

### Describe the solution you'd like

From best to worst on user POV, and worst to best on dev effort , either :
* detect changes in discovery config and publish the changes accordingly (including changes in the global ``homeassistant`` key 
* add an internal way of refreshing the discovery config on a device per device way or in a global way in settings/tools for example
* Document that behaviour and give a workaround by removing the retained discovered topic manually

### Describe alternatives you've considered

Right now the only way I see is to manually remove the retained topics :
``mosquitto_sub -h <mqtthost> -u<user> -P<pass> -t 'homeassistant/+/<0xieeAddr>/+/config' --retained-only --remove-retained -W 2``
Removing all discovery messages will also remove all other non z2m mqtt configs. **Be really careful** as you'll have to force a republish of those in the other publishers...
``mosquitto_sub -h <mqtthost> -u<user> -P<pass> -t 'homeassistant/+/+/+/config' --retained-only --remove-retained -W 2``

### Additional context

N/A

Koenkk, 2024-06-09:
With "no longer being picked up", do you mean that now a restart is required or are they not picked up at all anymore?

ghoz, 2024-06-09:
So the update to the discovery config change are not picked up by Home Assistant, after z2m restart I need to manually remove the retained topic for the discovery config to be picked up again by HA

But I might have been too fast to point fingers, I thought it was linked to the recent optimisation as I don't remember this behavior before, but It does looks like the mqtt config payload is correctly sent my z2m, but not updated in HA :

```
mosquitto_sub -h 127.0.0.1 -u user -P password -v -t 'homeassistant/+/0xieeeAddr/power/config'
homeassistant/sensor/0xieeeAddr/power/config {"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Salon/Media-PriseConso/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xieeeAddr"],"manufacturer":"Nous","model":"Smart plug (with power monitoring) (A1Z)","name":"Salon/Media-PriseConso","via_device":"zigbee2mqtt_bridge_0xcoordAddr"},"device_class":"power","enabled_by_default":true,"entity_category":"diagnostic","object_id":"salon/media-priseconso_power","origin":{"name":"Zigbee2MQTT","sw":"1.38.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Salon/Media-PriseConso","unique_id":"0xieeeAddr_power_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.power }}"}
... after z2m restart ...
homeassistant/sensor/0xieeeAddr/power/config {"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Salon/Media-PriseConso/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xieeeAddr"],"manufacturer":"Nous","model":"Smart plug (with power monitoring) (A1Z)","name":"Salon/Media-PriseConso","via_device":"zigbee2mqtt_bridge_0xcoordAddr"},"device_class":"power","enabled_by_default":true,"object_id":"salon/media-priseconso_power","origin":{"name":"Zigbee2MQTT","sw":"1.38.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Salon/Media-PriseConso","unique_id":"0xieeeAddr_power_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.power }}"}
```

ghoz, 2024-06-09:
Ok scratch all that, 
After some time and restarts HA finally picked it up. 
Weird is that it picked up the disabled entities much faster, but kept the enabled ones with the old config and took much longer to update . 

anyway my bad ...

closed 2024-06-09 by ghoz