Koenkk/zigbee2mqtt #22978: [Feature request]: Keep yaml config comments and  anchors

ghoz, 2024-06-08:
### Is your feature request related to a problem? Please describe

This one is a tough one to address IMO.
Right now adding comments or anchors to repeating sections of configuration won't survive a setting change in the UI / bridge

### Describe the solution you'd like

Ideally 
* comments would survive config changes
* Anchors and references would be kept unless changed

### Describe alternatives you've considered

0. keep a backup of the config with the comments / anchors, 
for every change I want to make to the config files:
1. Diff with runtime config to find what changed using the UI, 
2. intergrate those changes,
3. override  the config with the backup + UI changes + new changes
4. restart z2m

### Additional context

Would probably need a rewrite of the yaml integration . the current module is not comment aware (as it should for a simple yaml reader/writer)
I found eemli yaml module (https://eemeli.org/yaml/#parsing-documents) that has a way of parsing full documents in addition to the actual objects semantics and would allow to modify / add in place values without breaking the comments (with caveats of course) 

The anchor case I really don't know to be honest...

ghoz, 2024-06-08:
Some references of related previous Issues :
https://github.com/Koenkk/zigbee2mqtt/issues/13339 
https://github.com/Koenkk/zigbee2mqtt/issues/1148

Koenkk, 2024-06-08:
It's indeed a hard one, this library claims it supports it: https://github.com/eemeli/yaml

closed 2024-12-21 by github-actions[bot]