Koenkk/zigbee2mqtt #22979: [New device support]: TZ3000_mkhkxx1p  1CH Switch Module-ES

das399igste, 2024-06-08:
### Link

https://de.aliexpress.com/item/1005007022047722.html?spm=a2g0o.order_list.order_list_main.4.5b925c5fWG0dhU&gatewayAdapt=glo2deu

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0xa4c138a04882c662","nwkAddr":33793,"manufId":4417,"manufName":"_TZ3000_mkhkxx1p","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_mkhkxx1p","zclVersion":3,"appVersion":72,"modelId":"TS0001","powerSource":1}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ccdf7b5","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b001ccdf7b5","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b001ccdf7b5","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":559524985},"lastSeen":1717848178587}

### Comments

Status:
{
    "linkquality": 117,
    "voltage": 32,
    "power_outage_memory": "restore",
    "state": "OFF",
    "switch_type": "state",
    "current": 0,
    "power": 0,
    "energy": 0
}

### External definition

```shell
const {identify, onOff, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0001'],
    model: 'TS0001',
    vendor: '_TZ3000_mkhkxx1p',
    description: 'Automatically generated definition',
    extend: [identify(), onOff({"powerOnBehavior":false}), electricityMeter()],
    meta: {},
};

module.exports = definition;
```

Koenkk, 2024-06-08:
Does it work with this external definition?

das399igste, 2024-06-08:
It works with the external definition.

Only the picture don't fit at all.

Power seems to have the known problem of tuya. With an LED light source with 5W, 0 is displayed. Values (12-13W) are then displayed for a light source with 11W.

Koenkk, 2024-06-10:
I checked, but it is supported in 1.38.0 already.

closed 2024-06-10 by Koenkk