Koenkk/zigbee2mqtt #22980: [Feature request]: Retain the network map data

deviantintegral, 2024-06-08:
### Is your feature request related to a problem? Please describe

When debugging offline devices, it’s helpful to know what their parent device is. However, since the device is offline, building the map won’t show any useful information.

### Describe the solution you'd like

If the map response data is retained in MQTT, it provides somewhat of a “last known” parent device. In the future, this could be triggered to run a scan periodically (say in the middle of the night) so a version of the map data is always instantly available.

### Describe alternatives you've considered

Another idea I had was to keep a copy of the path a message took each time its received and store it in the database. But, I’m not sure that information is actually surfaced by the firmware.

### Additional context

If we do this, we may need to also surface a last scanned date / time to the front end.

iotola, 2024-06-09:
I agree it would be helpful to retain the last generated map,  and the last message routing information would be great if possible.

Nerivec, 2024-06-11:
The problem with archiving the network map data for re-use, is that it may bring more confusion than benefits.
Since devices can change parents any time they want, the chances of a network map being still relevant drop drastically with time (and potentially even more in the context described of an "offline device"). Requesting the network map at regular short intervals is undesirable due to the load on the network.

However keeping track of the last route record could provide a lightweight alternative (since that's always up-to-date in the background), albeit not a full network map. I'm just not sure how much info is available on different stacks.

`ember`: https://github.com/Koenkk/zigbee-herdsman/blob/2b5b07462ece53f15fc987341b16451a486621ad/src/adapter/ember/ezsp/ezsp.ts#L4693

@Koenkk Is there some kind of route update notify in zstack?

Koenkk, 2024-06-11:
Zstack has the `ZDO_SRC_RTG_IND` command 

> This message is an indication to inform host device the receipt of a source route to a given device.

However I'm not sure how accurate the source routes reflect the actual parent.

closed 2024-12-21 by github-actions[bot]