Koenkk/zigbee2mqtt #22981: [New device support]: 2 Gang UK Socket with Power Monitoring and USB A and C ports

nblount, 2024-06-08:
### Link

https://www.ebay.co.uk/itm/395416536036?var=664100492450

### Database entry

{"id":74,"type":"Router","ieeeAddr":"0xa4c1388e7dbff816","nwkAddr":2476,"manufId":4417,"manufName":"_TZ3000_bep7ccew","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�/�-\u0012�/�-\u0012","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_bep7ccew","zclVersion":3,"appVersion":78,"modelId":"TS011F","powerSource":1}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":231,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,0]}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":0,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":2619665,"53251":123,"53252":11944,"53253":1374,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b86e15","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b86e15","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b86e15","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null}],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b86e15","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":78,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1717855745097}

### Comments

Can't figure out how to setup the definition. Device has two plug sockets only left one can be controlled via HomeAssistant by default. Current and Voltage seem to be shared by both sockets. Power only appears to report for the second socket. On/off switch only works for left socket. Power outage memory and Indicator mode seems to work. There is no information on the device or box as to who the manufacturer is or model number.

### External definition

```shell
const {deviceEndpoints, identify, onOff, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS011F'],
    model: 'TS011F',
    vendor: '_TZ3000_bep7ccew',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), identify(), onOff({"powerOnBehavior":false,"endpointNames":["1","2"]}), electricityMeter()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

nblount, 2024-06-08:
This works but suspect it can be improved:
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const {deviceEndpoints, identify, onOff, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS011F'],
    model: 'TS011F', // Update this with the real model of the device (written on the device itself or product page)
    vendor: '_TZ3000_bep7ccew', // Update this with the real vendor of the device (written on the device itself or product page)
    description: '2 Gang Socket with power monitoring and USB', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    extend: [deviceEndpoints({"endpoints":{"left":1,"right":2}}), onOff({"powerOnBehavior":false,"endpointNames":["left","right"]}), identify()],
    fromZigbee: [fz.identify, fz.on_off, fz.electrical_measurement], // We will add this later
    toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.power().withEndpoint('left'), e.voltage().withEndpoint('left'), e.current().withEndpoint('left')], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the f
    meta: {multiEndpoint: true},
};

module.exports = definition;

[image]

nblount, 2024-06-10:
Doesn't appear to, just appears to have voltage, current, energy and power for the both sockets. I could be wrong though.

Koenkk, 2024-06-10:
Does it also work with:

```js
const definition = {
zigbeeModel: ['TS011F'],
model: 'TS011F', // Update this with the real model of the device (written on the device itself or product page)
vendor: '_TZ3000_bep7ccew', // Update this with the real vendor of the device (written on the device itself or product page)
description: '2 Gang Socket with power monitoring and USB', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
extend: [deviceEndpoints({"endpoints":{"left":1,"right":2}}), onOff({"powerOnBehavior":false,"endpointNames":["left","right"]}), identify(), electricityMeter()],
};
```

nblount, 2024-06-10:
Will have to wire it up again to test, will get back to you.

nblount, 2024-06-10:
State:
{
    "current_left": 230,
    "linkquality": 214,
    "power_left": 18,
    "state_left": "OFF",
    "state_right": "ON",
    "voltage_left": 230,
    "current": null,
    "energy": null,
    "power": null,
    "voltage": null
}
I have something plugged into right but it shows power left, if I plug into the left it will still show the power etc on the left.

nblount, 2024-06-10:
[image]

Koenkk, 2024-06-10:
And with: 

```js
const definition = {
zigbeeModel: ['TS011F'],
model: 'TS011F', // Update this with the real model of the device (written on the device itself or product page)
vendor: '_TZ3000_bep7ccew', // Update this with the real vendor of the device (written on the device itself or product page)
description: '2 Gang Socket with power monitoring and USB', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
extend: [deviceEndpoints({"endpoints":{"left":1,"right":2}}), onOff({"powerOnBehavior":false,"endpointNames":["left","right"]}), identify(), electricityMeter()],
        meta: {multiEndpointSkip: ['current', 'voltage', 'power', 'energy']},
};
```

nblount, 2024-06-10:
{
    "current_left": 0,
    "linkquality": 207,
    "power_left": 0,
    "state_left": "OFF",
    "state_right": "OFF",
    "voltage_left": 231,
    "current": 143,
    "energy": null,
    "power": 14,
    "voltage": null
}

[image]

nblount, 2024-06-10:
Voltage came in eventually
{
    "current_left": 0,
    "linkquality": 204,
    "power_left": 0,
    "state_left": "ON",
    "state_right": "OFF",
    "voltage_left": 231,
    "current": 71,
    "energy": null,
    "power": 10,
    "voltage": 230
}

nblount, 2024-06-10:
Energy never appear to populate so probably not a valid option. Current is in mA, Power is W. The default definition allowed to change the settings for the light and set the power on setting (restore, on or off).

nblount, 2024-06-10:
Looking though my other zigbee devices this looks very similar to some power monitoring plugs I have which also appear as Zigbee Model TS011F Manufacturer Tuya Model BSD29_1. They have Power outage memory, Indicator mode and Child lock state these features where available on this 2 gang socket as well before I tried to build a custom config to support the two switches.

Nick-Sills, 2024-06-11:
Hi, Same device, same TS011F reported device issue. If it helps with manufacturer identification for the db, I purchased mine from AliExpress in the Shawader store. The device seems to be identified as a "Shawader Smart UK Zigbee Wall Socket".

Nick-Sills, 2024-06-11:
@Koenkk OK, so I'm experiencing similar behaviour to @nblount 

I've tried changing the external definition with what seemed logical, but to no avail, and now left and right switches are operating together. I'm new to this but if I can be pointed in the right direction, I'll see what I can do. Thanks.

Nick-Sills, 2024-06-11:
Basic question. After changing the external JS definition, do I  need to re-pair the device, restart Z2M, or both. When the state or MQTT payload does or doesn't change, it's difficult to know what's the right thing to do. Thanks

nblount, 2024-06-12:
Didn't have to repair but it did sometimes take a while or if you have errors in the .js it continues to use the previous configuration. Check the logs for any errors in the .js file.

elgerg, 2024-06-14:
Hi @Koenkk 

I added this converter and my double socket now works perfectly.

However my single plugs now look like this:

[image]

when they did look like:
[image]

Is there any way I can get both to appear separately from each other with the right configs?

Thanks in advance!

Koenkk, 2024-06-20:
Added! Config will be separated from the plug.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-06-20 by Koenkk