Koenkk/zigbee2mqtt #22985: BLE-YL01:  Free Chlorine is 10x the value

mzakharo, 2024-06-08:
### What happened?

BLE-YL01 Reports Free Chlorine levels, and zigbee2mqtt integration reports the value to be 10x the actual value in ppm. For example, according to ORP/pH tables, found [online](https://github.com/mzakharo/tubby/blob/main/charts/servovendi.png), Free chlorine  for  ORP=632  and pH=7.5, the Free Chlorine value should be 0.6ppm, but the reported integration value is 6ppm.
[image]

### What did you expect to happen?

chlorine ppm value reported should be multiplied by 0.1.

### How to reproduce it (minimal and precise)

Pair the BLE-YL01 and start using it

### Zigbee2MQTT version

 1.38.0 

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff E

### Setup

Docker container

closed 2024-06-10 by Koenkk

Burki24, 2024-06-16:
@Koenkk 
@mzakharo 

Can we please reverse this fix. The Chlor is meassured in mg/L I tested it with my prof. PoolLab 2. And there, I have 3,41 mg/l and in Z2M it is now 0,3 mg/l, so I think we have to use the .raw version:
[image]
[image]

mzakharo, 2024-06-16:
The value BLE-YL01 delivers is from the table mentioned above. They state this on the aliexpress page. The  0.1 fix is correct.