Koenkk/zigbee2mqtt #22987: Inovelli VZM36 - Fan doesn't turn off when setting speed to 0%

MangoScango, 2024-06-09:
### What happened?

When trying to turn off fans in Home Assistant, the default tile card turns off fans by calling fan.set_percentage, and setting the fan to 0%.

When this happens with the Inovelli VZM36, an error is logged for genLevelCtrl.moveToLevelWithOnOff, stating that the level parameter is missing.

### What did you expect to happen?

The Fan state should be set to off

### How to reproduce it (minimal and precise)

Call the following service in Home Assistant

service: fan.set_percentage
target:
  entity_id:
    - fan.master_bed_ceiling_fan
data:
  percentage: 0

### Zigbee2MQTT version

1.38.0-dev commit: a4a0c6f

### Adapter firmware version

7.4.2 [GA]

### Adapter

Sonoff Dongle-E running EmberZNet

### Setup

HassOS on RPi4

### Debug log

Error 2024-06-08 22:10:19z2m: Publish 'set' 'fan_mode' to 'Master Bed Ceiling Fan' failed: 'Error: ZCL command 0x881a14fffe41a944/2 genLevelCtrl.moveToLevelWithOnOff({"transtime":65535}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Parameter 'level' is missing)'

Koenkk, 2024-06-10:
@InovelliUSA could you check this?

InovelliUSA, 2024-06-13:
I am not seeing this. If I create a tile card or a card for the fan entity I don't get that error (the fan turns off as it should):

service: fan.set_percentage
metadata: {}
data:
  percentage: 0
target:
  device_id: 03a836548c71279101954dc837bdb142

```
Zigbee2MQTT:debug 2024-06-13 11:44:57: Received MQTT message on 'zigbee2mqtt/0xb0c7defffe61d5a3/set/fan_mode' with data 'off'
Zigbee2MQTT:debug 2024-06-13 11:44:57: Publishing 'set' 'fan_mode' to '0xb0c7defffe61d5a3'
Zigbee2MQTT:info  2024-06-13 11:44:57: MQTT publish: topic 'zigbee2mqtt/0xb0c7defffe61d5a3', payload '{"autoTimerOff_1":null,"autoTimerOff_2":null,"breezeMode":null,"brightness":170,"currentLevel":170,"defaultLevelRemote_1":null,"defaultLevelRemote_2":null,"dimmingSpeedDownRemote_1":null,"dimmingSpeedDownRemote_2":null,"dimmingSpeedUpRemote_1":null,"dimmingSpeedUpRemote_2":null,"fan_mode":"off","fan_state":"ON","higherOutputInNonNeutral_1":null,"ledColorWhenOn_1":null,"ledIntensityWhenOn_1":null,"linkquality":171,"maximumLevel_1":null,"maximumLevel_2":null,"minimumLevel_1":null,"minimumLevel_2":null,"outputMode_1":null,"outputMode_2":null,"quickStartLevel_1":null,"quickStartTime_1":null,"quickStartTime_2":null,"rampRateOffToOnRemote_1":null,"rampRateOffToOnRemote_2":null,"rampRateOnToOffRemote":127,"rampRateOnToOffRemote_1":null,"rampRateOnToOffRemote_2":null,"smartBulbMode_1":null,"smartBulbMode_2":null,"state":"ON","stateAfterPowerRestored_1":null,"stateAfterPowerRestored_2":null,"update":{"installed_version":67174657,"latest_version":67174657,"state":"idle"},"update_available":null}'
Zigbee2MQTT:debug 2024-06-13 11:44:57: Received Zigbee message from '0xb0c7defffe61d5a3', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":85}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2024-06-13 11:44:57: MQTT publish: topic 'zigbee2mqtt/0xb0c7defffe61d5a3', payload '{"autoTimerOff_1":null,"autoTimerOff_2":null,"breezeMode":null,"brightness":170,"currentLevel":170,"defaultLevelRemote_1":null,"defaultLevelRemote_2":null,"dimmingSpeedDownRemote_1":null,"dimmingSpeedDownRemote_2":null,"dimmingSpeedUpRemote_1":null,"dimmingSpeedUpRemote_2":null,"fan_mode":"low","fan_state":"ON","higherOutputInNonNeutral_1":null,"ledColorWhenOn_1":null,"ledIntensityWhenOn_1":null,"linkquality":171,"maximumLevel_1":null,"maximumLevel_2":null,"minimumLevel_1":null,"minimumLevel_2":null,"outputMode_1":null,"outputMode_2":null,"quickStartLevel_1":null,"quickStartTime_1":null,"quickStartTime_2":null,"rampRateOffToOnRemote_1":null,"rampRateOffToOnRemote_2":null,"rampRateOnToOffRemote":127,"rampRateOnToOffRemote_1":null,"rampRateOnToOffRemote_2":null,"smartBulbMode_1":null,"smartBulbMode_2":null,"state":"ON","stateAfterPowerRestored_1":null,"stateAfterPowerRestored_2":null,"update":{"installed_version":67174657,"latest_version":67174657,"state":"idle"},"update_available":null}'
Zigbee2MQTT:debug 2024-06-13 11:44:59: Received Zigbee message from '0xb0c7defffe61d5a3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID 0
```

MangoScango, 2024-06-13:
I wonder if it has to do with one of fan settings then. Here's the full state, anything i should try tweaking?

`2M@home.mangoplex.xyz

About
Exposes
Bind
Reporting
Settings
Settings (specific)
State
Clusters
Scene
Dev console
{
    "autoTimerOff_1": 0,
    "autoTimerOff_2": 0,
    "breezeMode": {
        "speed1": "high",
        "speed2": "medium",
        "speed3": "low",
        "speed4": "high",
        "speed5": "medium",
        "time1": 180,
        "time2": 60,
        "time3": 60,
        "time4": 300,
        "time5": 120
    },
    "brightness": 254,
    "defaultLevelRemote_1": 255,
    "defaultLevelRemote_2": 255,
    "dimmingSpeedDownRemote": 0,
    "dimmingSpeedDownRemote_1": 127,
    "dimmingSpeedDownRemote_2": 0,
    "dimmingSpeedUpRemote_2": 0,
    "fan_mode": "high",
    "fan_state": "OFF",
    "higherOutputInNonNeutral_1": "Disabled (default)",
    "ledIntensityWhenOn_1": 0,
    "linkquality": 216,
    "maximumLevel_1": 254,
    "maximumLevel_2": 255,
    "minimumLevel": 1,
    "minimumLevel_1": 3,
    "minimumLevel_2": 1,
    "outputMode": 1,
    "outputMode_1": "Dimmer",
    "outputMode_2": "Ceiling Fan (3-Speed)",
    "quickStartTime_2": 0,
    "rampRateOffToOnRemote_2": 0,
    "rampRateOnToOffRemote": 5,
    "rampRateOnToOffRemote_1": 5,
    "rampRateOnToOffRemote_2": 5,
    "smartBulbMode_1": "Disabled",
    "smartBulbMode_2": "Disabled",
    "state": "OFF",
    "stateAfterPowerRestored_1": 255,
    "stateAfterPowerRestored_2": 255,
    "update": {
        "installed_version": 67174657,
        "latest_version": 67174657,
        "state": "idle"
    },
    "dimmingSpeedUpRemote_1": null,
    "ledColorWhenOn_1": null,
    "quickStartLevel_1": null,
    "quickStartTime_1": null,
    "rampRateOffToOnRemote_1": null,
    "update_available": null
}`

MangoScango, 2024-06-13:
I've tried resetting as much of the fan settings as I can to the defaults, but the behavior still occurs. The only thing I can't put back to null values are the Min and Max levels for the fan. I'm thinking it must be related to that field being set in my config.

MangoScango, 2024-06-14:
So I tried removing from from my stick. factory resetting, and adding it back, and the behavior still remains. I've been comparing my state to yours, and noticed that your fan_mode gets set to "off", but that's not a setting my install seems to expose:
[image]

I messed around in the dev console, and setting the fanMode to 0 there turns the fan off as expected, but the fan mode attribute in z2m remains at the last value. So my new hypothesis is that this is whats causing the command to fail, the mapping to from fanMode 0 to "off" is missing on my install. But I have no idea how I would go about fixing that.

MangoScango, 2024-06-15:
I think I've figured out what needs changed to fix this, but not sure how I can test it. I submitted a PR if you can look it over and make sure it makes sense @InovelliUSA

closed 2024-06-15 by Koenkk

InovelliUSA, 2024-06-20:
@MangoScango can you try the Edge version of z2m to see if your issue is resolved? I am not sure why it wouldn't work for you to begin with, but hopefully your change will make a difference.

MangoScango, 2024-06-20:
Yeah, sorry I didn't mention. I updated to the edge build once my PR got merged in, and it resolved it the problem.