Koenkk/zigbee2mqtt #22990: Error: when I change driver from ezsp to ember z2m won't start

lpt2007, 2024-06-09:
### What happened?

I cahnged driver from ezsp to ember

### What did you expect to happen?

I see in logs this warning

`[2024-06-09 11:28:18] warning: 	zh:ezsp: Deprecated driver 'ezsp' currently in use, 'ember' will become the officially supported EmberZNet driver in next release. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462`

### How to reproduce it (minimal and precise)

If I change driver ezsp to ember a I get error:

```
[11:22:30] INFO: Preparing to start...
[11:22:30] INFO: Socat not enabled
[11:22:31] INFO: Starting Zigbee2MQTT...
[2024-06-09 11:22:33] info: 	z2m: Logging to console, file (filename: log.log)
[2024-06-09 11:22:33] info: 	z2m: Starting Zigbee2MQTT version 1.38.0 (commit #unknown)
[2024-06-09 11:22:33] info: 	z2m: Starting zigbee-herdsman (0.49.2)
[2024-06-09 11:22:33] info: 	zh:ember: Using default stack config.
[2024-06-09 11:22:33] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-06-09 11:22:33] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-06-09 11:22:33] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-09 11:22:33] info: 	zh:ember:uart:ash: Serial port opened
[2024-06-09 11:22:33] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-06-09 11:22:34] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-06-09 11:22:34] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-06-09 11:22:34] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-06-09 11:22:34] error: 	z2m: Error while starting zigbee-herdsman
[2024-06-09 11:22:34] error: 	z2m: Failed to start zigbee
[2024-06-09 11:22:34] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-06-09 11:22:34] error: 	z2m: Exiting...
[2024-06-09 11:22:34] error: 	z2m: Error: NCP EZSP protocol version of 12 does not match Host version 13
```

And If I put back driver ezsp I get warning:
`[2024-06-09 11:28:18] warning: 	zh:ezsp: Deprecated driver 'ezsp' currently in use, 'ember' will become the officially supported EmberZNet driver in next release. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462`

### Zigbee2MQTT version

1.38.0 commit: unknown

### Adapter firmware version

7.3.1.0 build 176

### Adapter

usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Setup

haos on virtual macine on xcp-ng host

### Debug log

```
[11:22:30] INFO: Preparing to start...
[11:22:30] INFO: Socat not enabled
[11:22:31] INFO: Starting Zigbee2MQTT...
[2024-06-09 11:22:33] info: 	z2m: Logging to console, file (filename: log.log)
[2024-06-09 11:22:33] info: 	z2m: Starting Zigbee2MQTT version 1.38.0 (commit #unknown)
[2024-06-09 11:22:33] info: 	z2m: Starting zigbee-herdsman (0.49.2)
[2024-06-09 11:22:33] info: 	zh:ember: Using default stack config.
[2024-06-09 11:22:33] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-06-09 11:22:33] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-06-09 11:22:33] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-09 11:22:33] info: 	zh:ember:uart:ash: Serial port opened
[2024-06-09 11:22:33] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-06-09 11:22:34] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-06-09 11:22:34] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-06-09 11:22:34] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-06-09 11:22:34] error: 	z2m: Error while starting zigbee-herdsman
[2024-06-09 11:22:34] error: 	z2m: Failed to start zigbee
[2024-06-09 11:22:34] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-06-09 11:22:34] error: 	z2m: Exiting...
[2024-06-09 11:22:34] error: 	z2m: Error: NCP EZSP protocol version of 12 does not match Host version 13
```

LaurentChardin, 2024-06-09:
You will need to update the firmware of your dongle : https://smarthomescene.com/guides/how-to-use-ember-driver-with-efr32mg21-zigbee-adapters/ , before choosing the `ember` adapter.

It is actually failing because the versions do not match.

Nerivec, 2024-06-09:
See [docs](https://www.zigbee2mqtt.io/guide/adapters/emberznet.html#logging-specific-to-this-stack) for details on other ember-specific messages.

closed 2024-06-09 by Nerivec