Koenkk/zigbee2mqtt #22994: Issue with mosquitto:latest and large zigbee networks using z2m

bgreet, 2024-06-09:
### What happened?

Running HA 2024.6.1.  Using eclipse-mosquitto:latest, devices populate into the HA integration but key controls do not (ie: switch on/off, light bulb on/off).  Certain data does populate like power consumption, and other states.  Additionally when home assistant is restarted, all devices become unavailable until coordinator is restarted.  Tried on fresh home assistant install with same outcome.  Tried to remove/readd mosquito home assistant integration.

Large device network (two zigbee networks each with individual mqtt topic, one network with 156 devices the other with 90) both using the same coordinator.

### What did you expect to happen?

All entities would be available for all devices.  Restart would automatically allow all devices to be available.

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20221226

### Adapter

TubeZB (CC2652)

### Setup

Server is bare metal with home assistant , mosquito and zigbee2mqtt installed on docker

richardstone, 2024-06-10:
I see the same. I'm using Mosquitto in docker with HA and z2m.

Recently updated z2m to 1.38.0 and all my devices became unavailable. When I deleted a device from HA, restarted z2m it got discovered but without any entities (i.e. switch/light, etc..)

Mosquitto: eclipse-mosquitto:2.0.15

vvrockhound, 2024-06-20:
same issues with a small network - I have to downgrade z2m to 1.35 for it to respond to home assistant commands.
Mosquitto: eclipse-mosquitto:2.0.18

closed 2024-12-29 by github-actions[bot]