Koenkk/zigbee2mqtt #22997: zigbee2mqtt does not respond to the MQTT topic

Sergej569, 2024-06-09:
### What happened?

zigbee2mqtt does not respond to the MQTT topic "Status". I cannot change the status. If I change it directly on the Zigbee2mqtt website, this status changes in the MQTT topic

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

 20210708

### Adapter

zStack3x0

### Setup

Pi

hbk001, 2024-06-10:
Same with 1.38.0

francisp2, 2024-06-12:
what exact message do you send ?

zigbee2mqtt/status ?

Sergej569, 2024-06-12:
I send "on" or "off" but nothing happens on zigbee2mqtt.

Koenkk, 2024-06-12:
Thats not a valid MQTT topic for z2m, look at the docs for your device for the correct topics

closed 2024-06-12 by Koenkk

hbk001, 2024-06-12:
Hope I got it correctly from Mqtt, this does not work on 1.38.0 but works fine on 1.37.0:
{"countdown":0,"linkquality":78,"power_on_behavior":"previous","state":"OFF","switch_type":"state"}
Just clicking a device on/off.

Sergej569, 2024-06-14:
Thanks for the help. It works now