Koenkk/zigbee2mqtt #23000: TuYa CK-BL702-AL-01 Feature "Do not siturb" no working

dan73800, 2024-06-10:
### What happened?

When I pair a bulb TuYa CK-BL702-AL-01 in my Z2M, I have a feature "Do not disturb" supposed to keep last status after power lost, but this feature is not working

### What did you expect to happen?

If I choose "Do not disturb = true" when power is lost, my bulb should stay off, but even with true or false, it does not change and the bulb lights on after a power lost

[image]

### How to reproduce it (minimal and precise)

Put the light Off, set the switch "do not disturb : true"
Power off the bulb as a current lost, then turn in back on and you will se the light turning on. It is the same case with "do not disturb set to false"

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

7.4.3 [GA]

### Adapter

Sonoff ZDongle-E EmberZNet

### Setup

arm64 Jeedom

### Debug log

empty log for this

rmxs, 2024-06-23:
After power outage i realize the same issue. All CK-BL702-AL-01 from off before the power outage turn on after power return.

defenseindepth1337, 2024-09-06:
I am facing the same issue with my bulbs even on the latest version(1.40.0-1). 

@Koenkk : Could you please look this this? Thanks in advance!

DlargeX, 2024-09-18:
I also have Gu10 and E27 lights and they are equipped with FW 1.2.3. Does anyone know which FW is current since I don't have a Tuya Bridge?

dimacoupe, 2024-11-01:
[CK-BL702-AL-01] in ceiling light form factor.
really too buggy. i have the same issue + device state offline but react on all action.

JosephTico, 2024-11-14:
Can confirm the issue

insleys, 2024-11-15:
Same issue here with all my E14 versions of this bulb.  I'm wondering if someone can confirm whether Do Not Disturb is actually an implemented feature on the controller on this bulb?  I suppose it's possible that this isn't a Z2M issue.  Has anyone tested in any other non-Z2M environment?

dimacoupe, 2024-11-15:
> Same issue here with all my E14 versions of this bulb. I'm wondering if someone can confirm whether Do Not Disturb is actually an implemented feature on the controller on this bulb? I suppose it's possible that this isn't a Z2M issue. Has anyone tested in any other non-Z2M environment?

I think this is a problem with the entire product line. I have 4 E27 light bulbs and 3 ceiling lamps with this marking, they all have the same problem in Z2M. There are no settings at all in TUYA, including “do not disturb”

insleys, 2024-11-15:
> I think this is a problem with the entire product line. I have 4 E27 light bulbs and 3 ceiling lamps with this marking, they all have the same problem in Z2M. There are no settings at all in TUYA, including “do not disturb”

I guess you get what you pay for then.  I suppose £2.71 a bulb of AliExpress was too good to be true.  Thanks for confirming.

RuStyC0der, 2024-12-06:
Hi community, a was able to test same device with ZHA, it does not have "do not disturb" mode but I was able to set power-on behavior off and it works.

Maybe by chance someone knows how to get raw command ZHA sends to bulb so I can send it from Z2M and achieve same behavior.

In my country problem with power outages is common and it`s very irritating to get all those bulbs on when power restores

RuStyC0der, 2024-12-06:
Guys, I think I found solution for this issue just go to dev console, then select parameters like this(`cluster OnOff, attribute startUpOnOff, value 255`) and press write)
[image]

after doing so, my bulb is not enables by itself after power restore anymore but only if it was enabled before. if you want it to remain off even it was powered before outage jsut set value to 0 instead of 255

dan73800, 2024-12-06:
Many thanks  @RuStyC0der , it works like a charm

dimacoupe, 2024-12-06:
> Guys, I think I found solution for this issue just go to dev console, then select parameters like this(`cluster OnOff, attribute startUpOnOff, value 255`) and press write) [...]

thanks, bro!

@Koenkk can you provide this in z2m by default?

rmxs, 2024-12-07:
> Guys, I think I found solution for this issue just go to dev console, then select parameters like this(`cluster OnOff, attribute startUpOnOff, value 255`) and press write) [...]

Thanks for the solution!
@Koenkk Please add this to z2m

mremedi2023, 2025-01-02:
> Guys, I think I found solution for this issue just go to dev console, then select parameters like this(`cluster OnOff, attribute startUpOnOff, value 255`) and press write) [...]

MASTER OF ALL MASTER!!!

faguirre89, 2025-02-08:
@RuStyC0der thanks a lot man, I was looking for that solution.

faguirre89, 2025-02-08:
> Guys, I think I found solution for this issue just go to dev console, then select parameters like this(`cluster OnOff, attribute startUpOnOff, value 255`) and press write) [image]
>
> after doing so, [...]

@RuStyC0der thanks a lot man, I was looking for that solution

JosephTico, 2025-04-10:
Any chance this gets implemented permanently?

Feriman22, 2025-05-13:
I can confirm that issue still exists with E14 bulbs.

FW: 1.2.3
Zigbee2MQTT version: 2.3.0

alvarillo88, 2025-05-28:
> Guys, I think I found solution for this issue just go to dev console, then select parameters like this(`cluster OnOff, attribute startUpOnOff, value 255`) and press write) [image]
>
> after doing so, [...]

Hi, thanks for the solution.

I have found that **if you set a value of 2, the 'do not disturb' option works like it must**.

Bjk8kds, 2025-11-09:
Same issue here