Koenkk/zigbee2mqtt #23003: [New device support]: Add Sunricher SR-ZG9030A-MW

87insane, 2024-06-10:
### Link

https://www.sunricher.com/zigbee-microwave-sensor-sr-zg9030a-mw.html

### Database entry

{"id":20,"type":"Router","ieeeAddr":"0x60a423fffef174f3","nwkAddr":51209,"manufId":4644,"manufName":"Sunricher","powerSource":"Mains (single phase)","modelId":"HK-DIM","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[3,4,5,6,8,25,4096],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":263,"inClusterList":[0,3,1030],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":262,"inClusterList":[0,3,1024],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.5.3_r54","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1718025108990}

### Comments

I have tried to pair the device but it is always recognised as the following:
Zigbee model: HK-DIM
Zigbee manufacturer: Sunricher
Description: LED dimmable driver

The following endpoints are displayed in the instructions:
**Endpoint 1:**
Input clusters:
0x0000: Basic
0x0003: Identify
0x0004: Groups
0x0005: Scenes
0x0006: On/off
0x0008: Level Control
0x0b05: Diagnostics
0x0100: zll commissioning
Output clusters:
0x0019: OTA
0x0100: zll commissioning
**Endpoint 2:**
Input Clusters:
0x0000: Basic
0x0003: Identify
0x0400: illuminance measurement
0x0406: occupancy sensing

https://www.sunricher.com/media/resources/manual/SR-ZG9030A-MW%20instruction.pdf

### External definition

```shell
const {deviceEndpoints, identify, onOff, occupancy, illuminance, commandsOnOff, commandsLevelCtrl} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['HK-DIM'],
    model: 'HK-DIM',
    vendor: 'Sunricher',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2,"3":3}}), identify(), onOff({"powerOnBehavior":false}), occupancy(), illuminance({"endpointNames":["3"]}), commandsOnOff(), commandsLevelCtrl()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

87insane, 2024-07-15:
Is somebody here, he can help?

niracler, 2024-12-25:
@87insane 

The support for this device has been merged. You can try it by updating to the dev branch.

https://github.com/Koenkk/zigbee-herdsman-converters/pull/8509

**Note** : When the device is first connected to Zigbee2MQTT, it may be incorrectly recognized as HK-DIM. Make sure to manually select ZG9030A-MW in the configuration and then power cycle the device for it to be correctly identified as ZG9030A-MW.

[image]

Feel free to reach out if you have any further questions.

closed 2024-12-26 by Koenkk

87insane, 2025-01-08:
Not working for me...

See: https://github.com/Koenkk/zigbee-herdsman-converters/pull/8509#discussion_r1905251670

87insane, 2025-01-08:
> @87insane
> 
> The support for this device has been merged. You can try it by updating to the dev branch.
> 
> [Koenkk/zigbee-herdsman-converters#8509](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8509)
> 
> **Note** : When the device is first connected to Zigbee2MQTT, it may be incorrectly recognized as HK-DIM. Make sure to manually select ZG9030A-MW in the configuration and then power cycle the device for it to be correctly identified as ZG9030A-MW.
> 
> <img alt="image" width="500" src="https://private-user-images.githubusercontent.com/24842631/398497863-2a5ffa9f-b3da-4c86-90a5-7ef831f3702f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYzMjUyMDEsIm5iZiI6MTczNjMyNDkwMSwicGF0aCI6Ii8yNDg0MjYzMS8zOTg0OTc4NjMtMmE1ZmZhOWYtYjNkYS00Yzg2LTkwYTUtN2VmODMxZjM3MDJmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA4VDA4MjgyMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWEwYTQ1NTk4NGI3MjI0NzRhMDg2YTVmZmI3NTU0NWUzZmRhOWFkZmEyMTNlZjQ5YzQ5NzQyNmU2OGFlYjliODYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.cEcxruiRc3YI_VdyYu0eFtmZiqSfxn6XjYZkRyodEos">
> Feel free to reach out if you have any further questions.

Not working for me...

See: https://github.com/Koenkk/zigbee-herdsman-converters/pull/8509#discussion_r1905251670

niracler, 2025-01-08:
@87insane I've received your question. I'll take a look at it in the next two days this week.

niracler, 2025-01-09:
@87insane Thank you for your feedback. I’ve retested it on my end, and everything seems fine.

1. After switching the modeId, you need to re-power the device for zigbee2mqtt to recognize it as ZG9030A-MW.
2. This device has quite a few configuration options. You might want to check which ones are not set to their default values. Sometimes, certain settings can reduce the sensor’s sensitivity, causing it to miss detecting people passing by.

before
[image]

after
[image]

87insane, 2025-01-09:
@niracler Thanks for your quick reply.

1. I tried that several times. It's a bit annoying to take the device out of the ceiling but unfortunately that doesn't help.
How I did it exactly:
A) Deleted the device from z2m.
B) Put the device in pairing mode (press Prg button 5 times)
C) When z2m found the device, selected the model
D) Removed the device from the ceiling and removed the power cable (1min)
What am I doing wrong?

2. I left the indicator LED on the device. So I can see when the sensor recognizes me or when it doesn't. Many of these options are only activated after a restart. At least if you set up the device via BT.

z2m Version: 2.0.0 commit: 060ae99