Koenkk/zigbee2mqtt #23004: [New device support]: GiEX GX02 (new Manufacturer Code)

MisterE1970, 2024-06-10:
### Link

https://de.aliexpress.com/item/1005006547328147.html?spm=a2g0o.order_list.order_list_main.59.4e6b5c5fOrhvjh&gatewayAdapt=glo2deu

### Database entry

0xa4c138fb29305ab8

### Comments

The device is not recognized in zigbee2mqtt, although it is in the database. It probably has a new vendor ID and is therefore not recognized. Please add the ID!
Produkt-Link [https://de.aliexpress.com/item/1005006547328147.html?spm=a2g0o.order_list.order_list_main.59.4e6b5c5fOrhvjh&gatewayAdapt=glo2deu](url)
Z2M-Database: https://www.zigbee2mqtt.io/devices/GX02.html 
Product-Label: [image]

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_7ytb3h8u',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

Koenkk, 2024-06-10:
Will be supported in the next z2m version!

closed 2024-06-10 by Koenkk

MisterE1970, 2024-06-11:
when will the release be?

Koenkk, 2024-06-11:
Every 1st of the month

NordJung, 2024-07-02:
Hello @MisterE1970,
is your GX02 recognized correctly after the update to Z2M 1.390? My Giex 02 (_TZE200_7ytb3h8u) is recognized as QT06_2.

MisterE1970, 2024-07-02:
[image]

It works!

DSchumacher2104, 2024-07-07:
Mine is recognized Now too, but not working correct. Shows wrong times, Looses the ability to switch and shows not the actual state, even it is shown as online. Don‘t know what to do.

The picture is wrong to, as my valve is the green and Not the grey one. It shows: _TZE200_7ytb3h8u

I have another green one with _TZE204_7ytb3h8u which works perfect.