Koenkk/zigbee2mqtt #23006: The control of Vimar 02973.B thermostat has some problems

amaciuc, 2024-06-10:
### What happened?

Hi,

I recently bought a Vimar 02973.B thermostat and I found that the control of it with Z2M has some problems:

- The system mode is only `heat` & `cool`. I succeeded to load an external converter where I put `off` as first value in the list of `withSystemMode`. Firstly it worked, but after some tests I don't know what I have broken and I started to received:
```
Publish 'set' 'system_mode' to 'vimar_devel_thermostat' failed: 'Error: ZCL command 0xf4ce36a7b879b96f/10 hvacThermostat.write({"systemMode":3}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_VALUE')'
```
- `RunningState` is missing. I also tried to load it in my external converter but I got errors
- `SetPoint` doesn't work with precision of `0.1`, only with `0.5`. It can be read with `0.1` but published only with `0.5` unfortunately. Maybe you have some suggestions here in order to fix it.
When it is set with `0.1` (as it comes from [source code](https://github.com/Koenkk/zigbee-herdsman-converters/blob/8994426959208acb41941e7ed04403c711f52dab/src/devices/vimar.ts#L72C1-L94C11)), the termostat never set that value, it always tends to set the closest value of `0.5` or `x.0`. That is a little strange. Could it be a firmware problem with the thermostat ? 
- `LocalTemperatureCalibrarion` is missing. I tried with external converter, it doesn't work.

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

20211210

### Adapter

Sonoff Dongle-P

### Setup

Intel NUC with HAOS

### Debug log

Here is my external converter  that I have worked with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
// const ea = exposes.access;

// function roundToOneDecimalPlace(value) {
//   return Math.round(value * 10) / 10;
// }

const definition = {

  zigbeeModel: ['Thermostat_v0.1', 'WheelThermostat_v1.0'],
  model: '02973.B',
  vendor: 'Vimar',
  description: 'Vimar IoT thermostat',
  fromZigbee: [fz.thermostat],
  toZigbee: [
    // {
    //   key: ['occupied_heating_setpoint'],
    //   convertSet: async (entity, key, value, meta) => {
    //     value = roundToOneDecimalPlace(value);
    //     await tz.thermostat_occupied_heating_setpoint.convertSet(entity, key, value, meta);
    //   },
    // },
    // tz.thermostat_running_state,
    tz.thermostat_local_temperature,
    // tz.thermostat_local_temperature_calibration,
    tz.thermostat_occupied_heating_setpoint,
    tz.thermostat_occupied_cooling_setpoint,
    tz.thermostat_system_mode,
  ],
  exposes: [
    e.climate()
      .withLocalTemperature()
      .withSetpoint('occupied_heating_setpoint', 4, 40, 0.5)
      .withSetpoint('occupied_cooling_setpoint', 4, 40, 0.5)
      // .withLocalTemperatureCalibration(-5, 5, 0.1)
      .withSystemMode(['off', 'heat', 'cool']),
      // .withRunningState(['idle', 'heat', 'cool']),
  ],
  configure: async (device, coordinatorEndpoint) => {
      const endpoint = device.getEndpoint(10);
      const binds = ['genBasic', 'genIdentify', 'hvacThermostat'];
      await reporting.bind(endpoint, coordinatorEndpoint, binds);
      await reporting.thermostatTemperature(endpoint);
  },

};

module.exports = definition;
```

DavideCarboni, 2024-06-21:
I have same problems, 
I noticed that the system_mode "off" commands and the command with the previous system_mode state work, but I can't change from "heat" to "cool" or vice versa

closed 2024-12-29 by github-actions[bot]

Belox86, 2025-01-15:
i'm thinking to buy the thermostat and using it with z2m did you managed to solve your issue? thx

sdp-17, 2025-10-18:
i also have the same issue with missing running_state. actually, i managed to get it published correctly over zigbee, but i don't know how to get it published as an attribute on HA. Please see the issue I have created recently:

https://github.com/Koenkk/zigbee2mqtt/issues/29078