Koenkk/zigbee2mqtt #23008: WZ5 Led controller controller in RGBW mode recognized as TuYa TS0504B

lukiii16, 2024-06-10:
### What happened?

WZ5 Led controller controller in RGBW mode recognized as TuYa TS0504B

After 1.5 year this is still the same issue as here:
https://github.com/Koenkk/zigbee2mqtt/issues/16338

### What did you expect to happen?

show up as WZ5 RGBW so i can contol withe collor as well

### How to reproduce it (minimal and precise)

Not possibile without device

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Setup

Home Assistant 2024.5.5 Addons

Legot, 2024-06-22:
Also having this issue.

Zigbee2MQTT version
   1.38.0

Coordinator type
    EZSP v13

Coordinator revision
    7.4.1.0 build 0

Setup
   Docker

jtherin, 2024-06-24:
same issue

closed 2024-12-29 by github-actions[bot]