Koenkk/zigbee2mqtt #23009: [Wrong device]: Nedis ZBSM10WT detected as TS0202

tjorim, 2024-06-10:
### Link

https://nedis.nl/nl-nl/product/veiligheid-en-beveiliging/detectoren/bewegingsdetectoren/550726145/smartlife-bewegingssensor-zigbee-30-batterij-gevoed-ip20-detectiehoek-120-0-detectiebereik-5-m-max-batterijduur-12-maanden-wit

### Model

ZBSM10WT

### Description

Motion sensor

### Vendor

Nedis

### Picture (link)

https://cdn.nedis.com/images/products_high_res/ZBSM10WT_P40.JPG

### Database entry

{"id":32,"type":"EndDevice","ieeeAddr":"0xa4c138c7122a2bca","nwkAddr":9418,"manufId":4417,"manufName":"_TZ3000_nss8amz9","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002390dc9e","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":74,"batteryVoltage":28}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002390dc9e","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1475368908},"lastSeen":1718048822098}

### Notes

Z2M detects the Nedis ZBSM10WT motion sensor as a [TuYa TS0202](https://www.zigbee2mqtt.io/devices/TS0202.html).
However, according to the supported devices page, it should be a TuYa RH3040.

Koenkk, 2024-06-11:
Is this just a cosmetic issue? (all functionality works?)

tjorim, 2024-06-11:
As far as I noticed, yes.

tjorim, 2024-06-11:
Basically. However, the Tamper sensor does not work. (And sometimes false positives but probably not related?)

Koenkk, 2024-06-11:
Fixed, removed the tamper.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-06-11 by Koenkk