Koenkk/zigbee2mqtt #23012: Editing Blocklist brings error

kfussmann, 2024-06-11:
### What happened?

I wanted to edit the blocklist by removing a device by clicking the minus in the UI. After clicking the minus the following errormessange was shown and the device was not deleted from the blocklist:
2024-06-11 11:54:42Request 'zigbee2mqtt/bridge/request/options' failed with error: 'blocklist must be array'

### What did you expect to happen?

I expected the device to be deleted without problem. 

### How to reproduce it (minimal and precise)

Every time I try to remove the same happens.

### Zigbee2MQTT version

1.38.0 HA Addon

### Adapter firmware version

20240315

### Adapter

ZigStar Stick V4

### Setup

Add-on on Home Assistant Green

### Debug log

See above mentioned error message

Koenkk, 2024-06-11:
Already fixed in dev and will be fixed in the next release

closed 2024-06-11 by Koenkk

Mafus1, 2024-06-21:
For anyone that does not want to wait for the release; its possible to remove blocked devices by editing the `configuration.yaml`. Just find the line that says `blocklist:` and remove either items from the list or the entire line.

gstolze, 2024-06-22:
line bloclist removed in the configuration.yaml, after reloading nothing happened.

goprojojo, 2025-01-16:
Same issue here. After I deleted de device from the configuration.yaml file Z2MQTT addon doesn’t start and shows the message: Refusing to start because configuration is not valid, found the following errors:
- blocklist must be array