Koenkk/zigbee2mqtt #23014: Zigbee2MQTT dashboard (port 9099)  button actions disappear too quickly to read

scargill, 2024-06-11:
### What happened?

Various zigbee buttons - in zigbee dashboard - click on device - exposes - look at ACTION - shows empty string("") normally - press the button and the message "1 single" or whatever, appears only for milliseconds and reverts back to empty string("")  FAR too quickly to read.. even my samsung phone on 60fps could only ~JUST capture a button press event.

### What did you expect to happen?

Iwould expect that th action would stay onscreen long enough to spot it easily.

### How to reproduce it (minimal and precise)

take for example any Aqara oe Aliexpress Zigbee button and press (one of) the button(s) and ty to view in zigbe2mqtt - go to device -exposes and view Action... 

### Zigbee2MQTT version

1.38.9 commit fe048e6

### Adapter firmware version

no version option - did you mean  SETTINGS - FRONTEND?

### Adapter

ZZH

### Setup

Zigbee2mqtt on Raspberry pi 5

Nerivec, 2024-06-11:
Opened a feature request in frontend for this (was on my TODO list 😉).
https://github.com/nurikk/zigbee2mqtt-frontend/issues/2047

closed 2024-06-11 by Nerivec

scargill, 2024-06-12:
Well done - thank you for listening...