Koenkk/zigbee2mqtt #23017: WARNING (MainThread) [supervisor.addons.options] Unknown option 'version' for Zigbee2MQTT (45df7312_zigbee2mqtt)

BusinessPerson, 2024-06-12:
### What happened?

**HA config**
Using HA Yellow with HA 2024.6.2
Supervisor 2024.06.0
OS 12.3
Frontend 20240610.0
Zigbee2mqtt Version 1.38.0-1
Mosquitto 6.4.1

**Zigbee2mqtt config**
>data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: ""
  log: false
mqtt:
  base_topic: zigbee2mqtt
serial:
  port: /dev/ttyAMA1
  adapter: ezsp

**Issue**
Zigbee2mqtt appears to be working, UI is up and accessible and the platform is serving requests, 
problem is error messages being spammed in Supervisor logs as follows:

> 2024-06-12 12:55:13.328 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
2024-06-12 12:55:13.504 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)

Current config has the following in the mqtt section of config (which is where I suspect the error / issue is:
`mqtt:
  version: 5`

replacing this with
`mqtt:
  base_topic: zigbee2mqtt`

results in the same error with:
>WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)

### What did you expect to happen?

No errors?

### How to reproduce it (minimal and precise)

Minimal config with the settings in mqtt as above

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

### Setup

HA Yellow with Z2M add-on

Koenkk, 2024-06-12:
Remove the `base_topic`, you have to configure it via the Z2M frontend.

closed 2024-06-12 by Koenkk

BusinessPerson, 2024-06-12:
Removing all content from the mqtt section results in 
`Failed to save add-on configuration, Missing option 'mqtt' in root in Zigbee2MQTT (45df7312_zigbee2mqtt). Got {'data_path': '/config/zigbee2mqtt', 'socat': {'enabled': False, 'master': 'pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777', 'slave': 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5', 'options': '', 'log': False}, 'serial': {'port': '/dev/ttyAMA1', 'adapter': 'ezsp'}}
`
Same issue if I edit it in YAML :-(