Koenkk/zigbee2mqtt #23021: Network/Route Error after updating to 1.38.0-1 with Ember driver

Krzysztonek, 2024-06-12:
### What happened?

[image]

### What did you expect to happen?

No errors

### How to reproduce it (minimal and precise)

From time to time the above cyan messages are displayed. It hasn't happened so far when using ezsp.
However, in common Z2M add-on seems to work fine, except few devices, which are being disconnected..

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

7.4.2

### Adapter

SkyConnect

### Setup

add-on, x86

diesel437, 2024-06-13:
I'm having the same issue.. but my adapter and firmware version are different. Also, I'm running some NPM issue, since I can only start the z2m if I run npm start. When I use systemctl start zigbee2mqtt.service it doen'st start the frontend, neither the network, and I get no errors.

**Zigbee2MQTT version**
1.38.0

**Adapter firmware version**
7.4.1

**Adapter**
SLZB-06M

**Setup**
LXC, x86

**Debug Log**

[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5]
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=0] Added to rxQueue
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5]
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=1] Added to rxQueue
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2]
[2024-06-13 06:14:38] debug:    zh:ember:ezsp: <=== [FRAME: ID=196:"INCOMING_NETWORK_STATUS_HANDLER" Seq=12 Len=8]
[2024-06-13 06:14:38] debug:    zh:ember:ezsp: ezspIncomingNetworkStatusHandler(): callback called with: [errorCode=ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY], [target=60542]
[2024-06-13 06:14:38] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "60542".
[2024-06-13 06:14:38] debug:    zh:ember:ezsp: <=== [FRAME: ID=128:"INCOMING_ROUTE_ERROR_HANDLER" Seq=12 Len=8]
[2024-06-13 06:14:38] debug:    zh:ember:ezsp: ezspIncomingRouteErrorHandler(): callback called with: [status=MAC_INDIRECT_TIMEOUT], [target=60542]
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5]
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=2] Added to rxQueue
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5]
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=3] Added to rxQueue
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4]
[2024-06-13 06:14:38] debug:    zh:ember:ezsp: <=== [FRAME: ID=196:"INCOMING_NETWORK_STATUS_HANDLER" Seq=12 Len=8]
[2024-06-13 06:14:38] debug:    zh:ember:ezsp: ezspIncomingNetworkStatusHandler(): callback called with: [errorCode=ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY], [target=60542]
[2024-06-13 06:14:38] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "60542".
[2024-06-13 06:14:38] debug:    zh:ember:ezsp: <=== [FRAME: ID=128:"INCOMING_ROUTE_ERROR_HANDLER" Seq=12 Len=8]
[2024-06-13 06:14:38] debug:    zh:ember:ezsp: ezspIncomingRouteErrorHandler(): callback called with: [status=MAC_INDIRECT_TIMEOUT], [target=60542]
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5]
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=4] Added to rxQueue
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5]
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=5] Added to rxQueue
[2024-06-13 06:14:38] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=6]
[2024-06-13 06:14:38] debug:    zh:ember:ezsp: <=== [FRAME: ID=196:"INCOMING_NETWORK_STATUS_HANDLER" Seq=12 Len=8]
[2024-06-13 06:14:38] debug:    zh:ember:ezsp: ezspIncomingNetworkStatusHandler(): callback called with: [errorCode=ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY], [target=60542]
[2024-06-13 06:14:38] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "60542".
[2024-06-13 06:14:38] debug:    zh:ember:ezsp: <=== [FRAME: ID=128:"INCOMING_ROUTE_ERROR_HANDLER" Seq=12 Len=8]
[2024-06-13 06:14:38] debug:    zh:ember:ezsp: ezspIncomingRouteErrorHandler(): callback called with: [status=MAC_INDIRECT_TIMEOUT], [target=60542]
[2024-06-13 06:14:47] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-13 06:14:47] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5]
[2024-06-13 06:14:47] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=6] Added to rxQueue
[2024-06-13 06:14:47] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7]
[2024-06-13 06:14:47] debug:    zh:ember:ezsp: <=== [FRAME: ID=89:"INCOMING_ROUTE_RECORD_HANDLER" Seq=12 Len=20]
[2024-06-13 06:14:47] debug:    zh:ember:ezsp: ezspIncomingRouteRecordHandler(): callback called with: [source=28540], [sourceEui=0x60b647fffe41ac77], [lastHopLqi=208], [lastHopRssi=-48], [relayCount=1], [relayList=48604]
[2024-06-13 06:14:50] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-13 06:14:50] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5]

Nerivec, 2024-06-22:
See details on network/route errors https://nerivec.github.io/z2m-ember-helper/#help (some of this is also in [Z2M docs](https://www.zigbee2mqtt.io/guide/adapters/emberznet.html#logging-specific-to-this-stack)).
You can also use the tool to try to identify if any device is causing more troubles than average, if you have a long enough log file (too short won't give good results, need several hours of runtime).
_Note: You will not see these displayed with `ezsp` because they only log in `debug`. You can see them in a `debug` log file though (search for lines starting with `handleRouteError`). `ember` now displays them because they can be helpful in identifying trouble devices and other such network issues._

PS: `MAX_MESSAGE_LIMIT_REACHED` means the adapter is overloaded (something is spamming too much).

closed 2024-07-21 by Nerivec