Koenkk/zigbee2mqtt #23023: Tuya ZG-204ZM Presence Sensor - erro changing motion detection mode

wafliron, 2024-06-12:
### What happened?

Changing the "Motion detection mode" setting on a ZG-204ZM presence sensor throws the error:

"Publish 'set' 'motion_detection_mode' to 'xxx' failed: 'Error: Value: 'PIR+Dadar' not found in: [only_pir, pir_and_dadar, only_dadar]'

'PIR+Dadar' will change to whatever value you're trying to set, out of the three available ones ("Only PIR", "PIR+Dadar", "Only Dadar".

My ZG-204ZM is on firmware 0122052017, which is a high enough version for this setting to be supported.

### What did you expect to happen?

Motion detection mode to change

### How to reproduce it (minimal and precise)

Choose any option from "Motion detection mode" setting in "Exposes" tab for the device.

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20230507

### Adapter

Sonoff ZBDongle-P

### Setup

Add-on on Home Assistant OS, running in an ESXi VM

WayneManion, 2024-06-12:
I have no idea if this is related, but I have two Tuya microwave sensors:
ZY-M100-S_2
ZG-205Z/A
I can't change any of the parameters for either sensor.

The ZY-M100-S_2 has configurable values for: target_distance, radar_sensitivity, minimum_range, maximum_range, detection_delay, fading_time. I get error messages when trying to change any of them.  

The ZG-205Z/A has configurable values for: large_motion_detection_sensitivity, large_motion_detection_distance, fading_time, medium_motion_detection_distance, medium_motion_detection_sensitivity, indicator, small_detection_distance, small_detection_sensitivity. I get error messages when trying to change any of them. 

The ZG 

Z2M 1.38.0
smlight SLZB-06 PoE Zigbee coordinator (Zigbee fw 20240315, ESP fw 2.0.18)
docker container on Debian 12

Sample error message (ZG-205Z/A):
Publish 'set' 'large_motion_detection_distance' to 'Bed side presence' failed: 'Error: ZCL command 0xa4c138ce723d4d7c/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":4,"datatype":2,"data":[0,0,1,71]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'

Judufu, 2024-06-18:
I am having the same problem as @wafliron

thedom4302, 2024-06-18:
I also have this issue, Presence value always has null too!

wafliron, 2024-06-19:
> I also have this issue, Presence value always has null too!

FYI Presence value being null might be (likely is?) a separate issue - that one works fine for me. In fact the whole sensor works OK, except for being unable to change the Motion Detection Mode setting as per my OP.

wafliron, 2024-06-19:
> I have no idea if this is related, but I recently added two Tuya microwave sensors to my setup: ZY-M100-S_2 ZG-205Z/A I can't change any of the parameters for either sensor. I don't think either of them are working at all honestly. They constantly report presence even though no one is in the room.
> 
> The ZY-M100-S_2 has configurable values for: target_distance, radar_sensitivity, minimum_range, maximum_range, detection_delay, fading_time. I get error messages when trying to change any of them.
> 
> The ZG-205Z/A has configurable values for: large_motion_detection_sensitivity, large_motion_detection_distance, fading_time, medium_motion_detection_distance, medium_motion_detection_sensitivity, indicator, small_detection_distance, small_detection_sensitivity. I get error messages when trying to change any of them.
> 
> Z2M 1.38.0 (docker container on Debian 12) smlight SLZB-06 PoE Zigbee coordinator (Zigbee fw 20240315, ESP fw 2.0.18)
> 
> Sample error message (ZG-205Z/A): Publish 'set' 'large_motion_detection_distance' to 'Bed side presence' failed: 'Error: ZCL command 0xa4c138ce723d4d7c/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":4,"datatype":2,"data":[0,0,1,71]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'

Just a suggestion, but you should probably open a separate ticket for these issues given they're a different model of device to mine. I would think more likely to get attention that way.

Judufu, 2024-06-23:
Any idea on how to fix this?

wafliron, 2024-06-24:
Unfortunately not. It reads like a flat out code bug to me, like there’s a typo in the array of possible values or something like that. Hopefully it’ll get looked at at some stage.

Judufu, 2024-06-27:
Do you know any workaround? It used to work fine for me.. Maybe downgrade the z2mqtt version?

phjlljp, 2024-06-27:
I am also experiencing this issue. The error seems clear to me.

Looks to be caused by the mismatch between what Zigbee2MQTT is sending and what the device expects. Zigbee2MQTT is sending the update using the "+" character instead of the word "and" in the payload.

It should just require an update to fix the payload, right?

wafliron, 2024-07-02:
For anyone following this - looks like someone else had already reported the same issue (not sure how I missed that, I did search!) - and it's been fixed in Z2M 1.39.0: https://github.com/Koenkk/zigbee-herdsman-converters/issues/7590

closed 2024-07-02 by wafliron

thedom4302, 2024-07-02:
> fixed in Z2M 1.39.0: Koenkk/zigbee-herdsman-converters#7590

Awesome

Cheers,
Dom