Koenkk/zigbee2mqtt #23031: Can't add PowerTag A9MEM1570 Schneider electric (ZB Green power)

mti-sk, 2024-06-13:
### What happened?

Hi,
Apparently, the powertag A9MEM1570 is a device supporting the ZB greenpower (GP) standard. These devices cannot be paired to ZB2MQTT directly, but only through the ZB GreenPower Proxy (GPP). These devices are very difficult to find, they are not marked, they are identified only by the experiences of users. They usually contain a cluster of 242 input and output green power.
[image]

For example, the following were identified:
1. Router AOTEC WG001-Z01 (output GP cluster only, apparently cannot be used as GPP)
2. Philips Smart Plug HUE LOM008 (9290030509) (output GP cluster only, apparently cannot be used as GPP)
3. Ikea Bulb TRADFRI LED2103G5 (input and output cluster 242 greenPower) - now apparently the only device that can be used as GPP

But even if there is a device in the network that supports GPP, the powertag A9MEM1570 cannot be connected to the network.
If someone has succeeded, they are certainly not able to reproduce the progress that led to success. I consider the version of the firmware (Router, GPP, and the power tag itself) to be essential, which may once have enabled the device to be paired. What was once possible is no longer possible.
I would suggest to mark in the database of supported devices:
1. Mark devices that only support the ZB GreenPower standard - because they are very difficult to connect, if at all
2. Mark devices that can act as GreenPower proxies (GPP)

Furthermore, it is necessary to approach the GreenPower devices in the project (ZB2MQTT) somehow:
1. either greenpower support will be directly in ZB2MQQT
or
2. a list of supported GreenPower Proxy(GPP) will be created
or
4. greenPower devices will be marked as unsupported

I won't be the only user annoyed by greenpower devices.

### What did you expect to happen?

easy connection of the powertag A9MEM1570 to the ZB network.

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

7.4.2 [GA]

### Adapter

Sonoff ZB Dongle E, cfg as adapter: ember

### Setup

Odroid N2+, Ubuntu 24.04, source from github

### Debug log

[2024-06-13 09:18:41] info:     z2m: Logging to console, file (filename: log.log)
[2024-06-13 09:18:41] debug:    z2m: Loaded state from file /opt/zigbee2mqtt/data/state.json
[2024-06-13 09:18:41] info:     z2m: Starting Zigbee2MQTT version 1.38.0 (commit #0e6da85)
[2024-06-13 09:18:41] info:     z2m: Starting zigbee-herdsman (0.49.2)
[2024-06-13 09:18:41] debug:    z2m: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqt>
[2024-06-13 09:18:42] debug:    zh:adapter: Path '/dev/ttyACM0' is valid for 'EmberAdapter'
[2024-06-13 09:18:42] info:     zh:ember: Using default stack config.
[2024-06-13 09:18:42] debug:    zh:ember: Using delay=5.
[2024-06-13 09:18:42] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":xx,"panID":6754,"extendedPanID":[xx>
[2024-06-13 09:18:42] info:     zh:ember: ======== Ember Adapter Starting ========
[2024-06-13 09:18:42] debug:    zh:ember: Starting EZSP with stack configuration: "[object Object]".
[2024-06-13 09:18:42] info:     zh:ember:ezsp: ======== EZSP starting ========
[2024-06-13 09:18:42] info:     zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-13 09:18:42] info:     zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-06-13 09:18:42] debug:    zh:ember:uart:ash: Opening serial port with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true}
[2024-06-13 09:18:42] info:     zh:ember:uart:ash: Serial port opened
[2024-06-13 09:18:42] info:     zh:ember:uart:ash: ======== ASH starting ========
[2024-06-13 09:18:42] error:    zh:ember:uart:ash: Received frame with CRC error
[2024-06-13 09:18:42] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-13 09:18:42] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-13 09:18:42] error:    zh:ember:uart:ash: Received frame with CRC error
[2024-06-13 09:18:42] debug:    zh:ember:uart:ash: ---> [FRAME type=RST]
[2024-06-13 09:18:42] debug:    zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2024-06-13 09:18:42] debug:    zh:ember:uart:ash: Waiting for RSTACK... 100/2500
[2024-06-13 09:18:42] debug:    zh:ember:uart:ash: Waiting for RSTACK... 200/2500
[2024-06-13 09:18:42] debug:    zh:ember:uart:ash: Waiting for RSTACK... 300/2500
[2024-06-13 09:18:42] debug:    zh:ember:uart:ash: Waiting for RSTACK... 400/2500
[2024-06-13 09:18:42] debug:    zh:ember:uart:ash: Waiting for RSTACK... 500/2500
[2024-06-13 09:18:42] debug:    zh:ember:uart:ash: Waiting for RSTACK... 600/2500
[2024-06-13 09:18:43] debug:    zh:ember:uart:ash: Waiting for RSTACK... 700/2500
[2024-06-13 09:18:43] debug:    zh:ember:uart:ash: Waiting for RSTACK... 800/2500
[2024-06-13 09:18:43] debug:    zh:ember:uart:ash: Waiting for RSTACK... 900/2500
[2024-06-13 09:18:43] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1000/2500
[2024-06-13 09:18:43] debug:    zh:ember:uart:ash: <--- [FRAME type=RSTACK]
[2024-06-13 09:18:43] debug:    zh:ember: NCP version info: {"ezsp":13,"revision":"7.4.2 [GA]","build":0,"major":7,"minor":4,"patch":2,"special":0,"type":170}

Koenkk, 2024-06-15:
@burmistrzak the TI SDK doesnt support a green power proxy for the coordinator, so thats why a dedicated router is needed. From my knowledge, all Hue devices support green power.

Koenkk, 2024-06-15:
I guess it makes sense to add a GP part of the docs to confirm what devices (not on the device page itself as that would be too scattered). Would you like to create a PR for it?

Koenkk, 2024-06-15:
That would be a good place indeed!

chris-1243, 2024-06-20:
@mti-sk Have you tried to pair your powertag directly to the coordinator ? As you are on `ember,` you should normally be able to do it. I just had a try on a test network (Dongle-E fw 7.4.3) and a ZGP PTM216Z paired directly to the coordinator without any other devices in the network.

I had some tries at the beginning of `ember` driver and as far as I remember, it was difficult to pair a ZGP device via a specific router. I was unable to sniff traffic unless someone knows how to sniff with a TI CC2652P...

mti-sk, 2024-08-02:
Hi,
The problem with ZB GreenPower device pairing persists even in the new version of ZB2MQTT (1.39). Pairing does not work either directly or via the ZigBee GreenPower Proxy (in my case: Ikea LED2103G5).
The following message pops up when browsing: Error: Parameter 'tempMaster' is missing.
Is it a bug in the Sonoff Dognle-E firmware (v7.4.2 or v7.4.3), or is it a ZB2MQTT bug?
What version can I downgrade the dongle to to bypass: 'tempMaster' is missing ?
I also wondered if the ZB GreenPower devices do not require a specific zigbee channel, e.g.:ch5 ?? I think my network is on channel 11.
Why are some messages ignored as coming from the coordinator when they appear to be from ZG GreenPower? (last one in log)

[2024-08-02 05:17:23] info:     z2m: Starting Zigbee2MQTT version 1.39.0 (commit #0326926)
[2024-08-02 05:17:23] info:     z2m: Starting zigbee-herdsman (0.50.1)
[2024-08-02 05:17:25] debug:    zh:ember: NCP version info: {"ezsp":13,"revision":"7.4.3 [GA]","build":0,"major":7,"minor":4,"patch":3,"special":0,"type":170}
**[2024-08-02 05:18:27] error:    zh:controller:greenpower: Error: Parameter 'tempMaster' is missing**
[2024-08-02 05:18:27] debug:    zh:controller: Received payload: clusterID=33, address=0, groupID=2948, endpoint=242, destinationEndpoint=242, wasBroadcast=true, linkQuality=230, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":1,"commandIdentifier":0},"payload":{"options":0,"srcID":0,"frameCounter":4294967295,"commandID":227,"payloadSize":1,"commandFrame":{"nextChannel":5,"nextNextChannel":8}},"command":{"ID":0,"parameters":[{"name":"options","type":33},{"name":"srcID","type":35,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":0}]},{"name":"gpdIEEEAddr","type":240,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"gpdEndpoint","type":32,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"frameCounter","type":35},{"name":"commandID","type":32},{"name":"payloadSize","type":32},{"name":"commandFrame","type":1009},{"name":"gppNwkAddr","type":33,"conditions":[{"type":"bitMaskSet","param":"options","mask":16384}]},{"name":"gppGddLink","type":32,"conditions":[{"type":"bitMaskSet","param":"options","mask":16384}]}],"name":"notification"}}
[2024-08-02 05:18:27] debug:    z2m: Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"nextChannel":5,"nextNextChannel":8},"commandID":227,"frameCounter":4294967295,"options":0,"payloadSize":1,"srcID":0}' from endpoint 242 with groupID 2948, **ignoring since it is from coordinator**

mti-sk, 2024-08-09:
Hi,
I have to confirm that the ZBDongle P adapter (based on CC2652P)
works very well for pairing with PowerTag. I got my hands on one of the last pieces of ZBDongle-P and the pairing was done in 2s. Of course, direct pairing is not possible, a ZB GreenPower Proxy is needed (in my case: Ikea LED2103G5)

The Sonoff ZBDongle-E (V2 model, EFR32MG21) is a good coordinator, but to my great disappointment it is not possible to use it for ZB GreenPower devices. (error: zh:controller:greenpower: Error: Parameter 'tempMaster' is missing)
So now please try to advise whether I should migrate the entire network to the older ZBDongle-P or wait for the ZBDongle-E firmware bug to be fixed ???? or is the problem in ZB2MQTT ????

mti-sk, 2024-08-30:
Hi,
could someone please comment, what is your opinion, where is the problem with:
error: zh:controller:greenpower: Error: Parameter 'tempMaster' is missing
This is an error in:
1. in firmware Sonoff ZBDongle-E (V2 model, EFR32MG21) ?
2. directly in ZB2MQTT ???
Can this be solved somehow quickly?
@Koenkk?

mti-sk, 2024-09-03:
Hi,
This seems to fix: #1151 - Ember: fix GP proxied messages handling. solved, at least partially solved ZGP device pairing (at least for now: ZB2MQTT v1.40.0, ZBDongle-E fw7.4.4 ).
My PowerTag A9MEM1570 can now be paired via GPP to the network with the ZBDongle-E coordinator (V2 model, EFR32MG21) .
Problems with the missing parameter :greenpower: Error: Parameter 'tempMaster' is missing still persist, but seem to be non-fatal now and the A9MEM1570 is communicating with the network. Thank you all for your efforts.

senna1992, 2024-09-08:
Here's some error log for pairing r9mem powertag:
`error 2024-09-08 09:32:07zh:controller:greenpower: Error: Parameter 'tempMaster' is missing at ZclFrame.writePayloadCluster (/app/node_modules/zigbee-herdsman/src/zspec/zcl/zclFrame.ts:133:23) at ZclFrame.toBuffer (/app/node_modules/zigbee-herdsman/src/zspec/zcl/zclFrame.ts:77:18) at EmberAdapter.sendZclFrameToAll (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2595:31) at GreenPower.onZclGreenPowerData (/app/node_modules/zigbee-herdsman/src/controller/greenPower.ts:226:40) at processTicksAndRejections (node:internal/process/task_queues:95:5) at runNextTicks (node:internal/process/task_queues:64:3) at processImmediate (node:internal/timers:447:9) at Controller.onZclPayload (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:688:13)
info 2024-09-08 09:32:11zh:controller:greenpower: Received commissioning from '25115'
error 2024-09-08 09:32:11zh:controller:greenpower: Error: Parameter 'tempMaster' is missing at ZclFrame.writePayloadCluster (/app/node_modules/zigbee-herdsman/src/zspec/zcl/zclFrame.ts:133:23) at ZclFrame.toBuffer (/app/node_modules/zigbee-herdsman/src/zspec/zcl/zclFrame.ts:77:18) at EmberAdapter.sendZclFrameToAll (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2595:31) at GreenPower.onZclGreenPowerData (/app/node_modules/zigbee-herdsman/src/controller/greenPower.ts:154:44) at processTicksAndRejections (node:internal/process/task_queues:95:5) at runNextTicks (node:internal/process/task_queues:64:3) at processImmediate (node:internal/timers:447:9) at Controller.onZclPayload (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:688:13)
info 2024-09-08 09:32:11zh:controller:greenpower: Received commissioning from '46885'
info 2024-09-08 09:32:12z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x00000000e205621b/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x00000000e205621b"],"manufacturer":"","model":"Automatically generated definition (GreenPower_92)","name":"0x00000000e205621b","via_device":"zigbee2mqtt_bridge_0xf082c0fffe433135"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"0x00000000e205621b_linkquality","origin":{"name":"Zigbee2MQTT","sw":"1.40.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0x00000000e205621b","unique_id":"0x00000000e205621b_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
warning 2024-09-08 09:32:12z2m: Device '0x00000000e205621b' with Zigbee model 'GreenPower_92' and manufacturer name 'undefined' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info 2024-09-08 09:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Automatically generated definition","exposes":[{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"GreenPower_92","options":[],"supports_ota":false,"vendor":""},"friendly_name":"0x00000000e205621b","ieee_address":"0x00000000e205621b","status":"successful","supported":false},"type":"device_interview"}'
info 2024-09-08 09:32:13zh:controller:greenpower: Received commissioning from '25115'
error 2024-09-08 09:32:13zh:controller:greenpower: Error: Parameter 'tempMaster' is missing at ZclFrame.writePayloadCluster (/app/node_modules/zigbee-herdsman/src/zspec/zcl/zclFrame.ts:133:23) at ZclFrame.toBuffer (/app/node_modules/zigbee-herdsman/src/zspec/zcl/zclFrame.ts:77:18) at EmberAdapter.sendZclFrameToAll (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2595:31) at GreenPower.onZclGreenPowerData (/app/node_modules/zigbee-herdsman/src/controller/greenPower.ts:154:44) at processTicksAndRejections (node:internal/process/task_queues:95:5) at runNextTicks (node:internal/process/task_queues:64:3) at processImmediate (node:internal/timers:447:9) at Controller.onZclPayload (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:688:13)
info 2024-09-08 09:32:13zh:controller:greenpower: Received commissioning from '46885'
info 2024-09-08 09:32:14zh:controller:greenpower: Received commissioning from '25115'
error 2024-09-08 09:32:14zh:controller:greenpower: Error: Parameter 'tempMaster' is missing at ZclFrame.writePayloadCluster (/app/node_modules/zigbee-herdsman/src/zspec/zcl/zclFrame.ts:133:23) at ZclFrame.toBuffer (/app/node_modules/zigbee-herdsman/src/zspec/zcl/zclFrame.ts:77:18) at EmberAdapter.sendZclFrameToAll (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2595:31) at GreenPower.onZclGreenPowerData (/app/node_modules/zigbee-herdsman/src/controller/greenPower.ts:154:44) at processTicksAndRejections (node:internal/process/task_queues:95:5) at runNextTicks (node:internal/process/task_queues:64:3) at processImmediate (node:internal/timers:447:9) at Controller.onZclPayload (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:688:13)`

senna1992, 2024-09-08:
Here's the status report of r9m20 (1phase powertag):
`{
    "ac_frequency": 50,
    "current_phase_a": 0.18,
    "linkquality": 220,
    "power": 32,
    "power_apparent": 43,
    "power_factor": 76,
    "power_phase_a": 32,
    "voltage_phase_a": 238,
    "current_phase_b": null,
    "current_phase_c": null,
    "energy": 7.99,
    "energy_phase_a": 7.994,
    "energy_phase_b": null,
    "energy_phase_c": null,
    "power_phase_b": null,
    "power_phase_c": null,
    "voltage_phase_ab": null,
    "voltage_phase_b": null,
    "voltage_phase_bc": null,
    "voltage_phase_c": null,
    "voltage_phase_ca": null
}`

rbswift, 2024-10-01:
I've got the same problem using a A9MEM1570 powertag and CP210x usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus controller via an ikea LED1935C3 router.

Just this entry repeatedly logged but all values N/A except Linkquality.
zh:controller:greenpower: Received commissioning from '64135'

I'm very happy to experiment and provide more logs if that's helpful?

estimadarocha, 2024-11-12:
> Hi, This seems to fix: [#1151 ](https://github.com/Koenkk/zigbee-herdsman/issues/1151)- Ember: fix GP proxied messages handling. solved, at least partially solved ZGP device pairing (at least for [...]

@mti-sk tried exactly like you but with newer version of zb2mqtt and newer fw 8.0.2 from @Nerivec and no good results

estimadarocha, 2024-11-30:
@Koenkk and @Nerivec i was able to pair the powertags... but to be able to do this i had to use 2 adapters one as ncp and another as router. And if i remove the router one after the pairing after a few second's i start receiving messages failed broadcast and device stop communication.

mti-sk, 2024-12-02:
Hi,
I can confirm that pairing PowerTags (in most cases it is a Zigbee GreenPower devices - ZBGP) is possible with both the ZB-Dongle P coordinator (adapter: zstack ) and the ZB-Dongle E (adapter: ember).
This works since Zigbee2MQTT v1.40.0, ZBDongle-E fw7.4.4, apparently fixed with https://github.com/Koenkk/zigbee-herdsman/issues/1151 - Ember: fix GP proxied messages handling.
But be careful, in any case you need to have a ZB GreenPower Proxy(ZB GPP) device in the network (contains Endpoint 242 - Input and output, in my case: Ikea LED2103G5). Without a ZB GPP in the network, the Zigbee GreenPower device cannot be paired and obviously not operated.
Although the operation of the ZB GP device now works, error messages persist in the network, for example :error 2024-MM-DD zh:controller:greenpower: Error: Parameter 'tempMaster' is missing.
Fortunately, this is not fatal for communication.
**It would be advisable to add Endpoint 242 support to the coordinator so that it is possible to pair and operate ZB GP devices without ZB GPP**

elamperti, 2025-02-16:
Not stale (commenting to keep the issue open)

JonasMH, 2025-03-13:
Not stale

Nerivec, 2025-03-19:
@mti-sk could you try this again (about the errors) with latest-dev?
https://github.com/Koenkk/zigbee-herdsman/pull/1349 hopefully solves a lot of pending issues. 😉
_Please note the logging changed a bit, so you cannot rely on previous "search patterns" if you had any, to parse the logs._

closed 2025-03-27 by Nerivec