Koenkk/zigbee2mqtt #23033: Legrand/BTicino (FC80CC) 412171: Pairing Issues / Failed to configure

CodeForLove83, 2024-06-13:
### What happened?

Device: Legrand/Bticino 412171
Firmware build date: 20240308
Firmware version: 0053

Pairing procedure completes but failing to configure `haElectricalMeasurement.configReport` (see Debug Log).

### What did you expect to happen?

Configuration successfully completed.

### How to reproduce it (minimal and precise)

Put the device in pairing mode and enable Permit Join on Z2M. The device is paired but the configuration does not complete successfully.

### Zigbee2MQTT version

Zigbee2MQTT 1.38.0-1

### Adapter firmware version

0x26510900

### Adapter

ConBee3

### Setup

Z2A running on HA (Core 2024.6.2, Supervisor 2024.06.0)

### Debug log

`Failed to configure '0x00047400012497fc', attempt 12 (Error: ZCL command 0x00047400012497fc/1 haElectricalMeasurement.configReport([{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":5,"attribute":"activePower"},{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":null,"attribute":"rmsCurrent"},{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":null,"attribute":"rmsVoltage"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:316:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:771:26) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:557:9) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:76:13) at result.configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:1488:29) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:173:21) at Configure.configure (/app/lib/extension/configure.ts:121:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))`

CodeForLove83, 2024-06-24:
Hi, I could fix the issue with the following external definition.

Steps to follow for testing:

- Save the following piece of code to `FC80CC.js` and save the file to the zigbee2mqtt path where the `configuration.yaml` file is stored (e.g., `/usr/share/hassio/homeassistant/zigbee2mqtt/`)
- Modify the zigbee2mqtt `configuration.yaml` file as follow:
```
advanced:
  log_level: debug
external_converters:
  - FC80CC.js
```
- Restart zigbee2mqtt (if no errors occur in the log file then you should read `DIN contactor module (WIP) ` in the description of the device

If you try to reconfigure the device you shouldn't get any error.

That's my first attempt to play with external definition and with just a little experience with Zigbee.

```
const {identify, electricityMeter, onOff, commandsOnOff} = require('zigbee-herdsman-converters/lib/modernExtend');
const {fzLegrand, tzLegrand} = require('zigbee-herdsman-converters/lib/legrand');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: [' Contactor                     '],
	model: '412171',
    vendor: ' Legrand                       ',
	description: 'DIN contactor module (WIP)',
	whiteLabel: [{vendor: 'BTicino', model: 'FC80CC'}],
	ota: ota.zigbeeOTA,
    extend: [onOff()],
    fromZigbee: [fz.identify, fzLegrand.cluster_fc01, fz.ignore_basic_report, fz.ignore_genOta, fz.electrical_measurement],
    toZigbee: [tz.legrand_device_mode, tzLegrand.identify, tzLegrand.auto_mode, tz.electrical_measurement_power],
	meta: {},
	exposes: [
        e.power().withAccess(ea.STATE_GET),
		e.enum('device_mode', ea.ALL, ['switch', 'auto'])
		.withDescription('Switch: allow manual on/off, auto uses contact\'s C1/C2 wired actions for Peak/Off-Peak electricity rates'),
		e.enum('auto_mode', ea.STATE_SET, ['off', 'auto', 'on_override'])
		.withDescription('Off/auto/on (override) (works only if device is set to "auto" mode)'),
		],
	configure: async (device, coordinatorEndpoint) => {
		const endpoint = device.getEndpoint(1);
		await reporting.onOff(endpoint);
		await reporting.bind(endpoint, coordinatorEndpoint, ['genIdentify', 'genOnOff', 'haElectricalMeasurement']);
		await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
		await reporting.activePower(endpoint);
	},
};

module.exports = definition;
```

Koenkk, 2024-06-25:
Integrated the fix, thanks!

closed 2024-06-25 by Koenkk