Koenkk/zigbee2mqtt #23034: CC2652P Failed to start zigbee

DuchkPy, 2024-06-13:
### What happened?

Hello,

I have installed z2m on a debian container, by using tteck script on my Proxmox server.
Once done, I have setup the passthrough of the USB adaptor.

On the shell of my PVE, if I run this command: `ls -l /dev/serial/by-id` it returns:
```
total 0
lrwxrwxrwx 1 root root 13 Jun 13 14:29 usb-0658_0200-if00 -> ../../ttyACM0
lrwxrwxrwx 1 root root 13 Jun 13 13:25 usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8888dce83a96ed118cb56b4ce259fb3e-if00-port0 -> ../../ttyUSB0
```
so my ZigBee adapter is on port ttyUSB0 (ttyACM0 is a Zwave dapter, which is running well on a dedicated Zweave-JS-UI container).

On the shell of z2m container, if I run `ls -l /dev/tty*` it returns:
```
crw-rw-rw- 1 nobody nogroup   5, 0 Jun 13 14:47 /dev/tty
crw------- 1 root   tty     136, 1 Jun 13 14:56 /dev/tty1
crw--w---- 1 root   tty     136, 2 Jun 13 13:28 /dev/tty2
crw-rw-rw- 1 root   dialout 188, 0 Jun 13 14:48 /dev/ttyUSB0
```
And when I do `test -w /dev/ttyUSB0 && echo success || echo failure` it returns `success`.
So, I'm expecting the passthrough is working properly. Maybe, I'm wrong.

My `configuration.yaml` of z2m is the following:
```
frontend:
  port: 9572
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.1.110:1883
  user: XXXXXXX
  password: XXXXXXXXX
  keepalive: 60
  reject_unauthorized: true
  version: 4
serial:
  port: /dev/ttyUSB0
  adapter: ember
advanced:
  pan_id: GENERATE
  network_key: GENERATE
  channel: 20
```

When I try to start, I have an error indicating that it failed to start zigbee (see debug log for more detail).
I read the `https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html` page, try some modification without succeed (at least, I have the feeling this page helped me to make some step, compare to where I was in first step).

NB: I have never succeed to update the adapter firmware, and as z2m isn't starting, I'm not able to indicate what is the version of the firmware.

### What did you expect to happen?

I expected that z2m is starting

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

Unknown

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P Coordinator

### Setup

Container on NUC

### Debug log

cd /opt/zigbee2mqtt && npm start

> zigbee2mqtt@1.38.0 start
> node index.js

[2024-06-13 14:48:04] info:     z2m: Logging to console, file (filename: log.log)
[2024-06-13 14:48:04] info:     z2m: Starting Zigbee2MQTT version 1.38.0 (commit #f1847301)
[2024-06-13 14:48:04] info:     z2m: Starting zigbee-herdsman (0.49.2)
[2024-06-13 14:48:05] info:     zh:ember: Using default stack config.
[2024-06-13 14:48:05] info:     zh:ember: ======== Ember Adapter Starting ========
[2024-06-13 14:48:05] info:     zh:ember:ezsp: ======== EZSP starting ========
[2024-06-13 14:48:05] info:     zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-13 14:48:05] info:     zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-06-13 14:48:05] info:     zh:ember:uart:ash: Serial port opened
[2024-06-13 14:48:05] info:     zh:ember:uart:ash: ======== ASH starting ========
[2024-06-13 14:48:07] info:     zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-13 14:48:07] info:     zh:ember:uart:ash: ======== ASH starting ========
[2024-06-13 14:48:10] info:     zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-13 14:48:10] info:     zh:ember:uart:ash: ======== ASH starting ========
[2024-06-13 14:48:12] info:     zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-13 14:48:12] info:     zh:ember:uart:ash: ======== ASH starting ========
[2024-06-13 14:48:15] info:     zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-13 14:48:15] info:     zh:ember:uart:ash: ======== ASH starting ========
[2024-06-13 14:48:17] error:    z2m: Error while starting zigbee-herdsman
[2024-06-13 14:48:17] error:    z2m: Failed to start zigbee
[2024-06-13 14:48:17] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-06-13 14:48:17] error:    z2m: Exiting...
[2024-06-13 14:48:17] error:    z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:850:19)
    at EmberAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2683:24)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:127:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:62:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:109:27)
    at start (/opt/zigbee2mqtt/index.js:107:5)

Tonio16, 2024-06-13:
In adition to Siandco solution, here an advice, you should use the by-id instead the ttyUSBxyz port.

Antoine

DuchkPy, 2024-06-13:
Hello,

Thank you for your feedback.
It looks like I progressed, but still don't acheived the end

After update of the adapter, and launching again, I have this log :
```
> zigbee2mqtt@1.38.0 start
> node index.js

[2024-06-13 20:25:57] info:     z2m: Logging to console, file (filename: log.log)
[2024-06-13 20:25:57] info:     z2m: Starting Zigbee2MQTT version 1.38.0 (commit #f1847301)
[2024-06-13 20:25:57] info:     z2m: Starting zigbee-herdsman (0.49.2)
[2024-06-13 20:25:57] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-06-13 20:25:57] info:     zh:zstack:znp: Serialport opened
[2024-06-13 20:25:58] error:    zh:adapter:zstack:manager: Configuration is not consistent with adapter state/backup!
[2024-06-13 20:25:58] error:    zh:adapter:zstack:manager: - PAN ID: configured=29220, adapter=30044
[2024-06-13 20:25:58] error:    zh:adapter:zstack:manager: - Extended PAN ID: configured=00124b002e1e828e, adapter=00124b002e1e828e
[2024-06-13 20:25:58] error:    zh:adapter:zstack:manager: - Network Key: configured=47f6a55f111cb07e6e46f27a914794be, adapter=9f7064af1714d798d392b17b6835bef2
[2024-06-13 20:25:58] error:    zh:adapter:zstack:manager: - Channel List: configured=20, adapter=20
[2024-06-13 20:25:58] error:    zh:adapter:zstack:manager: Please update configuration to prevent further issues.
[2024-06-13 20:25:58] error:    zh:adapter:zstack:manager: If you wish to re-commission your network, please remove coordinator backup at /opt/zigbee2mqtt/data/coordinator_backup.json.
[2024-06-13 20:25:58] error:    zh:adapter:zstack:manager: Re-commissioning your network will require re-pairing of all devices!
[2024-06-13 20:25:58] error:    z2m: Error while starting zigbee-herdsman
[2024-06-13 20:25:58] error:    z2m: Failed to start zigbee
[2024-06-13 20:25:58] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-06-13 20:25:58] error:    z2m: Exiting...
[2024-06-13 20:25:58] error:    z2m: Error: startup failed - configuration-adapter mismatch - see logs above for more information
    at ZnpAdapterManager.determineStrategy (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:223:35)
    at ZnpAdapterManager.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:57:26)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:127:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:62:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:109:27)
    at start (/opt/zigbee2mqtt/index.js:107:5)
root@z2m:/opt/zigbee2mqtt# ^C
root@z2m:/opt/zigbee2mqtt# sudo systemctl stop zigbee2mqtt
root@z2m:/opt/zigbee2mqtt# nano /opt/zigbee2mqtt/data/configuration.yaml
root@z2m:/opt/zigbee2mqtt# cd /opt/zigbee2mqtt && npm start
```
It looks like the configuration isn't consistent with the adapter.

I can manually update the generated `PAN ID` to `30044`, but I don't know how to correctly update the `Network key` (I tried a copy/paste and it raised an error saying it should be an array).

Also, I know that's it's better to use the `by-id`, but I struggle to have it working.
I'm able to find this name, only on the shell of the PVE, not directly from the container. So, I'm not sure about the complete path (`/dev/by-id`, or `/by-id`, ...)
What is your advice to improve this ?

siandco, 2024-06-14:
Ok, according to the portion of log you've supplied your adapter is using `9f7064af1714d798d392b17b6835bef2` for its network key. To use this key, you need to modify your configuration.yaml have the following:  
`network_key: [ 0x9f, 0x70, 0x64, 0xaf, 0x17, 0x14, 0xd7, 0x98, 0xd3, 0x92, 0xb1, 0x7b, 0x68, 0x35, 0xbe, 0xf2 ]`

DuchkPy, 2024-06-14:
Perfect !
Now z2m started, and is working as it should be.

Thank you for your support.
On the `configuration.yaml` file it was displayed in another way, it wasn't that obvious to change it like that.

closed 2024-12-29 by github-actions[bot]