Koenkk/zigbee2mqtt #23035: Freezing HA when Restart the System or the zigbee2mqtt addon

grfernandes54, 2024-06-13:
### What happened?

Each time that I aske to restart my homeassiant, the VM freeze and need to be power off When I ask to restart zigbee2mqtt addon, the VM freeze to and only back to work if I power off the VM and power on again.

I've uninstall Z2M add-on and after uninstall, the HA was able to reload without any issue. After I install the Z2M again the HA start to freeze when I try to reload the system

### What did you expect to happen?

I expect that HA can be reload without need to power off the VM

### How to reproduce it (minimal and precise)

when reload HA or restart the zigbee2mqtt addon the VM freeze

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

 20230507

### Adapter

SMLIGHT CC2652P Zigbee USB Adapter SLZB-02

### Setup

Addon on Home Assistant on Virtual Box Machine

### Debug log

[log.log](https://github.com/user-attachments/files/15822506/log.log)
[log.log](https://github.com/user-attachments/files/15822506/log.log)

grfernandes54, 2024-06-21:
After upgrade do Home Assistant OS 12.4 and Home Assistant Core 2024.6.4 the situation become worriest. Now each time that z2m start the virtual machine freeze

turleveta, 2024-07-05:
tl;dr. Try switching the VirtualBox USB controller to USB 3.0 (xHCI), this might solve your issue. Note: A backup of the Zigbee stick nvram could be useful, and this potentially only works when using a USB 2.0 extension cable connected to the Zigbee stick. 

I experienced similar issues using VirtualBox 7.0.16 running the latest HA version, Zigbee2mqtt v1.38.0-1 addon, and the Sonoff Dongle Plus (P) running the latest coordinator firmware from 20230507 connected to a USB 2.0 extension cable plugged in a USB 3.0 port. The guest and host system would both crash during the shutdown / restart of the Zigbee2mqtt addon, which also prevented the upgrade to Zigbee2mqtt v1.39.0-1. During a clean start of HA the whole Zigbee network would work absolutely fine without delays. It's only the shutdown process of the Zigbee2mqtt addon which was an issue. I verified this behaviour using a completely clean HA install on both a Linux and Windows host using the latest HA VirtualBox image (haos_ova-12.4.vdi). I also tried flashing a known working nvram backup from the Zigbee stick and even resetting the nvram completely. This did not solve the issue. However, the Zigbee2mqtt addon worked perfectly fine on a clean HA install on a Raspberry Pi 4 using the same Zigbee stick, where the Zigbee2mqtt addon could be shutdown/ restarted without any issues. 

This indicated an issue with the VirtualBox USB connection to the Zigbee adapter, where the shutdown of the Zigbee adapter somehow made VirtualBox crash. I used the USB 2.0 port on the Raspberry Pi, but as a USB 2.0 extension cable was used for the Zigbee stick, the USB 3.0 port shouldn't be an issue using VirtualBox (and it hasn't been for the last two years). It seemed more likely there were new compatibility issues with the USB controller and VirtualBox. As VirtualBox recently did a major upgrade release from v6 to v7 it is more likely something changed here. This made me change the default USB controller in the VirtualBox host from USB 2.0 (OHCI + EHCI) to USB 3.0 (xHCI) found in Settings -> USB. 

After this change my clean HA installation started working properly where the Zigbee2mqtt addon could be shutdown / restarted. Although the usage of the USB 3.0 controller is generally not recommended I wonder if VirtualBox v7 introduced some changes for better compatibility. After the change to the USB 3.0 controller my Zigbee network was not recognized anymore on my live HA installation where I got the Zigbee2mqtt notification "Configuration is not consistent with adapter state/backup!". I would have expected this notification already to have happened before, as I used the Zigbee stick in my live system after I did the initial reset of the nvram of the Zigbee stick, but somehow only happened after the switch to the USB 3.0 controller.  After flashing my latest nvram backup, which was taken before the nvram reset, everything worked as expected in my live system without needing to repair all my devices. Shutdowns and restarts of the Zigbee2mqtt addon are working again, which also allowed me to upgrade to the latest Zigbee2mqtt version 1.39.0-1 without any issues.

I expect that the flashing of the Zigbee nvram backup is not necessary for most people as they wouldn't have performed a nvram reset, but I can't verify this. So it might be smart to do a backup of the nvram anyways. I also wonder if the usage of the USB 3.0 controller would cause issues in case no USB 2.0 extension cable is used for the Zigbee stick, but I haven't tested this, as the extension cable is recommended anyways when a USB 3.0 port is used.

JokeRsevilla, 2024-07-21:
Thank you! I had the same problem and I was able to fix it with your instructions!

KILYBMW, 2024-10-04:
Thanks, I was able to solve the problem too.

closed 2024-12-29 by github-actions[bot]