Koenkk/zigbee2mqtt #23040: [New device support]: Airzone Aidoo Zigbee AZAI6ZBEMHI

GitCharlieHub, 2024-06-13:
### Link

https://www.airzonecontrol.com/aa/en/control-solutions/aidoo/zigbee/

### Database entry

{"id":66,"type":"Router","ieeeAddr":"0x4c5bb3fffef0d5dc","nwkAddr":64481,"manufId":4747,"manufName":"Airzone","powerSource":"DC Source","modelId":"Aidoo Zigbee","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,4,6,513,514],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"powerSource":4,"disableLocalConfig":0}},"genOnOff":{"attributes":{"onOff":0}},"hvacThermostat":{"attributes":{"localTemp":2200,"systemMode":0}},"genIdentify":{"attributes":{"identifyTime":0}},"genGroups":{"attributes":{"nameSupport":0}},"hvacFanCtrl":{"attributes":{"fanMode":5}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeb772de","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":49,"stackVersion":1,"hwVersion":1,"dateCode":"2022","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1718308861220}

### Comments

The device only exposes on/off after discovery. However according to the technical manual here:
[https://doc.airzonecloud.com/Documentation/AZ6/AI6/MI_AZAI6ZBE_MUL.pdf](url)
It exposes several clusters and HVAC temperature / fan settings, ,etc...

I saw that Z2M seems to already support it:
https://github.com/home-assistant/core/pull/117191/files

As to contribute to the community I saw the https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html and tried following it to add support, however I don't see any converter messages in the debug log and have tried setting up a exposes: and configure: sections in a .js file based on other thermostats. However nothing seems to change anything on the device.

I got to this:
`const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const {identify, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['Aidoo Zigbee'],
    model: 'Aidoo Zigbee',
    vendor: 'Airzone',
    description: 'Airzone AIDOO mitsubishi control',
    fromZigbee: [fz.thermostat],
    toZigbee: [tz.thermostat_local_temperature, tz.thermostat_system_mode, tz.thermostat_running_state,
         tz.thermostat_control_sequence_of_operation, tz.thermostat_weekly_schedule],    
    extend: [onOff({"powerOnBehavior":false})],
    meta: {disableDefaultResponse: true},
    configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(5);
            const binds = ['genBasic', 'genIdentify', 'genGroups', 'genOnOff', 'hvacThermostat','hvacFanCtrl'];
            await reporting.bind(endpoint, coordinatorEndpoint, binds);
            await reporting.LocalTemperature(endpoint);
            await reporting.thermostatRunningState(endpoint);
            await reporting.thermostatSystemMode(endpoint);
            await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
            await reporting.thermostatTemperatureSetpointHold(endpoint);
            await reporting.thermostatTemperatureSetpointHoldDuration(endpoint);
        },
};`

Clearly I have no clue of how to do this...

If it helps I saw under the clusters in the Zigbee2MQTT frontend:
Endpoint 1-> Output clusters -> 'genIdentify' ,' genOta'
Endpoint 1-> input clusters -> GenBasic', 'genIdentify', 'genGroups', 'genOnOff', 'hvacThermostat','hvacFanCtrl'
Endpoint 242 -> Output clusters -> greenPower

And in the manual above we have the following clusters and attributes:
BASIC CLUSTER
- ZCL version
- Application version
- Stack version
- HW version
- Power source
- Disable local config 
- Reset to factory defaults

ON/OFF CLUSTER
- OnOff -> Off / On / Toggle

THERMOSTAT CLUSTER
- LocalTemperature = 100 x temperature
- OccupiedCoolingSetpoint
- OccupiedHeatingSetpoint 
- ControlSequenceOfOperation 
        - 0x00 = Cooling only 
        - 0x02 = Heating only
- SystemMode 
        - 0x00 = OFF
        - 0x01 = Auto
        - 0x03 = Cool
        - 0x04 = Heat
        - 0x07 = Fan Only
        - 0x08 = Dry
- ACLouverPosition
        - 0x01 = Fully closed
        - 0x02 = Fully open
        - 0x03 = 1/4 open
        - 0x04 = 1/2 open
        - 0x05 = 3/4 open

FAN CONTROL CLUSTER
- FanMode
        - 0x00 = OFF
        - 0x01 = Low
        - 0x02 = Medium
        - 0x03 = High
        - 0x04 = On
        - 0x05 = Auto
FanModeSequence
        - 0x02 = Low/Med/High/Auto

Any help more than welcome! Thank you!

### External definition

```shell
const {identify, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['Aidoo Zigbee'],
    model: 'Aidoo Zigbee',
    vendor: 'Airzone',
    description: 'Automatically generated definition',
    extend: [identify(), onOff({"powerOnBehavior":false})],
    meta: {},
};

module.exports = definition;
```

GitCharlieHub, 2024-06-13:
Hello again,

Managed to get help from ChatGPT!

Here's the working code:

const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Aidoo Zigbee'],
    model: 'Aidoo Zigbee',
    vendor: 'Airzone',
    description: 'Aidoo Zigbee Controller',
    fromZigbee: [fz.thermostat, fz.on_off, fz.fan],
    toZigbee: [
        tz.thermostat_local_temperature,
        tz.thermostat_occupied_heating_setpoint,
        tz.thermostat_occupied_cooling_setpoint,
        tz.thermostat_system_mode,
        tz.on_off,
        tz.fan_mode,
    ],
    exposes: [
        e.local_temperature(),
        e.numeric('occupied_heating_setpoint', ea.ALL).withUnit('°C').withDescription('Occupied heating setpoint'),
        e.numeric('occupied_cooling_setpoint', ea.ALL).withUnit('°C').withDescription('Specifies the cooling mode setpoint when the room is occupied.'),
        exposes.enum('system_mode', ea.ALL, ['off', 'auto', 'cool', 'heat', 'fan_only', 'dry'])
            .withDescription('Specifies the current operating mode of the thermostat.'),
        e.switch(),
        exposes.enum('fan_mode', ea.ALL, ['off', 'low', 'medium', 'high', 'on', 'auto']).withDescription('Fan mode'),
    ],
    endpoint: (device) => {
        return { 'system': 1 };
    },
    meta: { configureKey: 1 },
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['hvacThermostat', 'hvacFanCtrl', 'genOnOff']);
        await reporting.thermostatTemperature(endpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
        await reporting.onOff(endpoint);
        await reporting.fanMode(endpoint);
    },
};

module.exports = definition;

GitCharlieHub, 2024-06-14:
Hello,
Changes to make it a climate type device and a few other tweaks:

const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Aidoo Zigbee'],
    model: 'Aidoo Zigbee AZAI6ZBEMHI',
    vendor: 'Airzone',
    description: 'Gateway for two-way control and integration of AirCon Units.AZAI6ZBEMHI for Mitsubishi Heavy',
    fromZigbee: [fz.thermostat, fz.on_off, fz.fan],
    toZigbee: [
        tz.thermostat_local_temperature,
        tz.thermostat_occupied_heating_setpoint,
        tz.thermostat_occupied_cooling_setpoint,
        tz.thermostat_system_mode,
        tz.on_off,
        tz.fan_mode,
    ],
    exposes: [
        e.climate()
            .withLocalTemperature()
            .withSystemMode(['off', 'auto', 'cool', 'heat', 'fan_only', 'dry'])
            .withFanMode(['off', 'low', 'medium', 'high', 'on', 'auto'])
            .withSetpoint('occupied_heating_setpoint', 5, 30, 0.5)
            .withSetpoint('occupied_cooling_setpoint', 5, 30, 0.5),
        e.switch(),
    ],
    endpoint: (device) => {
        return { 'system': 1 };
    },
    meta: { configureKey: 1 },
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['hvacThermostat', 'hvacFanCtrl', 'genOnOff']);
        await reporting.thermostatTemperature(endpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
        await reporting.onOff(endpoint);
        await reporting.fanMode(endpoint);
    },
};

module.exports = definition;

Koenkk, 2024-06-15:
Lets continue in your PR

closed 2024-06-15 by Koenkk

wr5345, 2024-07-17:
Hello, I tried to control my Airzone Aidoo Zigbee AZAI6ZBEMHI device. For some reason, the "Fan mode" control does not work. 
I set the fan speed, but when the browser page is refreshed, "Fan mode" is reset to off.
What could be the problem?

https://github.com/user-attachments/assets/06158772-b3e9-4489-b374-5078baf47f23

wr5345, 2024-07-17:
Hello, I tried to control my Airzone Aidoo Zigbee AZAI6ZBEMHI device using your configuration file. For some reason, the "Fan mode" control does not work. 
I set the fan speed, but when the browser page is refreshed, "Fan mode" is reset to off.
What could be the problem?

https://github.com/user-attachments/assets/1d76b91a-3c25-4a5a-83f7-5322ff07faad

@Koenkk 
@GitCharlieHub

GitCharlieHub, 2024-07-17:
Heya. We’ve made some amends and incorporated the Airzone Aidoo Zigbee into the latest Zigbee2mqtt release. You shouldn’t need the config file from this thread. See here ->
Add support for Airzone Aidoo Zigbee - AZAI6ZBEMHI #7636

https://github.com/user-attachments/assets/1d76b91a-3c25-4a5a-83f7-5322ff07faad