Koenkk/zigbee2mqtt #23046: Lidl HG08130A GU10 temperature calibration is off

hayvan96, 2024-06-14:
### What happened?

When I try to change the color temperature of my GU10 spots, it seems that the temperature range is centered on 3800K. In other words, 6500K is cold as expected, but 2000K as well. When sliding from 6500 to 2000, color temperature gets warmer, around 3800 it's orange and when sliding down to 2000, it gets colder.
When using the mired slider in z2m, I have the same behavior.

### What did you expect to happen?

I would expect that the temperature range goes from the warmest to coldest, and not cold-warm-cold.

### How to reproduce it (minimal and precise)

Install Lidl GU10 [HG08130A] and try to change color temperature.

### Zigbee2MQTT version

1.38

### Adapter firmware version

20210708

### Adapter

sonoff-P

### Setup

add-on in home assistant (2024.5.5)

### Debug log

Spots are called WC-évier or WC.
[debug.log](https://github.com/user-attachments/files/15832215/debug.log)

hayvan96, 2024-07-01:
Hi
So I've bought a small Tuya Zigbee gateway to see if there was an issue with the spots :
- When they are paired to that gateway, no problem: when changing temperature
- When I remove and factory reset them, then add them again to Z2M, then the issue described above starts again : minimal mired and maximal mired are cool, somewhere in the middle it gets warmer and orange.

I can't use these GU10 spots with Z2M :( 

What kind of logs or data can I provide to help solving this issue ?

Thanks

RubenKelevra, 2024-11-01:
Weird, I got the similarly named HG08131A and those work mostly fine for me.

I however noticed, that they jump to absolute cold white in one of my rooms from time to time, while they are on, and then slowly dim back to a more appropriate warm color. I attributed this to a bug in Adaptive Lighting, as I never saw this in the other rooms, which run the same bulbs.

But now I'm not so sure anymore.

hayvan96, 2024-11-01:
Well, actually, I've solved my issue. These were second-hand bulbs. I bought a tuya gateway, paired them to it, unpaired them, repaired them in z2M, and they no longer have that behaviour.

closed 2024-11-01 by hayvan96

RubenKelevra, 2024-11-01:
Okay? That's the weirdest fix I've ever heard.

@Koenkk any idea why this has worked?

Koenkk, 2024-11-02:
That's really weird indeed, no clue tbh

RubenKelevra, 2024-11-02:
Alright, maybe just a firmware hickup.

As I said, I have some weird behavior on three of my bulbs as well, two GU10 one E14, while all other like 25 bulbs work fine. No clue what's going on there. 🤷‍♂️