Koenkk/zigbee2mqtt #23054: Please add Tongou TOWSMR1 1-40  _TZE204_kobbcyum

terzo33, 2024-06-14:
### Link

https://it.elcb.net/categoria-di-prodotto/interruttore-intelligente/

### Database entry

{"id":26,"type":"Router","ieeeAddr":"0xdc8e95fffe6546f9","nwkAddr":19851,"manufId":4098,"manufName":"_TZE204_kobbcyum","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�?�-i�?�-i�?�-i�?�-i�?�-i�?�-i�?�-i�?�-i","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_kobbcyum","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{}},"seMetering":{"attributes":{}},"msTemperatureMeasurement":{"attributes":{}}},"binds":[{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0005a5ebca","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-903893911},"lastSeen":1718367387424}

### Comments

with this configuration it partially works, i.e. some sensors are missing and those that work are updated every 5/10 minutes

the missing sensors are:

Temperature threshold
Temperature breaker
Power threshold
Power breaker
Over current threshold
Over current breaker
Over voltage threshold
Over voltage breaker
Under voltage threshold
Under voltage breaker
Power outage memory
Indicator mode
And other....

Thanks

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel

fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_kobbcyum',
        },
    ],
    model: 'TOWSMR1',
    vendor: 'Tongou',
    description: 'Smart circuit breaker',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [tuya.exposes.switch(), e.energy(), e.temperature(), e.power(), e.voltage(), e.current(),
        exposes.enum('fault', ea.STATE, ['clear', 'over_current_threshold', 'over_power_threshold',
            'over_voltage threshold', 'wrong_frequency_threshold']).withDescription('Fault status of the device (clear = nothing)'),
        exposes.enum('threshold_1', ea.STATE, ['not_set', 'over_current_threshold', 'over_voltage_threshold'])
            .withDescription('State of threshold_1'),
        exposes.binary('threshold_1_protection', ea.STATE, 'ON', 'OFF')
            .withDescription('OFF - alarm only, ON - relay will be off when threshold reached'),
        exposes.numeric('threshold_1_value', ea.STATE)
            .withDescription('Can be in Volt or Ampere depending on threshold setting. Setup the value on the device'),
        exposes.enum('threshold_2', ea.STATE, ['not_set', 'over_current_threshold', 'over_voltage_threshold'])
            .withDescription('State of threshold_2'),
        exposes.binary('threshold_2_protection', ea.STATE, 'ON', 'OFF')
            .withDescription('OFF - alarm only, ON - relay will be off when threshold reached'),
        exposes.numeric('threshold_2_value', ea.STATE)
            .withDescription('Setup value on the device'),
        exposes.binary('clear_fault', ea.STATE_SET, 'ON', 'OFF')
            .withDescription('Turn ON to clear last the fault'),
        exposes.text('meter_id', ea.STATE).withDescription('Meter ID (ID of device)'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'energy', tuya.valueConverter.divideBy100],
            [3, null, null], // Monthly, but sends data only after request
            [4, null, null], // Dayly, but sends data only after request
            [6, null, tuya.valueConverter.phaseVariant2], // voltage and current
            [10, 'fault', tuya.valueConverterBasic.lookup({'clear': 0, 'over_current_threshold': 1,
                'over_power_threshold': 2, 'over_voltage_threshold': 4, 'wrong_frequency_threshold': 8})],
            [11, null, null], // Frozen - strange function, in native app - nothing is clear
            [16, 'state', tuya.valueConverter.onOff],
            [17, null, tuya.valueConverter.threshold], // It's settable, but can't write converter
            [18, 'meter_id', tuya.valueConverter.raw],
            [20, 'clear_fault', tuya.valueConverter.onOff], // Clear fault
            [21, null, null], // Forward Energy T1 - don't know what this
            [22, null, null], // Forward Energy T2 - don't know what this
            [23, null, null], // Forward Energy T3 - don't know what this
            [24, null, null], // Forward Energy T4 - don't know what this
        ],
    },
};

module.exports = definition;
```

Koenkk, 2024-06-15:
Feel free to make a PR with the already working converter

terzo33, 2024-06-15:
@Koenkk Can you help me?

Koenkk, 2024-06-20:
Added!

Proncio, 2024-08-18:
Hi, 
this is my first post here so forgive me for any mistake/bad procedure spelling etc.

I have enhanced coding for TOWSMR1 device, based on @bifi2090 work for device TOQCB2-80

https://github.com/Koenkk/zigbee-herdsman-converters/pull/7752

Everything works, description in the Database can be changed. Devices are pretty the same, from the original code I just changed 

- Refresh time: 10 seconds
- Phases: this is single phase, code adapted
- Adding and formatting "Last event" messages (for some reasons DP name in Expose and Meta sections was different, last_event1 vs last_event, so it was not working)
- Added DP for leakage, but left commented as they should always stay default 30Mah (life risk)

[image]

[image]

If I should do something else to publish this new version, just tell me.

Thanks

Code:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {

fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_kobbcyum',
        },
    ],
    
    model: 'TOWSMR1',
    vendor: 'Tongou',
    description: 'Single-phase multifunction energy meter (DIN Module)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    // Important: respondToMcuVersionResponse should be false otherweise there are an avalanche of commandMcuVersionResponse messages every second.
    // queryIntervalSeconds: is doing a pooling to update device's parameters, now define to update data every 3 minutes.
    onEvent: tuya.onEvent({respondToMcuVersionResponse: false, queryIntervalSeconds: 10}),
    configure: tuya.configureMagicPacket,
    exposes: [
        tuya.exposes.switch(),
        e.energy(),
        e.power(),
        e.voltage(),
        e.current(),
        e.temperature(),
        e
        .enum('last_event', ea.STATE, [])
            .withDescription('Last event'),
        e.enum('over_current_setting', ea.STATE_SET, ['closed', 'alarm', 'trip']).withDescription('Action if threshold value is reached'),
        e
            .numeric('current_threshold', ea.STATE_SET)
            .withValueMin(1)
            .withValueMax(63)
[... 67 lines cut ...]
                    'Remote Off': tuya.enum(12),
                    'Manual On': tuya.enum(13),
                    'Manual Off': tuya.enum(14),
                    'Value 15': tuya.enum(15),
                    'Value 16': tuya.enum(16),
                    'Factory Reset': tuya.enum(17),
                }),
            ],
            [112, 'clear_fault', tuya.valueConverter.onOff],
            [113, 'factory_reset', tuya.valueConverter.onOff],
            [114, 'current_threshold', tuya.valueConverter.raw],
            [115, 'over_voltage_threshold', tuya.valueConverter.raw],
            [116, 'under_voltage_threshold', tuya.valueConverter.raw],
//            [117, 'leakage_threshold', tuya.valueConverter.raw],
            [118, 'temperature_threshold', tuya.valueConverter.divideBy10],
            [119, 'over_power_threshold', tuya.valueConverter.raw],
            //[125, 'forward_electricity', tuya.valueConverter.divideBy100],
            [131, 'temperature', tuya.valueConverter.divideBy10],
        ],
    },
};

module.exports = definition;

```

terzo33, 2024-08-18:
> Hi,
> this is my first post here so forgive me for any mistake/bad procedure spelling etc.
>
> I have enhanced coding for TOWSMR1 device, based on @bifi2090 work for device TOQCB2-80 [...]
 You are the best !!! @Koenkk please upgrade it. Thanks a lot

Proncio, 2024-08-18:
One note: the original coder @true9741 

https://github.com/Koenkk/zigbee2mqtt/issues/21943#issuecomment-2032208831

stated that in his configuration, polling too much congestes his zigbee network, so he used 3 minutes refresh time. I have for now a simple network, so for me it's not an issue, but for some others it could be.

So, a solution could be replacing the fixed number 10 (or 3 * 60 in the original config) with a new virtual DP/setting, representing the # seconds for refresh, and change it like all the other slider settings dinamically in the frontend. 

I would do it but I simply don't konw how, but I guess the experts here do ;) Any hint is more than well accepted

Koenkk, 2024-08-19:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

terzo33, 2024-08-19:
> One note: the original coder @true9741
>
> https://github.com/Koenkk/zigbee2mqtt/issues/21943#issuecomment-2032208831
>
> stated that in his configuration, polling too much congestes his zigbee network, [...]

@Proncio potresti spiegarmi meglio il problema dei 10 secondi ? Questo codice che hai scritto tu ogni quanti secondi aggiorna le entità?

terzo33, 2024-08-19:
> Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

Done :)

New code

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {

fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_kobbcyum',
        },
    ],
    
    model: 'TOWSMR1',
    vendor: 'Tongou',
    description: 'Single-phase multifunction energy meter (DIN Module)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    // Important: respondToMcuVersionResponse should be false otherweise there are an avalanche of commandMcuVersionResponse messages every second.
    // queryIntervalSeconds: is doing a pooling to update device's parameters, now define to update data every 3 minutes.
    onEvent: tuya.onEvent({respondToMcuVersionResponse: false, queryIntervalSeconds: 10}),
    configure: tuya.configureMagicPacket,
    exposes: [
        tuya.exposes.switch(),
        e.energy(),
        e.power(),
        e.voltage(),
        e.current(),
        e.temperature(),
        e.enum('last_event', ea.STATE, [])
            .withDescription('Last event'),
        e.enum('leakage_current', ea.STATE, [])
            .withDescription('Leakage Current'),
        e.enum('forward_electricity', ea.STATE, [])
            .withDescription('Forward Electricity'),
        e.enum('leakage_setting', ea.STATE_SET, ['closed', 'alarm', 'trip']).withDescription('Action if threshold value is reached - USE WITH CAUTION - RISK LIFE !!'),
        e
            .numeric('leakage_threshold', ea.STATE_SET)
            .withValueMin(30)
            .withValueMax(300)
            .withValueStep(1)
            .withUnit('A')
            .withDescription('leakage threshold setting'),
        e.enum('over_current_setting', ea.STATE_SET, ['closed', 'alarm', 'trip']).withDescription('Action if threshold value is reached'),
        e
            .numeric('current_threshold', ea.STATE_SET)
            .withValueMin(1)
            .withValueMax(63)
            .withValueStep(1)
            .withUnit('A')
            .withDescription('Current threshold setting'),
        e.enum('under_voltage_setting', ea.STATE_SET, ['closed', 'alarm', 'trip']).withDescription('Action if threshold value is reached'),
        e
            .numeric('under_voltage_threshold', ea.STATE_SET)
            .withValueMin(145)
            .withValueMax(220)
[... 57 lines cut ...]
                    'Alarm Voltage 2': tuya.enum(10),
                    'Remote On': tuya.enum(11),
                    'Remote Off': tuya.enum(12),
                    'Manual On': tuya.enum(13),
                    'Manual Off': tuya.enum(14),
                    'Value 15': tuya.enum(15),
                    'Value 16': tuya.enum(16),
                    'Factory Reset': tuya.enum(17),
                }),
            ],
            [112, 'clear_fault', tuya.valueConverter.onOff],
            [113, 'factory_reset', tuya.valueConverter.onOff],
            [114, 'current_threshold', tuya.valueConverter.raw],
            [115, 'over_voltage_threshold', tuya.valueConverter.raw],
            [116, 'under_voltage_threshold', tuya.valueConverter.raw],
            [117, 'leakage_threshold', tuya.valueConverter.raw],
            [118, 'temperature_threshold', tuya.valueConverter.divideBy10],
            [119, 'over_power_threshold', tuya.valueConverter.raw],
            [125, 'forward_electricity', tuya.valueConverter.divideBy100],
            [131, 'temperature', tuya.valueConverter.divideBy10],
        ],
    },
};

module.exports = definition;

terzo33, 2024-08-19:
> Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

Done :)

Code with new update. Now all features works fine

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {

fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_kobbcyum',
        },
    ],
    
    model: 'TOWSMR1',
    vendor: 'Tongou',
    description: 'Single-phase multifunction energy meter (DIN Module)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    // Important: respondToMcuVersionResponse should be false otherweise there are an avalanche of commandMcuVersionResponse messages every second.
    // queryIntervalSeconds: is doing a pooling to update device's parameters, now define to update data every 3 minutes.
    onEvent: tuya.onEvent({respondToMcuVersionResponse: false, queryIntervalSeconds: 10}),
    configure: tuya.configureMagicPacket,
    exposes: [
        tuya.exposes.switch(),
        e.energy(),
        e.power(),
        e.voltage(),
        e.current(),
        e.temperature(),
        e.enum('last_event', ea.STATE, [])
            .withDescription('Last event'),
        e.enum('leakage_current', ea.STATE, [])
            .withDescription('Leakage Current'),
        e.enum('forward_electricity', ea.STATE, [])
            .withDescription('Forward Electricity'),
        e.enum('leakage_setting', ea.STATE_SET, ['closed', 'alarm', 'trip']).withDescription('Action if threshold value is reached - USE WITH CAUTION - RISK LIFE !!'),
        e
            .numeric('leakage_threshold', ea.STATE_SET)
            .withValueMin(30)
            .withValueMax(300)
            .withValueStep(1)
            .withUnit('A')
            .withDescription('leakage threshold setting'),
        e.enum('over_current_setting', ea.STATE_SET, ['closed', 'alarm', 'trip']).withDescription('Action if threshold value is reached'),
        e
            .numeric('current_threshold', ea.STATE_SET)
            .withValueMin(1)
            .withValueMax(63)
            .withValueStep(1)
            .withUnit('A')
            .withDescription('Current threshold setting'),
        e.enum('under_voltage_setting', ea.STATE_SET, ['closed', 'alarm', 'trip']).withDescription('Action if threshold value is reached'),
        e
            .numeric('under_voltage_threshold', ea.STATE_SET)
            .withValueMin(145)
[... 59 lines cut ...]
                    'Remote On': tuya.enum(11),
                    'Remote Off': tuya.enum(12),
                    'Manual On': tuya.enum(13),
                    'Manual Off': tuya.enum(14),
                    'Value 15': tuya.enum(15),
                    'Value 16': tuya.enum(16),
                    'Factory Reset': tuya.enum(17),
                }),
            ],
            [112, 'clear_fault', tuya.valueConverter.onOff],
            [113, 'factory_reset', tuya.valueConverter.onOff],
            [114, 'current_threshold', tuya.valueConverter.raw],
            [115, 'over_voltage_threshold', tuya.valueConverter.raw],
            [116, 'under_voltage_threshold', tuya.valueConverter.raw],
            [117, 'leakage_threshold', tuya.valueConverter.raw],
            [118, 'temperature_threshold', tuya.valueConverter.divideBy10],
            [119, 'over_power_threshold', tuya.valueConverter.raw],
            [125, 'forward_electricity', tuya.valueConverter.divideBy100],
            [131, 'temperature', tuya.valueConverter.divideBy10],
        ],
    },
};

module.exports = definition;
```

terzo33, 2024-08-19:
[image]

Hi, I have these errors for leakage settings and leakage threshold. Someone can help me to fox it?

Thanks a lot

Koenkk, 2024-08-20:
Try with the latest [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

terzo33, 2024-08-20:
> Try with the latest [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) 

@Koenkk doesn't work :(

[image]

Koenkk, 2024-08-21:
Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Proncio, 2024-08-21:
Your code in leakage settings section indicates A as unit, it should be mA

terzo33, 2024-08-21:
> Your code in leakage settings section indicates A as unit, it should be mA

Yes

terzo33, 2024-08-21:
> Could you provide the debug log of this?
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

````
[2024-08-22 00:26:29] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-08-22 00:26:29] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-08-22 00:26:29] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-08-22 00:26:29] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-22 00:26:29] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,120,190]
[2024-08-22 00:26:29] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,120,190]
[2024-08-22 00:26:29] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,120] - 190
[2024-08-22 00:26:29] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":120}
[2024-08-22 00:26:29] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-22 00:26:29] debug: 	zh:zstack:unpi:parser: <-- [254,5,69,196,178,251,1,102,120,210]
[2024-08-22 00:26:29] debug: 	zh:zstack:unpi:parser: --- parseNext [254,5,69,196,178,251,1,102,120,210]
[2024-08-22 00:26:29] debug: 	zh:zstack:unpi:parser: --> parsed 5 - 2 - 5 - 196 - [178,251,1,102,120] - 210
[2024-08-22 00:26:29] debug: 	zh:zstack:znp: AREQ: <-- ZDO - srcRtgInd - {"dstaddr":64434,"relaycount":1,"relaylist":[30822]}
[2024-08-22 00:26:29] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-22 00:26:30] debug: 	zh:zstack:unpi:parser: <-- [254,30,68,129,0,0,0,239,178,251,1,1,0,57,0,220,115,14,0,0,10,9,118,2,0,238,102,4,0,1,2,102]
[2024-08-22 00:26:30] debug: 	zh:zstack:unpi:parser: --- parseNext [254,30,68,129,0,0,0,239,178,251,1,1,0,57,0,220,115,14,0,0,10,9,118,2,0,238,102,4,0,1,2,102]
[2024-08-22 00:26:30] debug: 	zh:zstack:unpi:parser: <-- [120,28,31]
[2024-08-22 00:26:30] debug: 	zh:zstack:unpi:parser: --- parseNext [254,30,68,129,0,0,0,239,178,251,1,1,0,57,0,220,115,14,0,0,10,9,118,2,0,238,102,4,0,1,2,102,120,28,31]
[2024-08-22 00:26:30] debug: 	zh:zstack:unpi:parser: --> parsed 30 - 2 - 4 - 129 - [0,0,0,239,178,251,1,1,0,57,0,220,115,14,0,0,10,9,118,2,0,238,102,4,0,1,2,102,120,28] - 31
[2024-08-22 00:26:30] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":64434,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":57,"securityuse":0,"timestamp":947164,"transseqnumber":0,"len":10,"data":{"type":"Buffer","data":[9,118,2,0,238,102,4,0,1,2]}}
[2024-08-22 00:26:30] debug: 	zh:controller: Received payload: clusterID=61184, address=64434, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=57, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":118,"commandIdentifier":2},"payload":{"seq":60928,"dpValues":[{"dp":102,"datatype":4,"data":{"type":"Buffer","data":[2]}}]},"command":{"ID":2,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataReport"}}
[2024-08-22 00:26:30] debug: 	zh:controller:endpoint: ZCL command 0x187a3efffe4b5765/1 manuSpecificTuya.defaultRsp({"cmdId":2,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"transactionSequenceNumber":118,"writeUndiv":false})
[2024-08-22 00:26:30] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x187a3efffe4b5765:64434/1 (0,0,1)
[2024-08-22 00:26:30] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":64434,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":121,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,118,11,2,0]}}
[2024-08-22 00:26:30] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,178,251,1,1,0,239,121,0,30,5,16,118,11,2,0,129]
[2024-08-22 00:26:30] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-22 00:26:30] debug: 	z2m: Received Zigbee message from 'Salvavita Seminterrato - Zigbee', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":102}],"seq":60928}' from endpoint 1 with groupID 0
[2024-08-22 00:26:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Salvavita Seminterrato - Zigbee', payload '{"automatic_closing":"OFF","automatic_reclosing":"OFF","clear_fault":"ON","current":0.87,"current_threshold":20,"energy":1422.96,"factory_reset":"OFF","forward_electricity":1419.7,"last_event":"Normal","leakage_current":8,"leakage_setting":"trip","leakage_threshold":30,"linkquality":57,"over_current_setting":"trip","over_power_setting":"closed","over_power_threshold":3500,"over_voltage_setting":"trip","over_voltage_threshold":270,"power":58,"state":"ON","temperature":34,"temperature_setting":"trip","temperature_threshold":65,"under_voltage_setting":"closed","under_voltage_threshold":180,"voltage":238}'
[2024-08-22 00:26:30] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-08-22 00:26:30] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-08-22 00:26:30] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-08-22 00:26:30] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-08-22 00:26:30] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-22 00:26:30] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,121,191]
[... cut ...]
[2024-08-22 00:26:30] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-22 00:26:30] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,124,186]
[2024-08-22 00:26:30] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,124,186]
[2024-08-22 00:26:30] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,124] - 186
[2024-08-22 00:26:30] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":124}
[2024-08-22 00:26:30] debug: 	zh:zstack:unpi:parser: --- parseNext []
````

Koenkk, 2024-08-22:
I don't see the error from https://github.com/Koenkk/zigbee2mqtt/issues/23054#issuecomment-2299589284 in this log.

terzo33, 2024-08-22:
> I don't see the error from https://github.com/Koenkk/zigbee2mqtt/issues/23054#issuecomment-2299589284 in this log.

what could this error be?

[image]

Koenkk, 2024-08-24:
I need the debug log for that, make sure to capture the logs at the moment this error is logged.

terzo33, 2024-08-27:
> I need the debug log for that, make sure to capture the logs at the moment this error is logged.!

Hi, see log please. Thanks a lot

[log-1.txt](https://github.com/user-attachments/files/16767710/log-1.txt)

Koenkk, 2024-08-28:
I only see the `error` level here, I also need the `debug` one.

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

terzo33, 2024-09-07:
> I only see the `error` level here, I also need the `debug` one.
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Hi @Koenkk , I see you fix it in 1.40.1

"This hotfix release primarily addresses the Cannot read properties of undefined (reading 'some') error (#23821)"

I send you 2 weeks go PR for new code with all features works

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {

fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_kobbcyum',
        },
    ],
    
    model: 'TOWSMR1',
    vendor: 'Tongou',
    description: 'Single-phase multifunction energy meter (DIN Module)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    // Important: respondToMcuVersionResponse should be false otherweise there are an avalanche of commandMcuVersionResponse messages every second.
    // queryIntervalSeconds: is doing a pooling to update device's parameters, now define to update data every 3 minutes.
    onEvent: tuya.onEvent({respondToMcuVersionResponse: false, queryIntervalSeconds: 10}),
    configure: tuya.configureMagicPacket,
    exposes: [
        tuya.exposes.switch(),
        e.energy(),
        e.power(),
        e.voltage(),
        e.current(),
        e.temperature(),
        e.enum('last_event', ea.STATE, [])
            .withDescription('Last event'),
        e.numeric('leakage_current', ea.STATE).withUnit('mA').withDescription('Leakage current is current that flows from an electrical circuit to the ground'),
        e.enum('leakage_setting', ea.STATE_SET, ['closed', 'alarm', 'trip']).withDescription('Action if threshold value is reached - USE WITH CAUTION - RISK LIFE !!'),
        e
            .numeric('leakage_threshold', ea.STATE_SET)
            .withValueMin(30)
            .withValueMax(300)
            .withValueStep(1)
            .withUnit('mA')
            .withDescription('leakage threshold setting - USE WITH CAUTION - RISK LIFE !! (default=30mA)'),
        e.enum('over_current_setting', ea.STATE_SET, ['closed', 'alarm', 'trip']).withDescription('Action if threshold value is reached'),
        e
            .numeric('current_threshold', ea.STATE_SET)
            .withValueMin(1)
            .withValueMax(40)
            .withValueStep(1)
            .withUnit('A')
            .withDescription('Current threshold setting'),
        e.enum('under_voltage_setting', ea.STATE_SET, ['closed', 'alarm', 'trip']).withDescription('Action if threshold value is reached'),
        e
            .numeric('under_voltage_threshold', ea.STATE_SET)
[... 59 lines cut ...]
                    'Alarm Voltage 2': tuya.enum(10),
                    'Remote On': tuya.enum(11),
                    'Remote Off': tuya.enum(12),
                    'Manual On': tuya.enum(13),
                    'Manual Off': tuya.enum(14),
                    'Value 15': tuya.enum(15),
                    'Value 16': tuya.enum(16),
                    'Factory Reset': tuya.enum(17),
                }),
            ],
            [112, 'clear_fault', tuya.valueConverter.onOff],
            [113, 'factory_reset', tuya.valueConverter.raw],
            [114, 'current_threshold', tuya.valueConverter.raw],
            [115, 'over_voltage_threshold', tuya.valueConverter.raw],
            [116, 'under_voltage_threshold', tuya.valueConverter.raw],
            [117, 'leakage_threshold', tuya.valueConverter.raw],
            [118, 'temperature_threshold', tuya.valueConverter.divideBy10],
            [119, 'over_power_threshold', tuya.valueConverter.raw],
            [131, 'temperature', tuya.valueConverter.divideBy10],
        ],
    },
};

module.exports = definition;
```

Koenkk, 2024-09-08:
> I send you 2 weeks ago PR for new code with all features works but I can't see new code in 1.40.1

I think I missed it, could you provide a link to it?

terzo33, 2024-09-08:
> > I send you 2 weeks ago PR for new code with all features works but I can't see new code in 1.40.1
> 
> I think I missed it, could you provide a link to it?

Yes @Koenkk 

https://github.com/Koenkk/zigbee-herdsman-converters/compare/master...terzo33:zigbee-herdsman-converters:patch-8

**I'm using zigbee2mqtt dev branch for "leakage_current" parameter. Can you fix this feature in normal zigbee2mqtt ?**

ars9, 2024-09-14:
@Koenkk @terzo33 Thank you for working on this. Can't wait for all of it to land up in a release build!

terzo33, 2024-09-14:
> @Koenkk @terzo33 Thank you for working on this. Can't wait for all of it to land up in a release build!

You're Welcome. I think will be ready in next release 💪

Koenkk, 2024-09-15:
`_TZE204_kobbcyum` will be supported in the next release, thanks all!

closed 2024-09-15 by Koenkk

terzo33, 2024-11-09:
> `_TZE204_kobbcyum` will be supported in the next release, thanks all!

Hi @Koenkk , towsmr1 integration has not been updated, everything is as before.

I sent you the PR for the update some time ago

Thanks a lot

Koenkk, 2024-11-10:
> I sent you the PR for the update some time ago

Can you provide a link to it?

terzo33, 2024-11-10:
> > I sent you the PR for the update some time ago
> 
> Can you provide a link to it?

https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {

fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_kobbcyum',
        },
    ],
    
    model: 'TOWSMR1',
    vendor: 'Tongou',
    description: 'Single-phase multifunction energy meter (DIN Module)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    // Important: respondToMcuVersionResponse should be false otherweise there are an avalanche of commandMcuVersionResponse messages every second.
    // queryIntervalSeconds: is doing a pooling to update device's parameters, now define to update data every 3 minutes.
    onEvent: tuya.onEvent({respondToMcuVersionResponse: false, queryIntervalSeconds: 10}),
    configure: tuya.configureMagicPacket,
    exposes: [
        tuya.exposes.switch(),
        e.energy(),
        e.power(),
        e.voltage(),
        e.current(),
        e.temperature(),
        e.enum('last_event', ea.STATE, [])
            .withDescription('Last event'),
        e.numeric('leakage_current', ea.STATE).withUnit('mA').withDescription('Leakage current is current that flows from an electrical circuit to the ground'),
        e.enum('leakage_setting', ea.STATE_SET, ['closed', 'alarm', 'trip']).withDescription('Action if threshold value is reached - USE WITH CAUTION - RISK LIFE !!'),
        e
            .numeric('leakage_threshold', ea.STATE_SET)
            .withValueMin(30)
            .withValueMax(300)
            .withValueStep(1)
            .withUnit('mA')
            .withDescription('leakage threshold setting - USE WITH CAUTION - RISK LIFE !! (default=30mA)'),
        e.enum('over_current_setting', ea.STATE_SET, ['closed', 'alarm', 'trip']).withDescription('Action if threshold value is reached'),
        e
            .numeric('current_threshold', ea.STATE_SET)
            .withValueMin(1)
            .withValueMax(40)
            .withValueStep(1)
            .withUnit('A')
            .withDescription('Current threshold setting'),
        e.enum('under_voltage_setting', ea.STATE_SET, ['closed', 'alarm', 'trip']).withDescription('Action if threshold value is reached'),
        e
            .numeric('under_voltage_threshold', ea.STATE_SET)
            .withValueMin(145)
            .withValueMax(220)
            .withValueStep(1)
            .withUnit('V')
            .withDescription('Under voltage threshold setting'),
[... 53 lines cut ...]
                    'Alarm Voltage 1': tuya.enum(9),
                    'Alarm Voltage 2': tuya.enum(10),
                    'Remote On': tuya.enum(11),
                    'Remote Off': tuya.enum(12),
                    'Manual On': tuya.enum(13),
                    'Manual Off': tuya.enum(14),
                    'Value 15': tuya.enum(15),
                    'Value 16': tuya.enum(16),
                    'Factory Reset': tuya.enum(17),
                }),
            ],
            [112, 'clear_fault', tuya.valueConverter.onOff],
            [113, 'factory_reset', tuya.valueConverter.raw],
            [114, 'current_threshold', tuya.valueConverter.raw],
            [115, 'over_voltage_threshold', tuya.valueConverter.raw],
            [116, 'under_voltage_threshold', tuya.valueConverter.raw],
            [117, 'leakage_threshold', tuya.valueConverter.raw],
            [118, 'temperature_threshold', tuya.valueConverter.divideBy10],
            [119, 'over_power_threshold', tuya.valueConverter.raw],
            [131, 'temperature', tuya.valueConverter.divideBy10],
        ],
    },
};

module.exports = definition;`

terzo33, 2024-11-10:
> Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

here is the PR, can you please check if it's ok?

https://github.com/Koenkk/zigbee-herdsman-converters/compare/master...terzo33:zigbee-herdsman-converters:patch-10

I add _TZE204_kobbcyum and _TZE284_kobbcyum

Koenkk, 2024-11-10:
I see a lot of unexpected changes under https://github.com/Koenkk/zigbee-herdsman-converters/compare/master...terzo33:zigbee-herdsman-converters:patch-10

terzo33, 2024-11-10:
> I see a lot of unexpected changes under https://github.com/Koenkk/zigbee-herdsman-converters/compare/master...terzo33:zigbee-herdsman-converters:patch-10

patch 10 is the latest and it is the one that works correctly. can you include it in the next release?

Koenkk, 2024-11-11:
Thanks, can you see if it works with z2m 1.41.0 with the following:

- Stop Z2M
- Open `data/database.db` with a text editor and change all occurrences of `_TZE284_kobbcyum ` to `_TZE204_q22avxbv`
- Start Z2M and see if the device now works

terzo33, 2024-11-11:
> Thanks, can you see if it works with z2m 1.41.0 with the following:
> 
> - Stop Z2M
> - Open `data/database.db` with a text editor and change all occurrences of `_TZE284_kobbcyum ` to `_TZE204_q22avxbv`
> - Start Z2M and see if the device now works
> 

In zigbee2mqtt folder?

terzo33, 2024-11-13:
> Thanks, can you see if it works with z2m 1.41.0 with the following:
> 
> - Stop Z2M
> - Open `data/database.db` with a text editor and change all occurrences of `_TZE284_kobbcyum ` to `_TZE204_q22avxbv`
> - Start Z2M and see if the device now works
> 

@Koenkk where can I find "data/database.db" ?

Koenkk, 2024-11-13:
Ah sorry, it's `zigbee2mqtt/database.db` indeed.

dimacoupe, 2024-11-24:
i have _TZE284_kobbcyum . used more than month with converter and can confirm - works 👌

terzo33, 2024-11-24:
@Koenkk please add new fixes to next release. Thanks a lot

dimacoupe, 2024-12-02:
z2m 1.42.0
_TZE284_kobbcyum unsupported 
[image]

terzo33, 2024-12-02:
> z2m 1.42.0
> _TZE284_kobbcyum unsupported 
> [image]
> 

I have sent the correct code(PR) to @Koenkk for kobbcyum_204(fix) and kobbcyum_284(new device) many times but I don't understand why @Koenkk keeps ignoring it. I've been waiting for 3 releases...

dimacoupe, 2024-12-02:
> I have sent the correct code(PR) to @Koenkk for kobbcyum_204(fix) and kobbcyum_284(new device) many times but I don't understand why @Koenkk keeps ignoring it. I've been waiting for 3 releases...

ok, I continue to use your custom converter

Koenkk, 2024-12-02:
Added for the 1 January release!

terzo33, 2024-12-02:
> Added for the 1 January release!

Thanks a lot 🙏🏻

Proncio, 2025-01-01:
> One note: the original coder @true9741
> 
> [#21943 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/21943#issuecomment-2032208831)
> 
> stated that in his configuration, polling too much congestes his zigbee network, so he used 3 minutes refresh time. I have for now a simple network, so for me it's not an issue, but for some others it could be.
> 
> So, a solution could be replacing the fixed number 10 (or 3 * 60 in the original config) with a new virtual DP/setting, representing the # seconds for refresh, and change it like all the other slider settings dinamically in the frontend.
> 
> I would do it but I simply don't konw how, but I guess the experts here do ;) Any hint is more than well accepted

Some months have passed, and I now have 8 of these devices. As already suspected, this polling is heavy: 

[image]

Every 10 seconds, CPU of the whole server raises from 10 to 45%, following the polling frequency declared in the code, in long term this is an issue. Solutions? 

1. avoid polling, but I can't find how
2. lower the frequency in code, but it's not flexible
3. adding a refresh rate parameter would be fine for most cases
4. ideas?

Does anyone know how to do solution 3 / send an example of a similar logic in other devices?