Koenkk/zigbee2mqtt #23055: Issue with complete pairing with BTicino K4027C

CodeForLove83, 2024-06-14:
### What happened?

The device cannot be successfully paired and configured _without_ double reset of the device.

Specifically, once the device is in pairing mode (green led blinking), Z2M confirms pairing and completion with success of the configuration. However, the device, K4027C, keeps blinking green and does emit any 'click' sound. To complete successfully the configuration, which means fixed green led (not blinking), you need to reset the device once more time (keep pressing the reset button, without deleting the device from Z2M, wait till the red light is fixed and then quick push once more the reset button to set the device with green light flashing. Z2M must allow devices to join (Permit Join(All)). Z2M will complete the configuration and the green light on the device stop flashing. 

Attached you can find an excerpt of the debug log. 

### What did you expect to happen?

Pair and configure the device without double resetting the device.

Expected process:
1. Reset the device, long press reset button -> fixed red light
2. Set the device in pairing mode, quick press the reset button -> green light flashing
3. Enable `Permit Join (All)` on Z2M
4. Wait till the device is successfully configured
5. Green light fixed on the device

### How to reproduce it (minimal and precise)

1. Reset the device, long press reset button -> fixed red light
2. Set the device in pairing mode, quick press the reset button -> green light flashing
3. Enable `Permit Join (All)` on Z2M
4. Wait till Z2M confirms that is successfully configured
5. Green light keeps flashing
6. Reset the device, long press reset button -> fixed red light
7. Set the device in pairing mode, quick press the reset button -> green light flashing
8. Wait till the device stop flashing the green light

### Zigbee2MQTT version

Zigbee2MQTT 1.38.0-1

### Adapter firmware version

0x26510900

### Adapter

ConBee3

### Setup

Z2A running on HA (Core 2024.6.2, Supervisor 2024.06.0)

### Debug log

[log.log](https://github.com/user-attachments/files/15840782/log.log)

jafar-atili, 2024-09-10:
I have the same problem now.
I suspect it has to do with their recent firmware.

Even after pairing them they continue to timeout and do not report state correctly

Crystian182, 2024-12-15:
I have the same error on 1.41.0-1

Crystian182, 2024-12-16:
i fixed it by upgrading the device to fw 0041 (20240722).
i tried multiple times before commenting here, but the progress kept getting stuck, so I thought the issue was related to the incomplete pairing process. However, after several attempts, I managed to update it, and the status LED is now turned off.

closed 2025-02-22 by github-actions[bot]