Koenkk/zigbee2mqtt #23056: Aqara ZNJLBL01LM - position shows closed when it's actually open

woutf, 2024-06-14:
### What happened?

I have found multiple issues around the Aqara E1 Roller Shade Driver, but not my specific one. Whenever I open the blinds through Home Assistant, the state gets updated to open and the position is set at 100%. Some time later, the position will reset to 0%, even though the blind hasn't moved. Because of this, I can't open the blind (button is greyed out)

If I then press 'Close' in Z2M, the position changes to 100% and starts closing. So to 'reset' the value when the blind is open but shows closed, I have to
-Press close
-Quickly press stop
-Press open again

[image]
In this screenshot, you can see the state gets changed to closed, then the device becomes unknown and then it's back at closed again. 

To be clear: the device was opened at 08:36 and not touched after that.

### What did you expect to happen?

I expect the state to reflect the actual state

### How to reproduce it (minimal and precise)

Open the shade 
Wait a while
Notice the reported state doesn't correspond to the physical state

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20230507

### Adapter

zzh!

### Setup

Add-on on Home Assistant Blue

woutf, 2024-06-19:
Things I've done so far, unfortunately without any effect

- Switched the network from channel 11 to 25
- Updated the coördinator firmware
- Forced the device to join through another router

The last thing I can think of is to swap out the zzh! for a SkyConnect, but that's just a shot in the dark.

woutf, 2024-06-19:
I managed to find something interesting: every time this happens, I can see 'device_announce' in the logs
`[2024-06-19 16:16:10] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Blinds ', payload '{"battery":85,"charging_status":false,"device_temperature":22,"linkquality":54,"motor_speed":"medium","motor_state":"stopped","position":100,"power_outage_count":4,"running":false,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
[2024-06-19 16:20:53] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Bedroom Blinds ","ieee_address":"0x54ef4410009934d8"},"type":"device_announce"}'
[2024-06-19 16:20:53] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Bedroom Blinds "},"type":"device_announced"}'
[2024-06-19 16:20:54] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Blinds ', payload '{"battery":85,"charging_status":false,"device_temperature":22,"linkquality":54,"motor_speed":"medium","motor_state":"stopped","position":100,"power_outage_count":4,"running":false,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
[2024-06-19 16:20:55] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Bedroom Blinds ","ieee_address":"0x54ef4410009934d8"},"type":"device_announce"}'
[2024-06-19 16:20:55] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Bedroom Blinds "},"type":"device_announced"}'
[2024-06-19 16:20:57] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Blinds ', payload '{"battery":85,"charging_status":false,"device_temperature":22,"linkquality":54,"motor_speed":"medium","motor_state":"stopped","position":0,"power_outage_count":4,"running":false,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
[2024-06-19 16:21:02] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Blinds ', payload '{"battery":85,"charging_status":false,"device_temperature":22,"linkquality":54,"motor_speed":"medium","motor_state":"stopped","position":0,"power_outage_count":5,"running":false,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'`

woutf, 2024-06-29:
I managed to exchange the device for a new one, but unfortunately this new one has the same issue

Koenkk, 2024-06-29:
> and the position is set at 100%. Some time later, the position will reset to 0%,

Could you provide the debug log when this happens?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

woutf, 2024-07-01:
> > and the position is set at 100%. Some time later, the position will reset to 0%,
> 
> Could you provide the debug log when this happens?
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

I'm running into a problem with log rotation. The problem occurs very intermittently, every few hours. This means I can't catch it when it happens and need to look into the logs after the fact. Because the logs fill up quite quickly at the debug log level, the relevant log information is already gone when I go and check. 

Is there any way to just enable debug information for one device? Or increase the amount of log files? Alternatively, I could set-up a syslog server, but I have no experience doing so. Using the log files would be easier.

Koenkk, 2024-07-02:
Probably the device crashes, can you try to disable availability for the cover and see if that fixes the issue?

woutf, 2024-07-03:
Availability isn't enabled for any device, including this one.

Koenkk, 2024-07-03:
Can you provide me your `configuration.yaml` file?

woutf, 2024-07-04:
> Can you provide me your `configuration.yaml` file?

```
data_path: /share/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/dev/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
  restartdelay: 1
  initialdelay: 1
  zigbee_herdsman_debug: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto
  user: mqtt
  password: ######
serial:
  port: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
advanced:
  log_level: debug
  log_syslog:
    host: 192.168.1.121
    port: 514
    protocol: udp4
    path: /dev/log
    pid: process.pid
    facility: local0
    localhost: localhost
    type: "5424"
    app_name: Zigbee2MQTT
    eol: \n
  pan_id: 6755
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental:
  new_api: true

```

and from devices.yaml
```
0x54ef441000844955':
  friendly_name: Bedroom Blinds
  availability: false
```

[image]

Koenkk, 2024-07-04:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/7abcbcf756b31e08402c1a39f8b64b25): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Also, could you provide the debug log when moving the curtain to e.g. 80%?

woutf, 2024-07-05:
Have it up and running, will check if this fixes things. Setting a position seems to be broken though
```
[2024-07-05 10:33:08] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Bedroom Blinds/set' with data '{ "position": 80 }'
[2024-07-05 10:33:08] error: 	z2m: No converter available for 'position' (80)
```

woutf, 2024-07-05:
The device has been stable for the past 9 hours, which is a record. It's also useless at the moment :D Because not only is there no converter available for 'position', the same goes for 'open' and 'close'.

Koenkk, 2024-07-06:
updated https://gist.github.com/Koenkk/7abcbcf756b31e08402c1a39f8b64b25, can you check again?

woutf, 2024-07-06:
`z2m: Publish 'set' 'state' to 'Bedroom Blinds' failed: 'ReferenceError: isString is not defined'`
`z2m: Publish 'set' 'position' to 'Bedroom Blinds' failed: 'ReferenceError: isString is not defined'`

Koenkk, 2024-07-07:
Updated https://gist.github.com/Koenkk/7abcbcf756b31e08402c1a39f8b64b25 again.

woutf, 2024-07-07:
Succes! Opening, closing and setting position works now. Now we wait and see if the position changes by itself...

woutf, 2024-07-09:
The curtain has been stable for the past two days, seems fixed. Thanks!

Koenkk, 2024-07-10:
Can you check if the device returns the correct position when reading it through the z2m frontend -> device -> dev console, read cluster `closuresWindowCovering` attribute `currentPositionLiftPercentage`?

woutf, 2024-07-10:
> Can you check if the device returns the correct position when reading it through the z2m frontend -> device -> dev console, read cluster `closuresWindowCovering` attribute `currentPositionLiftPercentage`?

Those two don't show up in the dropdown box with clusters to read. 
[image]

Koenkk, 2024-07-10:
Ah it's called `Closures` here.

woutf, 2024-07-10:
It returns 0, even though the cover is positioned halfway. Somehow Home Assistant does show the correct position

Koenkk, 2024-07-11:
Thanks, integrated the fix.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-07-11 by Koenkk

Cameleon99, 2024-08-17:
@Koenkk this appears to have broken my ZNJLBL01LM, now doesn't ever report update to position. See https://github.com/Koenkk/zigbee2mqtt/issues/23557

lizashcherbakova, 2024-11-28:
Problem is still there.  Zigbee2MQTT version 1.41.0

lizashcherbakova, 2024-11-28:
The problem is still there. Zigbee2MQTT version is 1.41.0

Marrrs, 2024-12-07:
Try resetting and setting the Min/Max positions using the hardware buttons as in the "Device Rotating Range Setup" part of the user manual. I found that the motor won't report its position if the positions are not set on the hardware first.  https://www.aqara.com/en/product/roller-shade-driver-e1/user-manual/

isibizi, 2024-12-08:
I‘m on 1.42, i have a tuya curtain. The state in z2m shows me closed but HA shows me open

{
    "backlight_mode": "ON",
    "calibration": "OFF",
    "calibration_time": 27,
    "indicator_mode": "off",
    "linkquality": 160,
    "motor_reversal": "OFF",
    "moving": "STOP",
    "position": 0,
    "state": "CLOSE"
}