Koenkk/zigbee2mqtt #23057: EcoDim throwing timeout errors

robinwo, 2024-06-14:
### What happened?

I'm using this dimmer: https://www.ecodim.nl/nl/eco-dim05-zigbee.html
Added to zigbee2mqtt, shows it's supported.

Now in the logs it keeps throwing timeout errors. I'm not seeing this for any other devices (red messages keeps popping in the lower right bottom of the browser). Changing state & brightness sometimes does, but most times does not work (leading up to this timeout). Anything I could do about this?

```
Error 2024-06-14 18:40:27Publish 'set' 'state' to 'EcoDim - DuoDimmer' failed: 'Error: ZCL command 0x84b4dbfffe089f3a/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 7135 - 1 - 228 - 6 - 11 after 10000ms)'
Error 2024-06-14 18:40:47Publish 'set' 'state' to 'EcoDim - DuoDimmer' failed: 'Error: ZCL command 0x84b4dbfffe089f3a/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 7135 - 2 - 229 - 6 - 11 after 10000ms)'
```

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

### Setup

HomeAssistant on Synology VM

Nerivec, 2024-06-22:
Did you try pairing the device to a specific router (or directly to the coordinator), see if it improves the situation?

xsienix, 2024-07-16:
Hi,

You should set the transition time to 1.

closed 2024-12-29 by github-actions[bot]