Koenkk/zigbee2mqtt #23059: TS0601 (_TZE200_whpb9yts) stopped responding to brightness requests

elgerg, 2024-06-14:
### What happened?

I upgraded to 1.38.0 from ~1.36 (before the automatic identification of capabilities was introduced) and now my dimmers dont work correctly. Sometimes the state ends up being the brightness so randomly wont turn off.

However the main issue is that I cant adjust the brightness anymore. only turn them on and off.

Even using the web UI (sending the right MQTT payload) does nothing.

EG sending the following payloads:
Info 2024-06-14 20:43:09MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":181,"level":319,"linkquality":222,"max_brightness":254,"min_brightness":3,"state":"ON"}'
Info 2024-06-14 20:43:15MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":92,"level":319,"linkquality":222,"max_brightness":254,"min_brightness":3,"state":"ON"}'
Info 2024-06-14 20:43:17MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":57,"level":319,"linkquality":222,"max_brightness":254,"min_brightness":3,"state":"ON"}'
Info 2024-06-14 20:43:27MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":161,"level":319,"linkquality":222,"max_brightness":254,"min_brightness":3,"state":"ON"}'

didnt change the brightness in any way.

What happened or what am I doing wrong now?

Thanks

### What did you expect to happen?

I expected the brightness to go up or down depending on the value supplied.

### How to reproduce it (minimal and precise)

Info 2024-06-14 20:43:09MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":181,"level":319,"linkquality":222,"max_brightness":254,"min_brightness":3,"state":"ON"}'
Info 2024-06-14 20:43:15MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":92,"level":319,"linkquality":222,"max_brightness":254,"min_brightness":3,"state":"ON"}'
Info 2024-06-14 20:43:17MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":57,"level":319,"linkquality":222,"max_brightness":254,"min_brightness":3,"state":"ON"}'
Info 2024-06-14 20:43:27MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":161,"level":319,"linkquality":222,"max_brightness":254,"min_brightness":3,"state":"ON"}'

### Zigbee2MQTT version

 1.38.0 commit: fe048e6e

### Adapter firmware version

20220219

### Adapter

slae.sh cc2652rb stick - slaesh's iot stuff

### Setup

NodeRed with Mosquitto

elgerg, 2024-06-14:
Oh it gets worse. When I change the values by using the physical button it doesnt report back :(

Actually it does but as the min brightness value..

Info 2024-06-14 20:50:09MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":207,"max_brightness":254,"min_brightness":33,"state":"ON"}'
Info 2024-06-14 20:50:09MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":204,"max_brightness":254,"min_brightness":37,"state":"ON"}'
Info 2024-06-14 20:50:09MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":207,"max_brightness":254,"min_brightness":41,"state":"ON"}'
Info 2024-06-14 20:50:10MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":207,"max_brightness":254,"min_brightness":46,"state":"ON"}'
Info 2024-06-14 20:50:10MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":207,"max_brightness":254,"min_brightness":50,"state":"ON"}'
Info 2024-06-14 20:50:10MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":207,"max_brightness":254,"min_brightness":54,"state":"ON"}'
Info 2024-06-14 20:50:10MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":207,"max_brightness":254,"min_brightness":57,"state":"ON"}'
Info 2024-06-14 20:50:10MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":207,"max_brightness":254,"min_brightness":61,"state":"ON"}'
Info 2024-06-14 20:50:10MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":207,"max_brightness":254,"min_brightness":65,"state":"ON"}'
Info 2024-06-14 20:50:10MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":204,"max_brightness":254,"min_brightness":69,"state":"ON"}'
Info 2024-06-14 20:50:10MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":204,"max_brightness":254,"min_brightness":72,"state":"ON"}'
Info 2024-06-14 20:50:10MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":207,"max_brightness":254,"min_brightness":77,"state":"ON"}'
Info 2024-06-14 20:50:10MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":204,"max_brightness":254,"min_brightness":81,"state":"ON"}'
Info 2024-06-14 20:50:11MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":204,"max_brightness":254,"min_brightness":85,"state":"ON"}'
Info 2024-06-14 20:50:11MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":204,"max_brightness":254,"min_brightness":88,"state":"ON"}'
Info 2024-06-14 20:50:11MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":211,"max_brightness":254,"min_brightness":90,"state":"ON"}'
Info 2024-06-14 20:50:28MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":189,"max_brightness":254,"min_brightness":90,"state":"ON"}'
Info 2024-06-14 20:50:28MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":189,"max_brightness":254,"min_brightness":86,"state":"ON"}'
Info 2024-06-14 20:50:28MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":193,"max_brightness":254,"min_brightness":82,"state":"ON"}'
Info 2024-06-14 20:50:28MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":189,"max_brightness":254,"min_brightness":78,"state":"ON"}'
Info 2024-06-14 20:50:28MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":204,"max_brightness":254,"min_brightness":76,"state":"ON"}'
Info 2024-06-14 20:50:38MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":182,"max_brightness":254,"min_brightness":76,"state":"ON"}'
Info 2024-06-14 20:50:38MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":182,"max_brightness":254,"min_brightness":80,"state":"ON"}'
Info 2024-06-14 20:50:38MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":182,"max_brightness":254,"min_brightness":84,"state":"ON"}'
Info 2024-06-14 20:50:38MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":182,"max_brightness":254,"min_brightness":87,"state":"ON"}'
Info 2024-06-14 20:50:38MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":182,"max_brightness":254,"min_brightness":92,"state":"ON"}'
Info 2024-06-14 20:50:38MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":182,"max_brightness":254,"min_brightness":96,"state":"ON"}'
Info 2024-06-14 20:50:38MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":182,"max_brightness":254,"min_brightness":100,"state":"ON"}'
Info 2024-06-14 20:50:39MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":193,"max_brightness":254,"min_brightness":103,"state":"ON"}'
Info 2024-06-14 20:50:39MQTT publish: topic 'zigbee2mqtt/LoungeMainDimmer', payload '{"brightness":86,"level":319,"linkquality":222,"max_brightness":254,"min_brightness":105,"state":"ON"}'

elgerg, 2024-06-14:
This is getting even more bizarre.

If I set the min brightness slider in the front end then the lights change as expected. However the brightness slider does not move.

Has brightness been replaced with min brightness?? If so why is brightness still there and no release notes that state this has been deprecated and why is the brightness slider still there?

Thanks

elgerg, 2024-06-15:
Having reverted to version 1.36.0 and then updating one version at a time I can confirm that the brightness slider and min_brightness start to get messed up from version 1.7.0 onwards.

This seems to be the case for the dimmer and a GL-C-008P

elgerg, 2024-06-18:
@Koenkk 

I've just set up a test lab using the latest version for testing purposes:
[image]

When changing the brightness from the light switch only the min_brightness value is passed (it should be brightness).

```
{
    "linkquality": 94,
    "state": "ON",
    "min_brightness": 254
}
{
    "linkquality": 94,
    "state": "ON",
    "min_brightness": 240
}
{
    "linkquality": 94,
    "state": "ON",
    "min_brightness": 220
}
```

Now sure where to go with this as I cant upgrade as most of my home automation for my lights will stop working properly.

Thanks in advance

Koenkk, 2024-06-22:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/0213908b3d1e8318e7b01fef1d738370): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

elgerg, 2024-06-23:
Hi @Koenkk 

Yep, that external converter has fixed the problem in my lab.

What is the next step?

Thanks!

Koenkk, 2024-06-24:
With 1.36, was the min brightness working?

Koenkk, 2024-06-25:
Great, integrated the fix!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-06-25 by Koenkk