Koenkk/zigbee2mqtt #23074: Philips Disney StoryLight (LLC013) stops responding after 20 minutes

DJBenson, 2024-06-16:
### What happened?

I've just switched all my devices over to a new coordinator (from an Electrolama zzh to a Smartlight SLZN-06M using the Ember driver) and a device which was causing me issues on my SkyConnect before I switched to the zzh is now causing problems again.

The device is a Philips Hue "Disney StoryLight" - baiscally a rebadged bloom. The issue appears to be the device stops responding to Z2M after approximately 20 minutes and then goes unresponsive. Restarting Z2M revives it but it goes offline again after the same period of time.

### What did you expect to happen?

The device to stay online!

### How to reproduce it (minimal and precise)

Join device
Wait for approximately 20 minutes
Send a command (refresh power status for example)
Device is unresponsive
Restart Z2M
Device is responsive.

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20240510

### Adapter

SLZB-06M

### Setup

HAOS, Proxmox, Lenovo Host.

### Debug log

[log.log](https://github.com/user-attachments/files/15855720/log.log)

DJBenson, 2024-06-16:
I have noticed another bulb in the network behaving the same. Both recover when Z2M is restarted.

Nerivec, 2024-06-22:
Are these devices paired directly to the coordinator?
Can you provide a `debug` log file from the start of Z2M, until after it stops working, so I can see what it is doing in that timeframe that might explain this?

closed 2024-12-29 by github-actions[bot]