Koenkk/zigbee2mqtt #23079: [New device support]: TRV ZIGBEE TS0601 / TZE200_ytryxh0a

DERBESSE, 2024-06-16:
### Link

https://www.xanlite.com/fr/regulation-denergie/1779-product-3700619438993.html

### Database entry

{"id":8,"type":"EndDevice","ieeeAddr":"0xa4c138c3459a0a60","nwkAddr":21209,"manufId":4417,"manufName":"_TZE200_ytryxh0a","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1718565228534}

### Comments

Hi,
I purchased this device but could not register it in HA. 
it is showing up on Z2M but is showing as Unsupported.
Thanks for your help

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_ytryxh0a',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

Mouni68, 2024-07-10:
Hello I have the same reference here the files I made

External definition
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee.js');
const tz = require('zigbee-herdsman-converters/converters/toZigbee.js');
const exposes = require('zigbee-herdsman-converters/lib/exposes.js');
const reporting = require('zigbee-herdsman-converters/lib/reporting.js');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend.js');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya.js');

const definition = {

    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_ytryxh0a',
        },
    ],
    model: 'TS0601_thermostat',
    vendor: 'Tuya',
    description: 'Radiator valve with thermostat',
	whiteLabel: [
            { vendor: 'Xanlite', model: 'KZTETHEAU' },
        ],
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
            e
                .climate()
                .withPreset(['auto', 'manual', 'holiday',])
		.withLocalTemperature(ea.STATE)
		.withSetpoint('current_heating_setpoint', 5, 30, 0.5, ea.STATE_SET)
                .withLocalTemperatureCalibration(-9, 9, 0.1, ea.STATE_SET),
            e.comfort_temperature().withValueMin(5).withValueMax(30),
            e.eco_temperature().withValueMin(5).withValueMax(30),
            e.holiday_temperature().withValueMin(0).withValueMax(30),
      	    e.window_detection(),
	    e.open_window_temperature().withValueMin(5).withValueMax(25),
	    e.binary('boost_heating', ea.STATE_SET, 'ON', 'OFF'),
             ...tuya.exposes.scheduleAllDays(ea.STATE, 'HH:MM/C'),
	    e.child_lock(),
	    e.battery().withUnit('%'),
	    tuya.exposes.errorStatus(),
    ],
    meta: {
        tuyaDatapoints: [
	[2, 'preset', tuya.valueConverterBasic.lookup({ auto: tuya.enum(0), manual: tuya.enum(1), holiday: tuya.enum(2), confort: tuya.enum(3) }), ],
	[16, 'current_heating_setpoint', tuya.valueConverter.divideBy2],
	[34, 'battery', tuya.valueConverter.batteryVoltToPercent],
        [30, 'child_lock', tuya.valueConverter.lockUnlock],
	[24, 'local_temperature', tuya.valueConverter.divideBy10],
	[45, 'error_status', tuya.valueConverter.raw], 
        [101, 'comfort_temperature', tuya.valueConverter.divideBy2],
        [102, 'eco_temperature', tuya.valueConverter.divideBy2],
	[103, 'holiday_start_stop', tuya.valueConverter.thermostatHolidayStartStop],
	[104, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration1],
	[105, 'holiday_temperature', tuya.valueConverter.divideBy2],
	[106, 'boost_heating', tuya.valueConverter.onOff],
	[107, 'window_detection', tuya.valueConverter.onOff],
	[109, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDP2],
        [110, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDP2],
        [111, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDP2],
        [112, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDP2],
        [113, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDP2],
        [114, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDP2],
[... 57 lines cut ...]
        to: (v) => {
           // const payload = [0];
           // const transitions = v.split(' ');
           //if (transitions.length != 5) {
           //     throw new Error('Invalid schedule: there should be 5 transitions');
           //}
           //for (const transition of transitions) {
           //     const timeTemp = transition.split('/');
           //     if (timeTemp.length != 2) {
           //         throw new Error('Invalid schedule: wrong transition format: ' + transition);
           //     }
           //     const hourMin = timeTemp[0].split(':');
           //     const hour = parseInt(hourMin[0]);
           //     const min = parseInt(hourMin[1]);
           //     const temperature = Math.floor(parseFloat(timeTemp[1]) * 10);
           //     if (hour < 0 || hour > 24 || min < 0 || min > 60 || temperature < 50 || temperature > 300) {
           //         throw new Error('Invalid hour, minute or temperature of: ' + transition);
           //     }
			//					
           //     payload.push(hour, min, (temperature & 0xff00) >> 8, temperature & 0xff);
           // }
           // return payload;
        },
    },
```

DERBESSE, 2024-07-10:
Hello Mouni68
Thanks you !
Now, the devices is supported.
But, please how to modify 'zigbee-herdsman-converters/lib/tuya.js' with new réferences  ?
Thanks for your help

Mouni68, 2024-07-11:
I'm using Zigbee2mqtt in a docker (alpine) on a Unraid server for me, I have acces to the app/data directory so I copy the tuya.js inside the data folder and execute in command ligne `cp node_modules/zigbee-herdsman-converters/devices/tuya.js data/tuya.js` and the file is accessible to modify after I reverse the process with `cp data/tuya.js node_modules/zigbee-herdsman-converters/lib/tuya.js`

But is just for testing, I don't know how to add this to the next update, if someone can do it?

Koenkk, 2024-07-11:
Could you make a pull request to add out-of-the-box support? This can be done by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)

aml57, 2024-07-17:
Hello Mouni68
Thanks you !
Now, my devices are supported too, 
But i have a trouble with the `` 'current_heating_setpoint `` i cannot set a manual temperatur , only the presset mod working, 
And off my m2qtt drawing i see the thermostat but it`s lonely ( for the signal ) ,
have you any idee maybe ?

Mouni68, 2024-07-18:
Hello,
since 5 day I'm trying to make it work, I bought 2 smart socket to cover more area. I have seen that the thermostat are very bad, low signal will broke all the thing you will send. I have 8 thermostat and sometime you need to turn the wheel to "wake up". When it's near the coordinator it's better, very bad antena inside. When you pair the device you need to see all the informations inside the stat tab, otherwise start pairing again.

Lulu09320, 2024-07-25:
Hello, 
I have the same thermostatic valves, and I tried to follow your procedures to integrate them into Z2M, without success.
I managed to create the "external definition" but I'm not sure how to modify 'zigbee-herdsman-converters/lib/tuya.js' with new references. In which part of the file should I copy/paste these values? 

The valve is a priori recognized without that, but like @aml57 I have the following error: "z2m: Publish 'set' 'current_heating_setpoint' to '0xa4c138cbe9115e11' failed: 'TypeError: Cannot read properties of undefined (reading 'to') '"

In France, these thermostatic valves are installed free of charge as part of a bonus. There is a good chance that the problem arises very often with these references, I hope that they will be integrated by default very soon. 
THANKS !

Lulu09320, 2024-09-07:
> Hello, since 5 day I'm trying to make it work, I bought 2 smart socket to cover more area. I have seen that the thermostat are very bad, low signal will broke all the thing you will send. I have 8 [...]

Have you made any progress on the subject? I tried things that didn't work... 
As indicated above, the equipment seems recognized after the modifications but modification of the set temperature remains impossible...
THANKS

Mouni68, 2024-10-12:
Sorry, too much time to make it work. Now that it's a little colder in France, I was able to test in a real situation, I'm getting back to the problem

When its close to my Sonoff Dongle-E Zigbee with NCP Firmware it's working but more away the set temperature does'nt work or you need to turn the wheel before to wake up... useless. I'm using Jeedom and I made a script that sends the information, checks and if it's not good sends it again up to 3 times. but despite this certain radiators do not take the new temperature. 

I also took advantage of the government's free offer but its products are of poor quality. poor reception and above all half of the thermosats fell from the radiators due to defective adapter ring.

It's strange because with the original router, this one I have to place it in the middle of the living room, which is not practical, the instructions seem to work every time. with Z2M it's not all the time. I don't know how to solve this problem

Amulligue, 2024-11-02:
> Désolé, trop de temps pour le faire fonctionner. Maintenant qu'il fait un peu plus froid en France, j'ai pu tester en situation réelle, je reviens au problème
> 
> Quand il est proche de mon Sonoff Dongle-E Zigbee avec Firmware NCP ça fonctionne mais plus loin la température réglée ne fonctionne pas ou il faut tourner la molette avant de se réveiller... inutile. J'utilise Jeedom et j'ai fait un script qui envoie l'information, vérifiez et si ce n'est pas bon la renvoyer jusqu'à 3 fois. mais malgré cela certains radiateurs ne prennent pas la nouvelle température.
> 
> J'ai aussi profité de l'offre gratuite du gouvernement mais ses produits sont de mauvaise qualité. mauvaise réception et surtout la moitié des thermosats sont tombées des radiateurs à cause d'une bague d'adaptation défectueuse.
> 
> C'est étrange car avec le routeur d'origine, celui-ci je dois le placer au milieu du salon, ce qui n'est pas pratique, les instructions semblent fonctionner à chaque fois. avec Z2M ce n'est pas tout le temps. Je ne sais pas comment résoudre ce problème

bonjour, ou dois je mettre les modifications "batteryVoltToPercent: exports.valueConverterBasic.scale(0, 147, 0, 100),
    divideBy2: exports.valueConverterBasic.divideBy(2), "  dans le fichier tuya.js ?

merci :)

Mouni68, 2024-12-04:
Désolé de la réponse tardive, je suis débordé par le taf.

Suite à mes déboires avec les têtes qui se décrochent, le SAV de certinov m'a remplacé toutes les têtes par des nouvelles de marque Mazda (rien a voir avec le constructeur auto c'est même marqué dessus) modèle MZVT1Z Constructeur Zigbee _TZE204_k6rdmisz, et franchement ça n'a rien à voir.

Z2M ne les reconnaissait pas non plus mais les instructions sont simples et surtout standards pas besoin de modifier des fichiers un simple convertisseur externe suffit. Une meilleure réception du signal permet de les utiliser correctement.

Faites les démarches pour qu'ils vous remplacent les têtes Xanlite.

fred591, 2024-12-07:
Bonjour Mouni68,
J'ai suivi tes conseils et mon thermostat est bien dans mes appareils Z2M mais ne remonte rien.
Mon Ha est installé sur un NUC en débian. J'ai chercher les fichiers zigbee-herdsman-converters/lib/tuya.js et autres herdsmann je ne les trouve pas. Je pense que cela vient de là, le convertisseur externe ne trouve pas les fichiers requis.
Je te remercie par avance pour ton aide.
Fréd

BreymsFR, 2025-02-06:
I have the same issue than fred591, impossible to found zigbee-herdsman-converters/lib/tuya.js. Can someone request to add this modification on the HA Core ?(or Z2M code). I'm so rooky for that :-)

fred591, 2025-03-04:
Bonjour DERBESSE,
Non je n'ai pas réussi à trouver une solution pour ces vannes thermostatiques.
Elle sont appairées avec Z2M (modéle : TS0601 et Fabricant : non-supporté) mais ne distribues aucune information.

Peu à peu je les remplace par des sonoff qui fonctionnent trè bien. je suis à 3/9 de remplacer.

Je tente régulièrement de continuer mes essais.
Fred

DERBESSE, 2025-03-07:
@Koenkk , 
please, what's wrong with the code above ?
The run return 
Error: Cannot find module '/app/node_modules/zigbee-herdsman-converters/dist/converters/fromZigbee.js.js' 
Thanks a lot for your help

fred591, 2025-03-26:
Bonjour DERBESSE,
Je suis en congés cette semaine, alors je passe un peu de temps sur les vannes thermostatiques XANLITE.
Je ma suis installer un home assistant d'essai sur un RPI4. Seul la tête thermo est appairée sur ce système.
Ma config: HASSIO V15.0 sur RPI4, Mosquito en version 6.5.0 et Z2M en version 2.1.3-1 avec une clé conbee3.
Avec, je l'avoue pas plus de réussite. Je te transmets les log reçu dans le journal de Z2M, si tu peu m'éclairer moi je n'y comprends rien.
`[2025-03-26 10:52:18] debug: 	z2m: Received Zigbee message from 'Tete thermo xanlite', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":22648}' from endpoint 1 with groupID 0
[2025-03-26 10:52:18] debug: 	z2m: Skipping message, still interviewing
[2025-03-26 10:52:18] debug: 	zh:deconz:driver: DATA_REQUEST - destAddr: 0x561a EP:1 SeqNr. 177 request id: 6
[2025-03-26 10:52:18] debug: 	zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 6 status: 0
[2025-03-26 10:52:18] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-03-26 10:52:18] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 100110
[2025-03-26 10:52:18] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-03-26 10:52:18] debug: 	zh:deconz:driver: query aps data confirm
[2025-03-26 10:52:18] debug: 	zh:deconz:driver: DATA_CONFIRM - sending data state request - SeqNr. 178
[2025-03-26 10:52:18] debug: 	zh:deconz:frameparser: DATA_CONFIRM RESPONSE - destAddr: 0x561a request id: 6 confirm status: 0
[2025-03-26 10:52:18] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2025-03-26 10:52:18] debug: 	zh:deconz: sendZclFrameToEndpoint - message send with transSeq Nr.: 152
[2025-03-26 10:52:18] debug: 	zh:deconz: false, true, false, 10000
[2025-03-26 10:52:18] debug: 	zh:deconz: resolve request (152)
[2025-03-26 10:52:18] debug: 	zh:deconz:driver: query aps data indication
[2025-03-26 10:52:18] debug: 	zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 179
[2025-03-26 10:52:18] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 179 srcAddr: 0x561a destAddr: 0x0 profile id: 0x104 cluster id: 0xef00 lqi: 214
[2025-03-26 10:52:18] debug: 	zh:deconz:frameparser: response payload: 0999247858
[2025-03-26 10:52:18] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-03-26 10:52:18] debug: 	zh:controller: Received payload: clusterID=61184, address=22042, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=214, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":153,"commandIdentifier":36},"payload":{"payloadSize":22648},"command":{"ID":36,"parameters":[{"name":"payloadSize","type":33}],"name":"mcuSyncTime"}}
[2025-03-26 10:52:18] debug: 	zhc: Getting definitions for: zemismart.js,13,zemismart.js,12,zemismart.js,11,zemismart.js,10,zemismart.js,9,zemismart.js,8,zemismart.js,6,zemismart.js,0,woox.js,3,weten.js,1,tuya.js,358,tuya.js,357,tuya.js,356,tuya.js,355,tuya.js,354,tuya.js,352,tuya.js,351,tuya.js,350,tuya.js,349,tuya.js,348,tuya.js,347,tuya.js,346,tuya.js,345,tuya.js,344,tuya.js,343,tuya.js,342,tuya.js,341,tuya.js,340,tuya.js,338,tuya.js,337,tuya.js,336,tuya.js,335,tuya.js,334,tuya.js,331,tuya.js,330,tuya.js,329,tuya.js,328,tuya.js,326,tuya.js,325,tuya.js,324,tuya.js,323,tuya.js,322,tuya.js,319,tuya.js,318,tuya.js,317,tuya.js,315,tuya.js,314,tuya.js,313,tuya.js,312,tuya.js,311,tuya.js,310,tuya.js,309,tuya.js,308,tuya.js,307,tuya.js,306,tuya.js,305,tuya.js,304,tuya.js,303,tuya.js,302,tuya.js,301,tuya.js,300,tuya.js,299,tuya.js,298,tuya.js,297,tuya.js,296,tuya.js,295,tuya.js,294,tuya.js,288,tuya.js,287,tuya.js,286,tuya.js,285,tuya.js,284,tuya.js,283,tuya.js,282,tuya.js,281,tuya.js,280,tuya.js,279,tuya.js,278,tuya.js,277,tuya.js,276,tuya.js,275,tuya.js,274,tuya.js,273,tuya.js,272,tuya.js,271,tuya.js,269,tuya.js,268,tuya.js,259,tuya.js,258,tuya.js,257,tuya.js,255,tuya.js,254,tuya.js,248,tuya.js,246,tuya.js,245,tuya.js,244,tuya.js,243,tuya.js,242,tuya.js,241,tuya.js,240,tuya.js,239,tuya.js,238,tuya.js,235,tuya.js,234,tuya.js,232,tuya.js,231,tuya.js,230,tuya.js,229,tuya.js,228,tuya.js,227,tuya.js,224,tuya.js,223,tuya.js,222,tuya.js,221,tuya.js,219,tuya.js,216,tuya.js,215,tuya.js,212,tuya.js,211,tuya.js,207,tuya.js,201,tuya.js,200,tuya.js,199,tuya.js,198,tuya.js,197,tuya.js,169,tuya.js,168,tuya.js,167,tuya.js,166,tuya.js,165,tuya.js,164,tuya.js,163,tuya.js,162,tuya.js,161,tuya.js,160,tuya.js,159,tuya.js,158,tuya.js,157,tuya.js,156,tuya.js,155,tuya.js,147,tuya.js,146,tuya.js,145,tuya.js,144,tuya.js,143,tuya.js,142,tuya.js,141,tuya.js,140,tuya.js,139,tuya.js,138,tuya.js,137,tuya.js,136,tuya.js,135,tuya.js,134,tuya.js,133,tuya.js,132,tuya.js,131,tuya.js,130,tuya.js,129,tuya.js,128,tuya.js,127,tuya.js,126,tuya.js,125,tuya.js,124,tuya.js,108,tuya.js,107,tuya.js,106,tuya.js,83,tuya.js,82,tuya.js,81,tuya.js,80,tuya.js,79,tuya.js,78,tuya.js,77,tuya.js,76,tuya.js,75,tuya.js,73,tuya.js,70,tuya.js,64,tuya.js,63,tuya.js,62,tuya.js,61,tuya.js,60,tuya.js,59,tuya.js,58,tuya.js,57,tuya.js,56,tuya.js,55,tuya.js,54,tuya.js,53,tuya.js,24,tuya.js,23,tuya.js,22,tuya.js,21,tuya.js,20,tuya.js,19,tuya.js,18,tuya.js [... cut ...]
            [115, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDP2],
            [116, 'open_window_temperature', tuya.valueConverter.divideBy2],
            [117, 'open_window_time', tuya.valueConverter.raw],
            [118, 'boost_time', tuya.valueConverter.countdown],                
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;'

Si tu as un peu de temps pour m'aider, je t'en remercie par avance.
Fréd

DERBESSE, 2025-04-02:
@fred591 .
J'ai réussi a appairer et récupérer les infos des têtes.
La démarche que tu reprends est la bonne.
En revanche, j'ai supprimer les extensions .js a la fin des déclarations des constantes 
Exemple : 'const fz = require('zigbee-herdsman-converters/converters/fromZigbee.js') remplacé par 'const fz = require('zigbee-herdsman-converters/converters/fromZigbee') 
Idem sur les autres lignes.
😉 Tiens moi au courant !

jojo01fr, 2026-01-17:
Hi everyone, having received one of this unit purchased 11.99€ here
[https://www.xanlite-store.com/fr/thermostats/1876-tete-thermostatique-radio-connectee-pour-radiateur-hydraulique-systeme-thermostats-connectes-3700619438993.html](url)

I've imported fred591's external converter, thank you fred ! And modified it a bit to remove some bugs and add the most important part, to be able to use the schedule. So, feel free to use it, I made a submission here
[https://github.com/Koenkk/zigbee2mqtt/issues/30689](url)

bisouilles aux moeuf !

guillaumepicard831-wq, 2026-08-28:
Bonjour,

Les précédentes issues concernant ce modèle étant désormais fermées, je partage les résultats obtenus après plusieurs heures de rétro-ingénierie et de tests sur un thermostat :

TS0601
manufacturerName: _TZE200_ytryxh0a

L'objectif est de partager les datapoints identifiés ainsi que les mesures réalisées sur le DP34 afin de comparer les résultats avec d'autres utilisateurs de cette variante.

Datapoints identifiés
DP16 = current_heating_setpoint
DP24 = local_temperature
DP30 = child_lock
DP34 = batterie (valeur brute)
DP101 = comfort_temperature
DP102 = eco_temperature
DP103 = holiday_start_stop
DP106 = boost_heating
DP107 = window_detection
DP116 = open_window_temperature
DP117 = open_window_time

DP16
La consigne réelle semble être portée par le DP16 avec la formule :

consigne = DP16 / 2

Exemples observés :

10.0°C -> DP16 = 20
16.0°C -> DP16 = 32
20.0°C -> DP16 = 40
29.5°C -> DP16 = 59

Le mapping suivant fonctionne correctement :

[16, 'current_heating_setpoint', tuya.valueConverter.divideBy2]

La synchronisation est maintenant fonctionnelle dans les deux sens :

Home Assistant -> thermostat
thermostat -> Home Assistant
DP34 (batterie)
Le DP34 correspond à l'information batterie (confirmé expérimentalement).

Les mesures suivantes ont été obtenues en comparant la valeur brute du DP34 avec la tension affichée dans l'application Tuya :

2.39 V -> 66
2.49 V -> 80
2.53 V -> 83
2.74 V -> 101-102
2.76 V -> 106
2.79 V -> 108
2.82 V -> 110
3.12 V -> 148

Le mapping actuellement utilisé est :

[34, 'battery_raw', tuya.valueConverter.raw]

Une formule empirique semble donner des résultats cohérents :

battery_voltage ≈ 1.80 + (battery_raw / 112)

Exemples :

66 -> 2.39 V
80 -> 2.51 V
102 -> 2.71 V
110 -> 2.78 V
148 -> 3.12 V

⚠️ Cette formule n'est pas validée officiellement et repose uniquement sur des mesures expérimentales.

Convertisseur actuellement fonctionnel
Le convertisseur est désormais pleinement fonctionnel pour les fonctions principales du thermostat :

✅ Consigne remontée du thermostat vers Home Assistant

✅ Consigne envoyée de Home Assistant vers le thermostat

✅ Température locale

✅ Verrou enfant

✅ Températures confort / éco

✅ Détection fenêtre ouverte

✅ Mode boost

✅ Batterie brute (DP34)

Informations de l'appareil testé
Type : EndDevice

IEEE Address : 0xa4c1384f6938d2c9

Model ID : TS0601

Manufacturer Name : _TZE200_ytryxh0a

Definition : TS0601_thermostat

Vendor : Tuya

Points restant à éclaircir
Validation ou correction de la formule de conversion DP34 -> Volts
Origine du champ "battery" qui reste bloqué à 102 même après suppression de e.battery().withUnit('%')
Fonctionnement du DP104 (local_temperature_calibration)
Si d'autres personnes ont travaillé sur cette variante _TZE200_ytryxh0a, disposent d'autres mesures batterie, ou ont déjà identifié officiellement le DP34 ou le DP104, je serais très intéressé par un partage d'informations afin de comparer les résultats.

Merci !

guillaumepicard831-wq, 2026-08-28:
J'ai remarqué que certains convertisseurs TS0601 utilisent :

[34, 'battery', tuya.valueConverter.batteryVoltToPercent]

et

[104, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration1]

Mon appareil _TZE200_ytryxh0a semble utiliser également DP34 pour la batterie.

Quelqu'un sait-il précisément quel format est attendu par batteryVoltToPercent et si ce convertisseur est adapté à cette variante ?

guillaumepicard831-wq, 2026-09-04:
Bonjour,

Voici une mise à jour après de nouveaux tests approfondis sur cette tête thermostatique :

- Modèle Zigbee : `TS0601`
- Fabricant : `_TZE200_ytryxh0a`
- Marque commerciale : Xanlite
- Référence : `KZTETHEAU`

## Mise à jour du convertisseur

Le convertisseur utilise maintenant la base Tuya moderne :

```javascript
extend: [
    tuya.modernExtend.tuyaBase({
        dp: true,
        timeStart: '1970',
    }),
],
```

Cette configuration remplace les anciennes déclarations suivantes :

```javascript
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
onEvent: tuya.onEventSetTime,
configure: tuya.configureMagicPacket,
```

## Résultats concernant le DP16

Les tests permettent de confirmer que le **DP16 est le datapoint qui commande réellement la consigne physique de la tête**.

Le DP16 utilise une valeur correspondant à la température multipliée par 2.

Valeurs confirmées expérimentalement :

```text
10,0 °C -> DP16 = 20
15,0 °C -> DP16 = 30
16,0 °C -> DP16 = 32
20,0 °C -> DP16 = 40
28,0 °C -> DP16 = 56
29,5 °C -> DP16 = 59
```

Tests réalisés avec succès :

- écriture directe de 28 °C sur le DP16 : moteur actif et affichage physique à 28 °C ;
- écriture directe de 15 °C sur le DP16 : moteur actif et affichage physique à 15 °C ;
- `climate.set_temperature` vers 28 °C : moteur actif et affichage physique à 28 °C ;
- `climate.set_temperature` vers 15 °C : moteur actif et affichage physique à 15 °C ;
- pilotage par Versatile Thermostat en mode `over_climate` : validé dans les deux sens.

Le mapping retenu est donc :

```javascript
[16, 'current_heating_setpoint', tuya.valueConverter.divideBy2],
```

## Résultats concernant le DP105

Le DP105 transporte également des valeurs ressemblant à une température multipliée par 2.

Exemples observés :

```text
16,0 °C -> DP105 = 32
[... 268 lines cut ...]
2,79 V          | 108
2,82 V          | 110
3,12 V          | 148
```

Une formule empirique a été estimée :

```text
battery_voltage ≈ 1,80 + (battery_raw / 112)
```

Cette formule n’est pas officiellement validée. Elle provient uniquement des mesures expérimentales.

Le champ `battery` publié par Zigbee2MQTT semble parfois rester bloqué à `102`, indépendamment de l’évolution de `battery_raw`.

## État actuel des tests

- Synchronisation Home Assistant vers la tête : validée via DP16.
- Synchronisation tête vers Home Assistant : validée via DP16.
- Pilotage par `climate.set_temperature` : validé.
- Pilotage par Versatile Thermostat `over_climate` : validé.
- Auto-régulation Versatile Thermostat testée en mode Légère.
- DP105 conservé uniquement comme donnée brute d’observation.

Merci.

guillaumepicard831-wq, 2026-10-17:
Bonjour,

J’ai poursuivi les tests sur la tête thermostatique Xanlite KZTETHEAU (TS0601, _TZE200_ytryxh0a) avec Zigbee2MQTT.

J’ai maintenant un convertisseur externe fonctionnel avec le pilotage réel de la consigne et la calibration de température opérationnels.

Mappings confirmés :

[16, 'current_heating_setpoint', tuya.valueConverter.divideBy2],
[24, 'local_temperature', tuya.valueConverter.divideBy10],
[104, 'local_temperature_calibration', tuya.valueConverter.divideBy10],
[105, 'dp105', tuya.valueConverter.raw],

DP16 - Consigne physique réelle

DP16 est bien la consigne physique réellement utilisée par la tête.

Exemples observés :

15,0 °C -> DP16 = 30
20,0 °C -> DP16 = 40
28,0 °C -> DP16 = 56
29,5 °C -> DP16 = 59

Une modification de current_heating_setpoint depuis l’entité climate de Zigbee2MQTT modifie bien la consigne affichée sur la tête et provoque le fonctionnement réel de la vanne.

Ce fonctionnement est également validé avec Versatile Thermostat dans Home Assistant en mode over_climate.

DP104 - Calibration de la température locale

DP104 est maintenant confirmé comme étant l’offset de calibration de la température interne de la tête.

Le mapping fonctionnel est :

[104, 'local_temperature_calibration', tuya.valueConverter.divideBy10],

avec l’exposition :

.withLocalTemperatureCalibration(-9, 9, 0.1, ea.STATE_SET)

DP104 utilise le datatype Tuya 2 avec une résolution de 0,1 °C.

Exemples :

+0,1 °C -> valeur brute DP104 = 1
+0,2 °C -> valeur brute DP104 = 2
0,0 °C -> valeur brute DP104 = 0

Les calibrations positives et négatives ont été testées avec succès, ainsi que le retour à zéro.

Réponse Tuya brute observée pour +0,1 °C :

dp: 104
datatype: 2
data: [0,0,0,1]

Pour un retour à 0 °C :

dp: 104
datatype: 2
data: [0,0,0,0]

Donc, sur cette tête, divideBy10 fonctionne correctement pour DP104.

Le convertisseur précédemment proposé :

tuya.valueConverter.localTempCalibration1

ne fonctionne pas avec ma version actuelle de zigbee-herdsman-converters.

IMPORTANT - Initialisation de DP104

Lorsque local_temperature_calibration n’a jamais été initialisé, la valeur peut apparaître null dans Zigbee2MQTT ou unknown dans Home Assistant.

Il suffit alors d’écrire une seule fois :

0

dans local_temperature_calibration.

Après cette initialisation, la lecture et l’écriture de DP104 fonctionnent normalement.

Synchronisation avec un thermomètre externe et Versatile Thermostat

J’ai également validé DP104 avec la fonction de synchronisation de la température interne de Versatile Thermostat en mode over_climate dans Home Assistant.

La température de référence provient d’un thermomètre Sonoff placé dans la pièce.

VTherm calcule automatiquement l’offset nécessaire et l’envoie à local_temperature_calibration.

Exemple réel sur une tête :

Thermomètre de la pièce : 21,2 °C
Calibration appliquée : -1,7 °C
Température locale de la tête après correction : 21,2 °C

Sur une deuxième tête située dans la même pièce :

Thermomètre de la pièce : 21,2 °C
Calibration appliquée : +3,1 °C
Température locale de la tête après correction : 21,2 °C

Le même fonctionnement a été validé sur deux autres têtes dans une autre pièce.

La synchronisation automatique par calibration est maintenant configurée sur mes 13 têtes thermostatiques.

DP105

DP105 ne doit pas être utilisé comme consigne physique.

Pendant mes tests, DP105 pouvait accepter et confirmer une valeur sans provoquer le déplacement du moteur ni modifier la consigne affichée physiquement sur la tête.

Je le conserve donc uniquement pour observation :

[105, 'dp105', tuya.valueConverter.raw],

Mappings principaux actuellement utilisés :

[16, 'current_heating_setpoint', tuya.valueConverter.divideBy2],
[24, 'local_temperature', tuya.valueConverter.divideBy10],
[104, 'local_temperature_calibration', tuya.valueConverter.divideBy10],
[105, 'dp105', tuya.valueConverter.raw],

Je peux également fournir le convertisseur externe complet actuellement utilisé si cela peut être utile pour intégrer officiellement le support de _TZE200_ytryxh0a.