Koenkk/zigbee2mqtt #23080: Error while starting zigbee-herdsman / Error Resource temporarily unavailable Cannot lock port

andreas-vester, 2024-06-16:
### What happened?

I am not able to start zigbee2mqtt.
I installed it via proxmox ve helper scripts in a LXC container.

It seems that I have some issues with my Sonoff coordinator. My feeling is that the passthrough into the LXC Container is working proberly.
```

root@zigbee2mqtt:/opt/zigbee2mqtt# ls -la /dev/serial/by-id/
total 0
drwxr-xr-x 2 root root 80 Jun 16 21:28 .
drwxr-xr-x 3 root root 60 Jun 16 22:00 ..
lrwxrwxrwx 1 root root 13 Jun 16 21:06 usb-0658_0200-if00 -> ../../ttyACM0
lrwxrwxrwx 1 root root 13 Jun 16 21:28 usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_2aa2163d1b3bec11bcc4a4957a0af07f-if00-port0 -> ../../ttyUSB0
```

```
root@zigbee2mqtt:/opt/zigbee2mqtt# test -w /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_2aa2163d1b3bec11bcc4a4957a0af07f-if00-port0 && echo success || echo failure
success
```

Here's my config:
```
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee
  server: mqtt://10.10.2.11:1883
  user: xxx
  password: xxx
  keepalive: 60
serial:
  port: /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_2aa2163d1b3bec11bcc4a4957a0af07f-if00-port0
  adapter: zstack
advanced:
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false
frontend: true
log_level: debug
```

### What did you expect to happen?

Successful start of zigbee2mqtt ;-)

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

?

### Adapter

usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Running in a proxmox LXC container

### Debug log

root@zigbee2mqtt:/opt/zigbee2mqtt# cd /opt/zigbee2mqtt && npm start

> zigbee2mqtt@1.38.0 start
> node index.js

[2024-06-16 23:04:13] info:     z2m: Logging to console, file (filename: log.log)
[2024-06-16 23:04:13] info:     z2m: Starting Zigbee2MQTT version 1.38.0 (commit #f184730)
[2024-06-16 23:04:13] info:     z2m: Starting zigbee-herdsman (0.49.2)
[2024-06-16 23:04:13] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_2aa2163d1b3bec11bcc4a4957a0af07f-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-06-16 23:04:13] error:    z2m: Error while starting zigbee-herdsman
[2024-06-16 23:04:13] error:    z2m: Failed to start zigbee
[2024-06-16 23:04:13] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-06-16 23:04:13] error:    z2m: Exiting...
[2024-06-16 23:04:13] error:    z2m: Error: Error Resource temporarily unavailable Cannot lock port

francisp2, 2024-06-17:
You don't have ZHA installed ?

andreas-vester, 2024-06-17:
No, I haven't. I don't even use HomeAssistant.
I am just interested in a stand-alone installation of zigbee2mqtt.

andreas-vester, 2024-06-17:
I figured it out by myself. I indeed did not correctly pass through the stick to my LXC container.

closed 2024-06-17 by andreas-vester