Koenkk/zigbee2mqtt #23084: zigbee2mqtt restart every 2 minutes

DuchkPy, 2024-06-17:
### What happened?

Hello,

Basically, about every 2 minutes, z2m restart.
I tried to look on logs, on debug, warning, ... to find the reason without success.

So far, I have integrated only 1 equipment (since my previous support https://github.com/Koenkk/zigbee2mqtt/issues/23034), then I let z2m runs to see how it's working.
And I discovered that it's not stable at all.

From log, you can see z2m starting from 10:33:51 to 10:33:57, then suddenly it just stop at 10:35:20 without raising any message indicating the reason (or I didn't look at the correct place, so far).
And `systemd` make z2m restart about 20 seconds later.

Please raise your hint and I will try to solve this, or to provide other logs.

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20210708

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P Coordinator

### Setup

ontainer on NUC

### Debug log

[2024-06-17 10:33:51] info:     z2m: Logging to console, file (filename: log.log)
[2024-06-17 10:33:51] info:     z2m: Starting Zigbee2MQTT version 1.38.0 (commit #f1847301)
[2024-06-17 10:33:51] info:     z2m: Starting zigbee-herdsman (0.49.2)
[2024-06-17 10:33:51] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-06-17 10:33:51] info:     zh:zstack:znp: Serialport opened
[2024-06-17 10:33:52] info:     z2m: zigbee-herdsman started (resumed)
[2024-06-17 10:33:52] info:     z2m: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'
[2024-06-17 10:33:52] info:     z2m: Currently 1 devices are joined:
[2024-06-17 10:33:52] info:     z2m: ZBmini (0x847127ffa6c5595d): ZBMINI - SONOFF Zigbee two way smart switch (Router)
[2024-06-17 10:33:52] info:     z2m: Zigbee: disabling joining new devices.
[2024-06-17 10:33:52] info:     z2m:mqtt: Connecting to MQTT server at mqtt://192.168.1.110:1883
[2024-06-17 10:33:52] info:     z2m:mqtt: Connected to MQTT server
[2024-06-17 10:33:52] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
[2024-06-17 10:33:52] info:     z2m: Started frontend on port 9572
[2024-06-17 10:33:52] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"f1847301","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"},"log_level":"info","network":{"channel":20,"extendedPanID":"0x00124b002e1e828e","panID":30044},"permit_join":false,"version":"1.38.0"}'
[2024-06-17 10:33:52] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
[2024-06-17 10:33:52] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBmini sous-pente', payload '{"linkquality":12,"state":"OFF"}'
[2024-06-17 10:33:52] info:     z2m: Zigbee2MQTT started!
[2024-06-17 10:33:57] info:     z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0x00124b002e1e828e/connection_state/config', payload '{"device":{"hw_version":"zStack3x0 20210708","identifiers":["zigbee2mqtt_bridge_0x00124b002e1e828e"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.38.0"},"device_class":"connectivity","entity_category":"diagnostic","name":"Connection state","object_id":"zigbee2mqtt_bridge_connection_state","origin":{"name":"Zigbee2MQTT","sw":"1.38.0","url":"https://www.zigbee2mqtt.io"},"payload_off":"offline","payload_on":"online","state_topic":"zigbee2mqtt/bridge/state","unique_id":"bridge_0x00124b002e1e828e_connection_state_zigbee2mqtt","value_template":"{{ value }}"}'
[2024-06-17 10:35:20] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
[2024-06-17 10:35:20] info:     z2m:mqtt: Disconnecting from MQTT server
[2024-06-17 10:35:20] info:     z2m: Stopping zigbee-herdsman...
[2024-06-17 10:35:29] info:     zh:controller: Wrote coordinator backup to '/opt/zigbee2mqtt/data/coordinator_backup.json'
[2024-06-17 10:35:29] info:     zh:zstack:znp: closing
[2024-06-17 10:35:29] info:     zh:zstack:znp: Port closed
[2024-06-17 10:35:29] info:     z2m: Stopped zigbee-herdsman
[2024-06-17 10:35:29] info:     z2m: Stopped Zigbee2MQTT

sstraus, 2024-06-23:
Same problem, but my log.log is always empty, and a new folder for the logs is created every minute.

flipreverse, 2024-08-08:
I'm affected by the same problem. On my opinion, this has something to do with `Type=notify` in the SystemD unit. `systemctl start zigbee2mqtt` hangs if I use this type of service.
If a change the unit type to `simple`, and disable `WatchdocSec=10s`, everything works.

closed 2024-12-29 by github-actions[bot]