Koenkk/zigbee2mqtt #23087: How can I flash SLZB-06M to cp-uart-hw-v7.4.1.0-slzb-06m-230400.gbl ?

MilesTEG1, 2024-06-17:
Hello,

> Hi folks,
> yesterday I switched to ember driver:
> Flashed my SLZB-06M to [this firmware](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/slzb-06m/ncp-uart-hw-v7.4.1.0-slzb-06m-230400.gbl) von @darkxst and changed the driver to ember in the frontend.
> After the mandantory z2m restart everything came back in seconds 👍 
> 
> Here my setup.
> 
> z2m is running on my proxmox server inside an LXC.
> Also running emqx inside another LXC as my MQTT broker.
> 
> z2m accesses the SLZB-06M via TCP.

@jusicgn How have you managed to flash ty `cp-uart-hw-v7.4.1.0-slzb-06m-230400.gbl` to the SLZB-06M ?

MilesTEG1, 2024-06-18:
@jusicgn 
Hello,
I have the embedded firmware, in version 2.3.6 :

[image]

When I try to update the Zigbee FW, with official version, I get this error:
[image]
I think I have a faulty device.

I try to put the .gbl file in this interface, but the flash hasn't work.
WHen I try to flash with https://darkxst.github.io/silabs-firmware-builder/ I get a communication error.

MilesTEG1, 2024-06-18:
> @MilesTEG1 had the same issue even when trying to flash stock ota. Please have a look at the baudrate settings in the zigbee section. Set it to 115200 then try flashing again.

The baudrate is already set to 115200.

MilesTEG1, 2024-06-18:
> That's strange. Setting the baudrate to 115200 fixed the bootloader err for me.
> 
> Did you try to flash by usb? For this you need to install the usb driver as described on the webflasher site. Sometimes it took me a few tries until it worked. Also not every usb cable worked. Needed to use a usb-c to usb-a cable. Usb-c to usb-c didn't work for me.
Yes I tried to flash by USB, with the USB-A (mini) to USB-C adaptor sold with the SLZB-06M.

MilesTEG1, 2024-06-18:
> Last idea I have on this is to power off the device for a while before trying to flash again. Maybe even do a reset by holding the button while powering it on.
> 
> You could also try to remove the ethernet cable while flashing via usb. Didn't liked to have both plugged in as I had my tries with this.

I have tested exactly what you propose here.
I think I have a faulty device. I'm fighting with the seller to get an exchange, or a refund.

Kuechenfenster, 2024-06-21:
I got 2 SLZB-06M running in my setup and the 1st one I was able to upgrade via OTA, but the 2nd one I get the 'Bootloader enable err" message and can't find a way around so far... I contacted Customer Support and waiting for a feedback.

Outsidewall, 2024-07-06:
@Kuechenfenster Did you get a response, I have the same issue.

MilesTEG1, 2024-07-06:
> @Kuechenfenster Did you get a response, I have the same issue.

Hi,
My first SLZB-06M seemed to be defective and was returned to the seller. I've bought another one, and I got one time the error `bootloader enable err`. I retry another time, and it passed.
So, if you try multiple times and you still get this error, your coordinator may be defective.

And be careful, you have to choose the right file for the right baudrate. The SLZB-06M use 115200. Not the one I wrote in the issue.
I'll edit to warn in my first post.

closed 2024-08-02 by MilesTEG1

Stephan-4711, 2024-08-14:
here the original zigbee firmware can get updated via ota.
trying to load the gbl file I get Canceled by target
and with USB I don't get a connection