Koenkk/zigbee2mqtt #23088: BMCT-RZ does not react to external switches after pairing with Zigbee2MQTT

cbrueser, 2024-06-17:
### What happened?

After pairing BMCT-RZ (relay) with Z2M, the connected physical switch could no longer turn on/off the relay.

### What did you expect to happen?

External switch allows the relay to be turned on and off as when using a Bosch Smart Home Controller as the coordinator.

### How to reproduce it (minimal and precise)

Pair relay with Zigbee2MQTT using install code. Afterwards an external switch will no longer function.

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20240315

### Adapter

Zigstar UZG-01

### Setup

Docker container

cbrueser, 2024-06-20:
@burmistrzak Thanks for checking. I also took a look at the converter code myself and compared to the BMCT-SLZ implementation (which works great by the way). As you say, looks like some basic initial config stuff is missing.

bigbeka, 2024-12-17:
I’m glad I’m not the only one experiencing this. I just renovated my home and all lights and switches in all rooms are using Bosch Smart Relay, and the only way to turn on/off the lights is via home assistant.

My setup:
BMCT-RZ > SLZB-06 (Zigbee Controller) > LAN > UniFi switch > Home Assistant (RPi 4B 8GB).

@burmistrzak any update on this so far?

gecomaia, 2025-02-15:
Same behavior on my side, exactly as you’ve described!

bigbeka, 2025-02-15:
@gecomaia 

I replaced my x6 BMCT-RZ with x3 BMCT-SLZ, works perfectly with Z2M. BMCT-SLZ have 2 relays in one device, which is great if you can mount the relays in your distribution box, they take up less space and you save on cost too.

gecomaia, 2025-02-15:
Thank you for the hint!
Unfortunately, this is not possible in my case because the BMCT-RZ has a galvanically isolated switch, which is necessary for my garage door switch since it operates on only 12V.
I've used the BMCT-SLZ, which also works great, but only in combination with the mains.

BTW, fantastic work!

cbrueser, 2025-06-18:
Still relevant. Hoping @burmistrzak may get around to take a look into this.

DerDreschner, 2025-09-07:
The reason for that issue is the unanswered `zclVersionReadRequest`, which all `BMCT` devices are affected by. I fixed that in https://github.com/Koenkk/zigbee-herdsman-converters/pull/9941 and expose a lot more device options for `BMCT-DZ` and `BMCT-RZ`.

Hope you like it!

DerDreschner, 2025-09-07:
The reason for that is that the internal `switchType` is set to `none` by default. The device ignores all external inputs then, like when `childLock` is activated or the switch input is `decoupled`.

I expose those and a whole lot more device options in https://github.com/Koenkk/zigbee-herdsman-converters/pull/9941. That way, you should be able to set up the device correctly and use external switches with it. OTA support for the `BMCT-RZ` is being enabled with that change as well.

Hope you like it!

cbrueser, 2025-09-07:
Thanks a lot! Looking forward to trying it once your PR is merged and available.

cbrueser, 2025-10-03:
Everything works now as expected. Thanks.

closed 2025-10-03 by cbrueser