Koenkk/zigbee2mqtt #23089: TS0042 without action

Cic0r, 2024-06-17:
### What happened?

Added the device Tuya TS0042, but No actions are found. 
[image]

### What did you expect to happen?

Find actions for -at least - Press. Double Press and hold would be even better.

### Zigbee2MQTT version

1.38.0-1

### Adapter firmware version

7.4.2 (or sth. similar, flashed it right now)

### Adapter

Sky Connect EZSP (with ember Driver)

### Setup

Addon on HA on RPi3

### Debug log

Debug gives me the following after pressing the Buttons:

[2024-06-17 21:40:33] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Lichtschalter', payload '{"action":"1_single","battery":100,"linkquality":240,"voltage":3000}'
[2024-06-17 21:40:33] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Lichtschalter', payload '{"action":"","battery":100,"linkquality":240,"voltage":3000}'
[2024-06-17 21:40:33] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Lichtschalter/action', payload '1_single'
[2024-06-17 21:40:34] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Lichtschalter', payload '{"action":"2_single","battery":100,"linkquality":236,"voltage":3000}'
[2024-06-17 21:40:34] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Lichtschalter', payload '{"action":"","battery":100,"linkquality":236,"voltage":3000}'
[2024-06-17 21:40:34] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Lichtschalter/action', payload '2_single'

martintilma, 2024-06-18:
I have the same with IKEA E2001/E2002. Not sure if it is the same problem. Since a day the remote controls don't work anymore.

```
zigbee2mqtt  | [2024-06-18 21:00:34] debug: 	zh:zstack:unpi:parser: <-- [254,7,69,196,102,10,2,121,194,157,82,156]
zigbee2mqtt  | [2024-06-18 21:00:34] debug: 	zh:zstack:unpi:parser: --- parseNext [254,7,69,196,102,10,2,121,194,157,82,156]
zigbee2mqtt  | [2024-06-18 21:00:34] debug: 	zh:zstack:unpi:parser: --> parsed 7 - 2 - 5 - 196 - [102,10,2,121,194,157,82] - 156
zigbee2mqtt  | [2024-06-18 21:00:34] debug: 	zh:zstack:znp: AREQ: <-- ZDO - srcRtgInd - {"dstaddr":2662,"relaycount":2,"relaylist":[49785,21149]}
zigbee2mqtt  | [2024-06-18 21:00:34] debug: 	zh:zstack:unpi:parser: --- parseNext []
zigbee2mqtt  | [2024-06-18 21:00:34] debug: 	zh:zstack:unpi:parser: <-- [254,29,68,129,0,0,5,0,102,10,1,1,0,63,0,21,167,47,0,0,9,21,124,17,17,7,1,1,13,0,157,82,27,173]
zigbee2mqtt  | [2024-06-18 21:00:34] debug: 	zh:zstack:unpi:parser: --- parseNext [254,29,68,129,0,0,5,0,102,10,1,1,0,63,0,21,167,47,0,0,9,21,124,17,17,7,1,1,13,0,157,82,27,173]
zigbee2mqtt  | [2024-06-18 21:00:34] debug: 	zh:zstack:unpi:parser: --> parsed 29 - 2 - 4 - 129 - [0,0,5,0,102,10,1,1,0,63,0,21,167,47,0,0,9,21,124,17,17,7,1,1,13,0,157,82,27] - 173
zigbee2mqtt  | [2024-06-18 21:00:34] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":5,"srcaddr":2662,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":63,"securityuse":0,"timestamp":3122965,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[21,124,17,17,7,1,1,13,0]}}
zigbee2mqtt  | [2024-06-18 21:00:34] debug: 	zh:controller: Received payload: clusterID=5, address=2662, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=63, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4476,"transactionSequenceNumber":17,"commandIdentifier":7},"payload":{"value":257,"value2":13},"command":{"ID":7,"parameters":[{"name":"value","type":33},{"name":"value2","type":33}],"name":"tradfriArrowSingle"}}
zigbee2mqtt  | [2024-06-18 21:00:34] debug: 	zh:zstack:unpi:parser: --- parseNext []
zigbee2mqtt  | [2024-06-18 21:00:34] debug: 	z2m: Received Zigbee message from 'Keuken afstandsbediening', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":257,"value2":13}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-06-18 21:00:34] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Keuken afstandsbediening', payload '{"action":"arrow_left_click","battery":85,"linkquality":63,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
zigbee2mqtt  | [2024-06-18 21:00:34] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Keuken afstandsbediening', payload '{"action":"","battery":85,"linkquality":63,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
zigbee2mqtt  | [2024-06-18 21:00:34] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Keuken afstandsbediening/action', payload 'arrow_left_click'
```

In my case the:
`"action":"arrow_left_click"`
followed directly by:
`"action":""`

martintilma, 2024-06-18:
I don't know why but I got it working again: 

- I replaced the blueprint with: lsismeiro/ikea_e2001_e2002.yaml (not sure if this was needed)
- I removed the device from Zigbee2MQTT
- I repaired the device
- Updated the name with: "Update Home Assistant entity ID" toggled on
- Connected the device to the blueprint

The Zigbee2MQTT UI still shows: Empty string("")

Nerivec, 2024-06-22:
Actions are ephemeral. The state updates just long enough to trigger whatever is using it, then reverts to nothing.
Docs usually mention the values used by devices:
https://www.zigbee2mqtt.io/devices/TS0042.html#action-enum
https://www.zigbee2mqtt.io/devices/E2001_E2002.html#button-action-mapping
(otherwise they can be found in the logs by pushing buttons and seeing what values come in: `MQTT publish: topic 'zigbee2mqtt/Lichtschalter/action', payload '1_single'`).

Cic0r, 2024-07-01:
Updated to 1.39.0 and Problem is still there. No actions, Status and so on. Can't Change anything.

Cic0r, 2024-07-01:
incorrectly closed

closed 2024-12-29 by github-actions[bot]