Koenkk/zigbee2mqtt #23090: TS110E stopped working (unsupported)

stewe93, 2024-06-17:
### What happened?

The following device stopped working after the latest update from dev-branch: 
```
Zigbee Model
TS110E
Zigbee Manufacturer
_TZ3210_k1msuvg6
Support status
Null [(how_to_add_support)](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html)
```
I don't think it's related to my EZSP driver (I know I have to migrate eventually, but everything else works, btw it it possible to migrate from an already working multipan setup to the zigbee only ember driver without repairing? I haven't found a guide for this) 

I could't find this manufacturer model/id in [tuya.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts), also checked the commits, nothing related as I can see, but  maybe somehow this is recognised now and wasn't until this update. 

On the dev console tab, generated the following config: 
```
const {identify, light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS110E'],
    model: 'TS110E',
    vendor: '_TZ3210_k1msuvg6',
    description: 'Automatically generated definition',
    extend: [identify(), light()],
    meta: {},
};

module.exports = definition;
``` 
but this does nothing for me, I guess. After a restart still unsupported.

I dont know if this is a problem or not, but I also could open a PR to add support for this if its just a missing id or converter if you would like.

### Zigbee2MQTT version

 1.38.0-dev commit: a85d3e4

### Adapter firmware version

7.3.1.0 build 0

### Adapter

EZSP 

### Setup

Docker: koenkk/zigbee2mqtt:latest-dev

Koenkk, 2024-06-18:
Fixed in the latest dev!

closed 2024-06-18 by Koenkk

stewe93, 2024-06-19:
Thank you very much for your fast response and fix!!!