Koenkk/zigbee2mqtt #23109: After updating from zigbee2mqtt docker version 1.37.1 to 1.38.0 FUT035Z+ are not controlable anymore via Home Assistant

gklank, 2024-06-19:
### What happened?

- can not switch on / off or adjust brightness of the FUT035Z+ via Home Assistant
- via zigbee2mqtt UI it is functioning already

### What did you expect to happen?

should function as before

### How to reproduce it (minimal and precise)

updating to docker image 1.38.0

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

CC2652R_coordinator - version from 20230507

### Adapter

CC2652R_coordinator

### Setup

docker

### Debug log

homeassistant/WandMitte = {"brightness":173,"color_mode":"color_temp","color_temp":500,"do_not_disturb":null,"linkquality":51,"state":"ON"}
              availability = online
              set = {"state":"ON","brightness":127}

1. changes via the zigbee2mqtt UI will change the entries in the "homeassistant/WandMitte" line - but just there
2. changes in Home Assistant is changing the entries in the "set = {"state":"ON","brightness":127}" line - but just there
   but no changes in the "homeassistant/WandMitte" line

- changes in "point 1"  are shown on the LED driven bei the FUT035Z+
- changes in "point 2"  are not shown on the LED driven bei the FUT035Z+

By downgrading back to "koenkk/zigbee2mqtt:1.37.1" the FUT035Z+ are controllable via Home Assistant again.

vvrockhound, 2024-06-21:
I cannot use 1.38 it does not listen to home assistant.
I have to back down to 1.37.1

max3163, 2024-06-22:
Same problem with Domoticz too with the same adapter in docker. 
Revert to 1.37.1 is OK.

Koenkk, 2024-06-24:
Could you provide the data/database.db entry of this device?

gklank, 2024-06-24:
Hi,

as I can see there is nothing private in there.

Therefore here the database.db in total - small environment, just started with zigbee2mqtt...

Regards

Gerhard
[database.db.txt](https://github.com/user-attachments/files/15952536/database.db.txt)

Koenkk, 2024-06-25:
I do not see any changes for this device, are you using the latest Home Assistant version?

gklank, 2024-06-25:
Hi,

yes:
Core 2024.6.4
Frontend 20240610.1

Regards

Gerhard

Koenkk, 2024-06-26:
Could you provide the debug log when trying to turn of the device via HA?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

floyd871, 2024-06-26:
I have the same Issue. Is this snippet enough?
``
[2024-06-26 14:31:27] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'homeassistant/HWR Licht', payload '{"action":null,"consumption":0.18121902644634247,"current":0,"device_temperature":33,"energy":0.18,"flip_indicator_light":null,"last_seen":"2024-06-26T14:31:27+02:00","led_disabled_night":null,"linkquality":182,"operation_mode":null,"power":0,"power_outage_count":5,"power_outage_memory":null,"state":"OFF","update":{"installed_version":2839,"latest_version":2839,"state":"idle"},"update_available":null,"voltage":235.425}'
[2024-06-26 14:31:28] [34mdebug[39m: 	zh:zstack:unpi:parser: <-- [254,28,68,129,0,0,4,11,206,160,1,1,0,43,0,76,194,107,0,0,8,24,4,10,11,5,41,51,0,142,24,28,246]
[2024-06-26 14:31:28] [34mdebug[39m: 	zh:zstack:unpi:parser: --- parseNext [254,28,68,129,0,0,4,11,206,160,1,1,0,43,0,76,194,107,0,0,8,24,4,10,11,5,41,51,0,142,24,28,246]
[2024-06-26 14:31:28] [34mdebug[39m: 	zh:zstack:unpi:parser: --> parsed 28 - 2 - 4 - 129 - [0,0,4,11,206,160,1,1,0,43,0,76,194,107,0,0,8,24,4,10,11,5,41,51,0,142,24,28] - 246
[2024-06-26 14:31:28] [34mdebug[39m: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":2820,"srcaddr":41166,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":43,"securityuse":0,"timestamp":7062092,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,4,10,11,5,41,51,0]}}
[2024-06-26 14:31:28] [34mdebug[39m: 	zh:controller: Received payload: clusterID=2820, address=41166, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=43, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":4,"commandIdentifier":10},"payload":[{"attrId":1291,"dataType":41,"attrData":51}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-06-26 14:31:28] [34mdebug[39m: 	zh:zstack:unpi:parser: --- parseNext []
[2

Koenkk, 2024-06-27:
I do not see any `Received MQTT message` when you control the device via HA. Can you confirm this? E.g. control the device via HA and check the log if any `Received MQTT message` appears there?

bjornsivertsen, 2024-06-27:
I need to test further, I had the following issue:

1. If I updated Z2M to 1.38.0, then I could not control lamps from HA either with my Ember based Skyconnect. I could still control the lamps from Z2Ms interface. Logs didn't show Z2M receiving any messages from HA.
2. If I downgraded to 1.37.1 again, then things worked as normal.
3. Now if I upgrade to 1.38.0 and then restart the Mosquitto service, things appear to work as normal again.
4. If I then restart the Z2M container, I get back to step 1. Things can't be controlled from HA again, atleast until I manually restart the Mosquitto container.

floyd871, 2024-06-28:
[zig.zip](https://github.com/user-attachments/files/16025511/zig.zip)

I send you both Debug-Logs from 1.37.1 and 1.38.0.
Watch out Device "Studio Licht"...

Koenkk, 2024-06-28:
@floyd871 it looks that you've set the z2m `base_topic` to `homeassistant`, can you try changing it to `zigbee2mqtt`?

floyd871, 2024-06-30:
@Koenkk Yes, that do the job. It's working again (1.38.0). At least for me. I didn't know why i've set the base_topic to "homeassistant". Mayby through some installation Videos on youtube. THX

Koenkk, 2024-06-30:
I've now added a check for this, Z2M will refuse when the `mqtt.base_topic` is equal to the `homeassistant.discovery_topic`.

closed 2024-06-30 by Koenkk

krzyk, 2024-12-10:
@Koenkk What's the difference between homeassistant.discovery_topic and mqtt.base_topic?

What is the discovery_topic, do I need to configure it in homeassistant? I have homeassistant configured with base_topic only.

Docs are not that clear and why I can't have those equal, it breaks something, but this change breaks a lot of current installs which had base_topic set to homeassistant.