Koenkk/zigbee2mqtt #23113: [New device support]: Support for this tuya garage door opener

rmupfumira, 2024-06-20:
### Link

https://www.aliexpress.com/item/1005006792508447.html?spm=a2g0o.order_list.order_list_main.17.51e918029N6maj

### Database entry

{"id":7,"type":"Router","ieeeAddr":"0xa4c13821593c2d35","nwkAddr":28764,"manufId":4417,"manufName":"_TZE20C_qw6nyd2u","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,4,5,6,8,768,4096,1280,61184],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x804b50fffea7eea1","zoneState":0}},"lightingColorCtrl":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":81,"stackVersion":1,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1718873498825}

### Comments

I tried to follow the add support guidelines but the device keeps being recognized as a light

### External definition

```shell
const {identify, light, iasZoneAlarm} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE20C_qw6nyd2u',
    description: 'Automatically generated definition',
    extend: [identify(), light({"colorTemp":{"range":[150,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}}), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]})],
    meta: {},
};

module.exports = definition;
```

halldark, 2024-12-11:
looks like https://www.zigbee2mqtt.io/devices/TS0603.html, but report as:
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE20C_qw6nyd2u',

closed 2025-04-04 by github-actions[bot]

albaphysic, 2026-10-28:
Thanks for that @privedmidved-gif 

This is what I used: 
```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const ea = exposes.access;

const trigger = {
    key: ['trigger'],
    convertSet: async (entity, key, value, meta) => {
        // Always send the relay's ON command; never send OFF.
        await tuya.sendDataPointBool(entity, 1, true);
    },
};

const definition = {
    fingerprint: [{
        modelID: 'TS0601',
        manufacturerName: '_TZE20C_zka46xbw',
    }],
    model: 'TS0601_garage_zka46xbw',
    vendor: 'Tuya',
    description: 'Garage door opener',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [trigger],
    onEvent: async (type, data, device, settings, state) => {
        await tuya.onEventSetTime(type, data, device, settings, state);
    },
    configure: tuya.configureMagicPacket,
    exposes: [
        exposes.binary('trigger', ea.SET, 'ON', 'OFF')
            .withDescription('Momentarily trigger the garage-door relay'),
        exposes.binary('contact', ea.STATE, false, true)
            .withDescription('Gate contact: false = open, true = closed'),
        exposes.enum('alarm', ea.STATE, ['none', 'unknown_1', 'contact_closed'])
            .withDescription('Alarm status'),
    ],
    meta: {
        tuyaDatapoints: [
            [3, 'contact', tuya.valueConverterBasic.lookup({'true': false, 'false': true})],
            [12, 'alarm', tuya.valueConverterBasic.lookup({
                'none': 0,
                'unknown_1': 1,
                'contact_closed': 2,
            })],
        ],
    },
};

module.exports = definition;
```

Note that mine is  `manufacturerName: '_TZE20C_zka46xbw` while yours is `manufacturerName: '_TZE20C_qw6nyd2u`, so change accordingly. 

I then added a Home Assistant automation script that I use as stateless button (not a toggle switch) and it works without hiccups.

In HA: 
Settings > Automations & scenes > Scripts
Add action > search for `MQTT: Publish`
Set Topic to: `zigbee2mqtt/YOUR_DEVICE_NAME/set`
Set Payload to: {"trigger":"ON"}
Save

Use Run to test if it opens and closes the door

On your HA dashboard, add a button card, leave Entity empty.
Interactions>Tap behavior>Perform Action>Action:Your script

Now it's a stateless "momentary" button on your HA dash and for me it works as one tap: open, one tap: close. Also one tap: stop if it's in the middle of movement.