Koenkk/zigbee2mqtt #23120: SLZB-06M, ember driver, z2m doesn't start

Herreropl, 2024-06-21:
### What happened?

I decided to switch to ember driver on my SLZB-06M. I flashed firmware via OTA (20240510 ). Unfortunately Zigbee2MQTT fails to start... I have checked different baud rates but without success. With ezsp driver works, but with errors.

current config:

port: tcp://10.10.10.245:6638
baudrate: 115200
adapter: ember
rtscts: false

### Zigbee2MQTT version

1.38.0-1

### Adapter firmware version

7.4.1.0 build 0

### Adapter

ember

### Setup

Add-on on Home Assistant OS on RPI4

### Debug log

[10:09:27] INFO: Preparing to start...
[10:09:28] INFO: Socat not enabled
[10:09:30] INFO: Starting Zigbee2MQTT...
[2024-06-21 10:09:36] error: 	zh:ember:uart:ash: Received frame with CRC error
[2024-06-21 10:09:36] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-06-21 10:09:36] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-06-21 10:09:36] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-06-21 10:09:38] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2024-06-21 10:09:38] error: 	zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
[2024-06-21 10:09:38] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | NCP status: ASH_NCP_FATAL_ERROR
[2024-06-21 10:09:38] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-06-21 10:09:38] error: 	zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!
[2024-06-21 10:14:37] error: 	z2m:mqtt: Not connected to MQTT server!
[2024-06-21 10:14:37] error: 	z2m:mqtt: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: '{"state":"offline"}
[10:14:38] INFO: Preparing to start...
[10:14:39] INFO: Socat not enabled
[10:14:41] INFO: Starting Zigbee2MQTT...
[2024-06-21 10:14:47] warning: 	zh:ezsp: Deprecated driver 'ezsp' currently in use, 'ember' will become the officially supported EmberZNet driver in next release. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462
[2024-06-21 10:14:47] error: 	zh:ezsp:uart: <-- CRC error: 5593499cc3af15c6d19874f0cf3488fce23ea5ebc9de6fa220c3e6e27e|e6e2|8816
[2024-06-21 10:14:47] error: 	zh:ezsp:uart: Error while parsing to NpiFrame 'Error: <-- CRC error: 5593499cc3af15c6d19874f0cf3488fce23ea5ebc9de6fa220c3e6e27e|e6e2|8816
    at Frame.checkCRC (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/frame.ts:87:19)
    at SerialDriver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:169:19)
    at Parser.emit (node:events:517:28)
    at Parser._transform (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/parser.ts:49:26)
    at Parser.Transform._write (node:internal/streams/transform:175:8)
    at writeOrBuffer (node:internal/streams/writable:392:12)
    at _write (node:internal/streams/writable:333:10)
    at Parser.Writable.write (node:internal/streams/writable:337:10)
    at Socket.ondata (node:internal/streams/readable:809:22)
    at Socket.emit (node:events:517:28)'

Nerivec, 2024-06-22:
Can you solve the MQTT issue first (`Not connected to MQTT server!`), then provide a `debug` log?

Herreropl, 2024-06-22:
I think the MQTT problem message is related to fact, that z2m not starting correctly. When the adapter is set to ember frontent it also does not work (bad gateway). After switching to ezsp, everything works (both frontend and mqtt).

[log.log](https://github.com/user-attachments/files/15935854/log.log)

robotsmarthomes, 2024-06-22:
Following- similar issue

Nerivec, 2024-06-22:
Can you confirm what devices are these (link to the docs pages would be great):
- 3662 (`0x0E4E`)
- 7979 (`0x1F2B`)

From your log, these two sent nearly 1500 messages in under 2 minutes, before Z2M even finished starting properly. Something's definitely wrong/bad there.
`ember` starts fine, but these two seem to mess with the end of the Z2M starting process with their spamming.
Can you try powering these two off (unplug, remove battery), then starting Z2M, to confirm?

Herreropl, 2024-06-22:
here you are :)

those two devices are electricity meters https://www.zigbee2mqtt.io/devices/SPM01.html#tuya-spm01

I have disabled them and got logs. The same symptoms :/ both frontend and zigbee network didn't work...

[log.log](https://github.com/user-attachments/files/15936636/log.log)

Herreropl, 2024-06-22:
ok, I have solved problem :) In my case if log level is set to warning and in the same time adapter is set to ember z2m doesn't start (bad gateway). I case of ezsp frontend works. The solution is to set log level to warn. In both cases frontent works. It's very strange...

Nerivec, 2024-06-23:
Did you modify anything else? Since you provided a log with `debug`, it seems strange that log level `warning`/`warn` would be the problem.

Herreropl, 2024-06-23:
ok, you are right log level is dead end. I restarted addon and again both frontend and z2m didn't work. 
So, I decided to start form scratch - what I did:
1. remove addon
2. remove zigbee2mqtt folder
3. install addon (ember driver) and repair above thirty devices
4. restart addon and again (bad gateway). 
5. a lot of test with different log levels but without success
6. switch to ezsp and everything works without errors
[log-warn(ember).log](https://github.com/user-attachments/files/15946476/log-warn.ember.log)
[log-info(ember).log](https://github.com/user-attachments/files/15946490/log-info.ember.log)
[log-debug(ember).log](https://github.com/user-attachments/files/15946499/log-debug.ember.log)

mmokrejs, 2024-06-25:
I had same issues when switching from `ezsp` to `ember`. It appears HomeAssistant caches the settings so it did NOT reflect that I edited the `configuration.yaml` manually via ssh+vim. I also had to delete the json and database files in that folder, leaving just the `configuration.yaml` and reboot.

My impression from the logs was that one cannot switch properly from one driver to the other. What really s*cks that the config values shown via HA web GUI do not reflect the actual content of the `/config/zigbee2mqtt_"$somenumber"/configuration.yaml`.

Nerivec, 2024-06-25:
@mmokrejs If the file and the UI are "synced", it shouldn't be a problem, but yeah, some operations break that link and the UI no longer reflects the content properly (like switching from regular to edge or vice versa). Otherwise, switching from one driver to the other is just a matter of changing the adapter config.
I don't think this is the same issue though, as @Herreropl redid the whole setup and still has the problem.

@Herreropl Seems to be indeed something with those devices that start spamming very early (`0xF738` this time) and Z2M not finishing its startup sequence for some reason.
@Koenkk any ideas? It seems to be stopping right after Herdsman is done (like the `start` function of the adapter is never actually returning even though it reached the end...).
It stops after [this](https://github.com/Koenkk/zigbee-herdsman/blob/99ee8f494d2c8d49040d8aa354772a3c16fa79c0/src/adapter/ember/adapter/emberAdapter.ts#L926), so, basically, [here](https://github.com/Koenkk/zigbee-herdsman/blob/99ee8f494d2c8d49040d8aa354772a3c16fa79c0/src/adapter/ember/adapter/emberAdapter.ts#L2724) but never gets [here](https://github.com/Koenkk/zigbee-herdsman/blob/99ee8f494d2c8d49040d8aa354772a3c16fa79c0/src/controller/controller.ts#L130), it never does the rest of the logic, so it eventually fails as soon as something calls code that hasn't been setup (MQTT here).
```log
[2024-06-24 00:01:21] info: 	zh:ember:queue: Request dispatching started.
...
[2024-06-24 00:02:29] error: 	z2m:mqtt: Not connected to MQTT server!
[2024-06-24 00:02:29] error: 	z2m:mqtt: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
```

Nerivec, 2024-06-26:
@voc0der Looks like your issue might be related to this one. Any chance you can provide a log with `debug` to confirm after what step it causes trouble?

voc0der, 2024-06-26:
@Nerivec thanks for getting me to the right thread. Here it is.

```
Using '/app/data' as data directory
[2024-06-26 11:13:38] info:     z2m: Logging to console, file (filename: log.log)
[2024-06-26 11:13:38] debug:    z2m: Loaded state from file /app/data/state.json
[2024-06-26 11:13:38] info:     z2m: Starting Zigbee2MQTT version 1.38.0 (commit #fe048e6)
[2024-06-26 11:13:38] info:     z2m: Starting zigbee-herdsman (0.49.2)
[2024-06-26 11:13:38] debug:    z2m: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[25],"extendedPanID":[222,158,171,2,134,204,14,1],"networkKey":"HIDDEN","panID":28602},"serialPort":{"adapter":"ember","path":"/dev/ttyACM0","rtscts":true}}'
[2024-06-26 11:13:38] info:     zh:ember: Using default stack config.
[2024-06-26 11:13:38] debug:    zh:ember: Using delay=5.
[2024-06-26 11:13:38] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[221,75,142,123,205,153,202,191,222,203,211,151,148,58,233,216],"panID":28602,"extendedPanID":[222,158,171,2,134,204,14,1],"channelList":[25]},"serialPort":{"rtscts":true,"path":"/dev/ttyACM0","adapter":"ember"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
[2024-06-26 11:13:38] info:     zh:ember: ======== Ember Adapter Starting ========
[2024-06-26 11:13:38] debug:    zh:ember: Starting EZSP with stack configuration: "[object Object]".
[2024-06-26 11:13:38] info:     zh:ember:ezsp: ======== EZSP starting ========
[2024-06-26 11:13:38] info:     zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-26 11:13:38] debug:    zh:ember:uart:ash: Opening serial port with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":true,"autoOpen":false,"parity":"none","stopBits":1,"xon":false,"xoff":false}
[2024-06-26 11:13:38] info:     zh:ember:uart:ash: Serial port opened
[2024-06-26 11:13:38] info:     zh:ember:uart:ash: ======== ASH starting ========
[2024-06-26 11:13:38] debug:    zh:ember:uart:ash: ---> [FRAME type=RST]
[2024-06-26 11:13:38] debug:    zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2024-06-26 11:13:38] debug:    zh:ember:uart:ash: Waiting for RSTACK... 100/2500
[2024-06-26 11:13:38] debug:    zh:ember:uart:ash: Waiting for RSTACK... 200/2500
[2024-06-26 11:13:38] debug:    zh:ember:uart:ash: Waiting for RSTACK... 300/2500
[2024-06-26 11:13:39] debug:    zh:ember:uart:ash: <--- [FRAME type=ERROR]
[2024-06-26 11:13:39] error:    zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=RESET_WATCHDOG.
[2024-06-26 11:13:39] error:    zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-06-26 11:13:39] error:    zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-06-26 11:13:39] info:     zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-26 11:13:39] info:     zh:ember:uart:ash: ======== ASH starting ========
[2024-06-26 11:13:39] debug:    zh:ember:uart:ash: ---> [FRAME type=RST]
[2024-06-26 11:13:39] debug:    zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2024-06-26 11:13:39] debug:    zh:ember:uart:ash: Waiting for RSTACK... 100/2500
[2024-06-26 11:13:39] debug:    zh:ember:uart:ash: Waiting for RSTACK... 200/2500
[2024-06-26 11:13:39] debug:    zh:ember:uart:ash: Waiting for RSTACK... 300/2500
[2024-06-26 11:13:39] debug:    zh:ember:uart:ash: Waiting for RSTACK... 400/2500
[2024-06-26 11:13:39] debug:    zh:ember:uart:ash: Waiting for RSTACK... 500/2500
[2024-06-26 11:13:39] debug:    zh:ember:uart:ash: Waiting for RSTACK... 600/2500
[2024-06-26 11:13:39] debug:    zh:ember:uart:ash: Waiting for RSTACK... 700/2500
[2024-06-26 11:13:39] debug:    zh:ember:uart:ash: Waiting for RSTACK... 800/2500
[2024-06-26 11:13:39] debug:    zh:ember:uart:ash: Waiting for RSTACK... 900/2500
[2024-06-26 11:13:40] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1000/2500
[2024-06-26 11:13:40] debug:    zh:ember:uart:ash: <--- [FRAME type=RSTACK]
[2024-06-26 11:13:40] info:     zh:ember:uart:ash: ======== ASH connected ========
[2024-06-26 11:13:40] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0]
[2024-06-26 11:13:40] info:     zh:ember:uart:ash: ======== ASH started ========
[2024-06-26 11:13:40] info:     zh:ember:ezsp: ======== EZSP started ========
[2024-06-26 11:13:40] debug:    zh:ember:ezsp: ===> [FRAME: ID=0:"VERSION" Seq=0 Len=4]
[2024-06-26 11:13:40] debug:    zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=0 frmRx=0]
[2024-06-26 11:13:41] warning:  zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2024-06-26 11:13:41] debug:    zh:ember:uart:ash: <--- [FRAME type=RSTACK]
[2024-06-26 11:13:41] error:    zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
[2024-06-26 11:13:41] error:    zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | NCP status: ASH_NCP_FATAL_ERROR
[2024-06-26 11:13:41] error:    zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-06-26 11:13:41] error:    zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!
[... cut ...]
[2024-06-26 11:13:43] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0]
[2024-06-26 11:13:43] debug:    zh:ember:ezsp: <=== [FRAME: ID=38:"GET_EUI64" Seq=46 Len=13]
[2024-06-26 11:13:43] debug:    zh:ember: [INIT] Network Ready! {"eui64":"0xbc026efffe23387d","parameters":{"extendedPanId":[222,158,171,2,134,204,14,1],"panId":28602,"radioTxPower":5,"radioChannel":25,"joinMethod":0,"nwkManagerId":0,"nwkUpdateId":0,"channels":134215680},"status":2}
[2024-06-26 11:13:43] info:     zh:ember:queue: Request dispatching started.
```

Nerivec, 2024-06-26:
Looks the same indeed (the retries on startup are just the hardware flow control issues -known to silabs-, only way to get rid of them entirely at the moment is to flash a [firmware without hardware flow control](https://github.com/darkxst/silabs-firmware-builder/tree/ember-nohw/firmware_builds) -and use `rtscts: false`-, but it shouldn't be an issue once it starts after that).
What is in-between `zh:ember:queue: Request dispatching started.` and `z2m:mqtt: Not connected to MQTT server!`?

voc0der, 2024-06-26:
This segment repeats. I never see the Not connected (I waited 2 minutes, which is as much as I can atm, I'll try more later).

```
[2024-06-26 13:23:49] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-26 13:23:49] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7]
[2024-06-26 13:23:49] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7 frmNum=0] Added to rxQueue
[2024-06-26 13:23:49] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1]
[2024-06-26 13:23:49] debug:    zh:ember:ezsp: <=== [FRAME: ID=89:"INCOMING_ROUTE_RECORD_HANDLER" Seq=46 Len=18]
[2024-06-26 13:23:49] debug:    zh:ember:ezsp: ezspIncomingRouteRecordHandler(): callback called with: [source=63928], [sourceEui=0x001788010db0d593], [lastHopLqi=140], [lastHopRssi=-65], [relayCount=0], [relayList=]
[2024-06-26 13:23:49] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-26 13:23:49] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7]
[2024-06-26 13:23:49] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7 frmNum=1] Added to rxQueue
[2024-06-26 13:23:49] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2]
[2024-06-26 13:23:49] debug:    zh:ember:ezsp: <=== [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=46 Len=37]
[2024-06-26 13:23:49] debug:    zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":11,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":152}], [lastHopLqi=140], [lastHopRssi=-65], [sender=63928], [bindingIndex=255], [addressIndex=255], [messageContents=01ec01000b10140102240001]
[2024-06-26 13:23:59] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-26 13:23:59] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7]
[2024-06-26 13:23:59] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7 frmNum=2] Added to rxQueue
[2024-06-26 13:23:59] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3]
[2024-06-26 13:23:59] debug:    zh:ember:ezsp: <=== [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=46 Len=37]
[2024-06-26 13:23:59] debug:    zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":11,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":153}], [lastHopLqi=140], [lastHopRssi=-65], [sender=63928], [bindingIndex=255], [addressIndex=255], [messageContents=01ed01000b10140102240001]
[2024-06-26 13:24:09] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-26 13:24:09] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7]
[2024-06-26 13:24:09] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7 frmNum=3] Added to rxQueue
[2024-06-26 13:24:09] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4]
[2024-06-26 13:24:09] debug:    zh:ember:ezsp: <=== [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=46 Len=37]
[2024-06-26 13:24:09] debug:    zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":11,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":154}], [lastHopLqi=140], [lastHopRssi=-65], [sender=63928], [bindingIndex=255], [addressIndex=255], [messageContents=01ee01000b10140102240001]
[2024-06-26 13:24:19] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-26 13:24:19] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7]
[2024-06-26 13:24:19] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7 frmNum=4] Added to rxQueue
[2024-06-26 13:24:19] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5]
[2024-06-26 13:24:19] debug:    zh:ember:ezsp: <=== [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=46 Len=37]
[2024-06-26 13:24:19] debug:    zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":11,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":155}], [lastHopLqi=140], [lastHopRssi=-65], [sender=63928], [bindingIndex=255], [addressIndex=255], [messageContents=01ef01000b10140102240001]
[2024-06-26 13:24:21] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-26 13:24:21] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7]
[2024-06-26 13:24:21] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7 frmNum=5] Added to rxQueue
[2024-06-26 13:24:21] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=6]
[2024-06-26 13:24:21] debug:    zh:ember:ezsp: <=== [FRAME: ID=89:"INCOMING_ROUTE_RECORD_HANDLER" Seq=46 Len=18]
[2024-06-26 13:24:21] debug:    zh:ember:ezsp: ezspIncomingRouteRecordHandler(): callback called with: [source=35006], [sourceEui=0x001788010db0d802], [lastHopLqi=236], [lastHopRssi=-41], [relayCount=0], [relayList=]
[2024-06-26 13:24:21] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-26 13:24:21] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7]
[2024-06-26 13:24:21] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7 frmNum=6] Added to rxQueue
[2024-06-26 13:24:21] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7]
[2024-06-26 13:24:21] debug:    zh:ember:ezsp: <=== [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=46 Len=37]
[2024-06-26 13:24:21] debug:    zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":11,"destinationEndpoint":1,"options":256," [... cut ...]
[2024-06-26 13:25:01] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4]
[2024-06-26 13:25:01] debug:    zh:ember:ezsp: <=== [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=46 Len=37]
[2024-06-26 13:25:01] debug:    zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":11,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":16}], [lastHopLqi=236], [lastHopRssi=-41], [sender=35006], [bindingIndex=255], [addressIndex=255], [messageContents=017401000b10140102240001]
```

Herreropl, 2024-06-26:
I think that the message "z2m:mqtt: Not connected to MQTT server!" appears when the addon is going to shutdown or restart. It seems that even frontend is offline something under the hood is working.

Nerivec, 2024-06-26:
Can either of you try the dev branch/edge, see if you get better results?

voc0der, 2024-06-26:
Same exact results with `latest-dev`, didn't even make it up on the first `docker compose up -d`

Herreropl, 2024-06-28:
> ed (the retries on startup are just the hardware flow control issues -known to silabs-, only way to get rid of them entirely at

I tried this no_hw firmware. After flashing first boot was ok, but after restart the same problem as before :/

`[09:35:23] INFO: Preparing to start...
[09:35:24] INFO: Socat not enabled
[09:35:26] INFO: Starting Zigbee2MQTT...
[2024-06-28 09:35:34] warning: 	zh:ember: [EzspConfigId] Failed to SET "APS_UNICAST_MESSAGE_COUNT" TO "32" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-06-28 09:35:34] warning: 	z2m: `permit_join` set to  `true` in configuration.yaml.
[2024-06-28 09:35:34] warning: 	z2m: Allowing new devices to join.
[2024-06-28 09:35:34] warning: 	z2m: Set `permit_join` to `false` once you joined all devices.
[09:36:19] INFO: Preparing to start...
[09:36:20] INFO: Socat not enabled
[09:36:22] INFO: Starting Zigbee2MQTT...
[2024-06-28 09:36:29] error: 	zh:ember:uart:ash: Received frame with CRC error
[2024-06-28 09:36:29] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-06-28 09:36:29] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-06-28 09:36:29] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-06-28 09:36:31] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2024-06-28 09:36:31] error: 	zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
[2024-06-28 09:36:31] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | NCP status: ASH_NCP_FATAL_ERROR
[2024-06-28 09:36:31] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-06-28 09:36:31] error: 	zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!
[2024-06-28 09:36:33] warning: 	zh:ember: [EzspConfigId] Failed to SET "APS_UNICAST_MESSAGE_COUNT" TO "32" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-06-28 09:37:42] error: 	z2m:mqtt: Not connected to MQTT server!
[2024-06-28 09:37:42] error: 	z2m:mqtt: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
[09:37:44] INFO: Preparing to start...
[09:37:44] INFO: Socat not enabled
[09:37:46] INFO: Starting Zigbee2MQTT...
[2024-06-28 09:37:52] error: 	zh:ember:uart:ash: Received frame with CRC error
[2024-06-28 09:37:52] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-06-28 09:37:52] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-06-28 09:37:52] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-06-28 09:37:55] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2024-06-28 09:37:55] error: 	zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
[2024-06-28 09:37:55] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | NCP status: ASH_NCP_FATAL_ERROR
[2024-06-28 09:37:55] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-06-28 09:37:55] error: 	zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!
[2024-06-28 09:37:56] warning: 	zh:ember: [EzspConfigId] Failed to SET "APS_UNICAST_MESSAGE_COUNT" TO "32" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.`

MilesTEG1, 2024-06-28:
Following- similar issue

Nerivec, 2024-06-28:
The no-hw was only for SkyConnect.
The 06-m doesn't have hardware flow control, so firmware is the same as you had before.
It is unusual to have such a messy startup sequence on it though. Are you on the latest "core firmware" (smlight one, 2.x.x)? They released a lot of fixes recently.
This won't fix the ultimate issue here though, just maybe a cleaner initial startup.

@voc0der How familiar are you with docker? Any chance you could edit a few files so we can figure out where exactly the darn thing is stopping?

mmokrejs, 2024-06-28:
I am not the oroginal reporter but as a side note, I received yesterday from SMLIGHT current dev OTA update 2.4.3_dev applicable to both SLZB-06 and -06M. I still see:

```
[12:31:00] INFO: Preparing to start...
[12:31:01] INFO: Socat not enabled
[12:31:06] INFO: Starting Zigbee2MQTT...
[2024-06-28 12:31:16] info: 	z2m: Logging to console, file (filename: log.log)
[2024-06-28 12:31:16] info: 	z2m: Starting Zigbee2MQTT version 1.38.0 (commit #unknown)
[2024-06-28 12:31:16] info: 	z2m: Starting zigbee-herdsman (0.49.2)
[2024-06-28 12:31:17] info: 	zh:ember: Using default stack config.
[2024-06-28 12:31:17] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-06-28 12:31:17] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-06-28 12:31:17] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-28 12:31:17] info: 	zh:ember:uart:ash: Socket ready
[2024-06-28 12:31:17] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-06-28 12:31:17] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-06-28 12:31:17] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-06-28 12:31:17] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-06-28 12:31:17] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-28 12:31:17] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-06-28 12:31:18] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-06-28 12:31:18] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-06-28 12:31:18] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-06-28 12:31:19] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2024-06-28 12:31:19] error: 	zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
[2024-06-28 12:31:19] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | NCP status: ASH_NCP_FATAL_ERROR
[2024-06-28 12:31:19] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-06-28 12:31:19] error: 	zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!
[2024-06-28 12:31:19] info: 	zh:ember:queue: Request dispatching stopped; queue=0 priorityQueue=0
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   Total frames: RX=2, TX=3
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   Cancelled   : RX=1, TX=0
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   DATA frames : RX=0, TX=1
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   DATA bytes  : RX=0, TX=4
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=1
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2024-06-28 12:31:19] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2024-06-28 12:31:19] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2024-06-28 12:31:19] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2024-06-28 12:31:20] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-06-28 12:31:20] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-06-28 12:31:20] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-28 12:31:20] info: 	zh:ember:uart:ash: Socket ready
[2024-06-28 12:31:20] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-06-28 12:31:21] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-06-28 12:31:21] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-06-28 12:31:21] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-06-28 12:31:21] info: 	zh:ember: [STACK STATUS] Network up.
[2024-06-28 12:31:21] info: 	zh:ember: [INIT TC] NCP network matches config.
[2024-06-28 12:31:21] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1247ms until next broadcast.
[2024-06-28 12:31:21] info: 	zh:ember:queue: Request dispatching started.
```

```
[08:38:01] setup | Starting firmware: v2.4.2.dev
[08:38:02] ConfigHelper | LittleFS mounted
[08:38:02] ConfigHelper | load config
[... 18 lines cut ...]
[08:38:11] Network | fireNetworkUp
[08:38:11] taskZB | Start task
[08:38:11] taskZB | Starting server on port: 6638
[08:38:11] Web | Webserver started
[08:38:11] time | time sync start
[08:38:11] internet | connected
[10:38:11] time | timezone: CET-1CEST,M3.5.0,M10.5.0/3
[... cut ...]
[10:38:12] Network | [POST] response: New record [on] created successfullyNew record [lo] created successfully
[10:38:12] stats | Statistics sent
[10:38:12] stats | Heap: 2384
[10:38:13] [ZBCHK] | Wrong answer
[10:38:13] [ZBVER] | Unknown
[10:38:14] [ZBCHK] | done
[12:30:54] taskZB | Client disconnected, id: 0
[12:31:17] taskZB | New client: 192.168.x.x id: 0
[12:31:19] taskZB | Client disconnected, id: 0
[12:31:20] taskZB | New client: 192.168.x.x id: 0
```

Note the `ZBCHK` and `ZBVER` commands failing.

voc0der, 2024-06-28:
@Nerivec  I'm pretty comfortable around docker/build, so if you have anything in mind I am open to trying.

Nerivec, 2024-06-28:
@voc0der Any chance you can find me on Discord (can DM me from zigbee2mqtt server)? It'll be easier to share stuff without polluting this thread too much (weird appears to only be getting weirder... 😅).

Nerivec, 2024-06-28:
@robotsmarthomes and @MilesTEG1 can you confirm your adapter type/model, and setup for z2m and if it's exactly the same, or just similar?

MilesTEG1, 2024-06-28:
> @robotsmarthomes and @MilesTEG1 can you confirm your adapter type/model, and setup for z2m and if it's exactly the same, or just similar?

Hello 👋🏻 
I posted [this issue](https://github.com/Koenkk/zigbee2mqtt/issues/23198) some minutes ago 😊
I have a SMLIGHT SLZB-06M, my Z2M is an addon inside HAOS witch runs as a proxmox VM.

fir3drag0n, 2024-06-28:
As user of the SLZB-06M I can also confirm that I need to start at least twice the z2m container in order to properly use it. Especially due to updates of the containers, it regularly occurs.

robotsmarthomes, 2024-06-29:
Hello,

Apologies I missed the notification earlier

I am currently running 2x SLZB-06M’s on two different zigbee2mqtt networks
using the same raspberry pi4.

Latest core firmware 2305
Latest Zigbee firmware - ember

Each network in the home has about 30x Zigbee Relays and also has 6x mmWave
human presence sensors (very chatty in the logs)

I am unable to start the zigbee2mqtt without deleting the
backup_config.json file.

It starts though but I’m finding this ember very slow and unresponsive at
times.

Ra72xx, 2024-07-02:
Following because of #22249, which also includes failures at starting with "ember" for me.

MilesTEG1, 2024-07-03:
Hello here,
I fixed my problem with error `ROUTE_ERROR_ADDRESS_CONFLICT`.
But I still have an issue with startup of Z2M: it crashes and I need to reboot one time at least to get it functionnal.
Here the log for the first book, the failed one:
```log
[2024-07-03 23:29:16] info: 	z2m: Logging to console, file (filename: log.log)
[2024-07-03 23:29:16] info: 	z2m: Starting Zigbee2MQTT version 1.39.0 (commit #unknown)
[2024-07-03 23:29:16] info: 	z2m: Starting zigbee-herdsman (0.50.1)
[2024-07-03 23:29:16] info: 	zh:ember: Using default stack config.
[2024-07-03 23:29:16] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-07-03 23:29:16] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-07-03 23:29:16] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-07-03 23:29:16] info: 	zh:ember:uart:ash: Socket ready
[2024-07-03 23:29:16] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-07-03 23:29:16] error: 	zh:ember:uart:ash: Received frame with CRC error
[2024-07-03 23:29:16] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-07-03 23:29:16] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-03 23:29:16] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-07-03 23:29:17] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-07-03 23:29:17] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-07-03 23:29:17] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-07-03 23:29:18] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-07-03 23:29:18] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-07-03 23:29:19] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2024-07-03 23:29:19] error: 	zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
[2024-07-03 23:29:19] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-03 23:29:19] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-07-03 23:29:19] error: 	zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!
[2024-07-03 23:29:19] info: 	zh:ember:queue: Request dispatching stopped; queue=0 priorityQueue=0
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   Total frames: RX=2, TX=3
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   Cancelled   : RX=1, TX=0
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   DATA frames : RX=0, TX=1
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   DATA bytes  : RX=0, TX=4
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=1
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2024-07-03 23:29:19] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2024-07-03 23:29:19] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2024-07-03 23:29:19] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-07-03 23:29:19] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash: Socket ready
[2024-07-03 23:29:19] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-07-03 23:29:20] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-07-03 23:29:20] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-07-03 23:29:20] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-07-03 23:29:21] info: 	zh:ember: [STACK STATUS] Network up.
[2024-07-03 23:29:21] info: 	zh:ember: [INIT TC] NCP network matches config.
[2024-07-03 23:29:21] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[2024-07-03 23:29:21] info: 	zh:ember:queue: Request dispatching started.
```

andriikut, 2024-07-09:
Has almost identical issue above

знову

[2024-07-09 22:25:55] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-07-09 22:25:55] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-09 22:25:55] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-07-09 22:25:55] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-07-09 22:25:55] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-07-09 22:25:55] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-07-09 22:25:55] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-07-09 22:25:56] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-07-09 22:25:56] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-07-09 22:25:56] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-07-09 22:25:57] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2024-07-09 22:25:57] error: 	zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
[2024-07-09 22:25:57] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-09 22:25:57] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-07-09 22:25:57] error: 	zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!

Switched to  "adapter: ezsp" from ember and looks like it works with latest firmware (got warnings regarding migration to ember driver) [2024-07-09 22:38:28] warning: 	zh:ezsp: Deprecated driver 'ezsp' currently in use, 'ember' will become the officially supported EmberZNet driver in next release.
but works so far

Nerivec, 2024-07-09:
@andriikut These errors should be bypassed and allow Z2M to start regardless. That is unless you are affected by the issue where Z2M freezes after `zh:ember:queue: Request dispatching started.`
Was this on slzb-06m too? If yes, what core firmware version, and do you have the logs from the SMLight interface (ESP side)? I'll pass them on.

Can anyone still affected by the freeze issue try with latest dev/edge and report back?

fir3drag0n, 2024-07-10:
I can confirm the problem with latest dev

Kuechenfenster, 2024-07-17:
is there any update on this matter?

I running 2 SLZB-06M sticks, 1 works perfectly and the 2nd one reports the above errors with not starting the Z2M network;
the setup of the 3rd stick is currently on hold as I need the 2nd one to work first.

Nerivec, 2024-07-17:
@Kuechenfenster Can you make a list of the possible differences between the setup that is working, and the one that isn't; hardware and software? And provide log samples from both?

tandarra, 2024-07-28:
Jumping on as I have been having issues and looking for a solution. SLZB-06M with Ember struggles to start, Mosquitto is running but Z2MQTT says 502 bad gateway.
Going directly to SLZB-06M IP I can see on the configuration page its connected to Z2MQTT. But no luck going to the HA UI page.

Changing the configuration back to ezsp seems to be smoothless. Next time I'll take note of the logs and share.

Herreropl, 2024-07-28:
> Jumping on as I have been having issues and looking for a solution. SLZB-06M with Ember struggles to start, Mosquitto is running but Z2MQTT says 502 bad gateway. Going directly to SLZB-06M IP I can see on the configuration page its connected to Z2MQTT. But no luck going to the HA UI page.
> 
> Changing the configuration back to ezsp seems to be smoothless.
> 
> Unsure what it is, but two log files [Ember - log.log](https://github.com/user-attachments/files/16403899/Ember.-.log.log) [working ezsp log.log](https://github.com/user-attachments/files/16403900/working.ezsp.log.log)

I have still the same problem. Temporary solution in few steps:
1. Stop z2m addon.
2. Change configuration to ember. Don't restart addon.
3. Restart SLZB-06M
4. Take some time (couple second to start adapter) - check if it started and it is ready.
5. Start z2m addon.

It will work till next restart of HA 🙂

Nerivec, 2024-07-28:
@tandarra Problem looks to be related to spamming on start (common thread I see in the logs here), in your case, between 23:53:10 and 00:01:00, I see nearly 6000 incoming messages... from devices: 57425 (0xe051), 47113 (0xb809), 41954 (0xa3e2) about a third each (most seem to be from the Tuya manufacturer-specific cluster). All that before Z2M is done starting.
Your ezsp logs don't have the startup sequence to compare though. Any chance you have the rest?

TDCroPower, 2024-08-23:
My Setup…
HA: 2024.8.2 (Docker Container)
z2m: 1.39.1 (Docker Container)
mosquitto: 2.0.18 (Docker Container)
SLZB-06M…
Core: v2.5.0
ZigBee: 20240510
Adapter: ember

I have the same problem as described here.
If I restart the z2m container in
running, it does not come up successfully.
Only when I restart the 06M stick does z2m start up successfully.

if I can test something please let me know, I have access to all files and configs.

rrakso, 2024-08-29:
I'm having similar problems #23761, but with `Sonoff ZBDongle-E`, and my z2m is working fine, but recently it is stopping working itself.. :c

DorjeDorf, 2024-09-17:
+1 for this problem for me with Z2M and SLZB-06M. I had to downgrade a lot and go back to ezsp to get the network to work (altough I often get timeouts and need to restart Z2M and the SLZB to get everything up and running again). 

The most stable setup for now is: 
v2.3.6 for the core
20240330 for the coordinator. 
Anything newer than that causes issues with Z2M (here, the version doesn't matter so much - I'm on 1.40.1). 

Current issues are frequent hang ups of the network, regular dropouts for individual devices that need a power cycle and I am unable to OTA any of the devices. I also regularly get a uart error in the logs. 

My Z2M is running standalone on a kubernetes cluster (that's why I got the SLZB-06M in the first place), but am starting to regret the switch and am eyeing my Slaesh dongle with more and more sympathy :D It was rock solid.

TDCroPower, 2024-09-17:
@DorjeDorf so you switched back from ember to ezsp and upgraded core down to v2.3.6 and coordinator down to 20240330?
Did you have to re-pair the ZigBee devices or did it work on the fly?

DorjeDorf, 2024-09-18:
> @DorjeDorf so you switched back from ember to ezsp and upgraded core down to v2.3.6 and coordinator down to 20240330? Did you have to re-pair the ZigBee devices or did it work on the fly?

Correct. No repairing needed. It all works. I do get the odd fault every couple weeks or so (then I need to power cycle either the device itself or the coordinator), but otherwise it works. Not the OTA, mind you. That is always broken and I get loads of uart errors in the logs when I try to OTA anything.

sushain97, 2024-10-27:
I experienced similar issues with the SLZB-06M and seem to have things working with a slightly newer version combo than @DorjeDorf (and using `ezsp` as well). I suspect keeping the coordinator downgraded is the important part:

- Core v2.5.6
- Coordinator 20240330
- Z2M 1.40.2

DorjeDorf, 2024-10-28:
An update from me - since the issues never went away - especially with OTAs and regular reboots needed - I trashed the SLZB-06M and went back to Slaesh's trusty USB adapter. 

All problems went away. I could finally OTA my devices and in general, the overall feeling is of a much better system. I don't know where the problem is - whether it's the SLZB or the driver or something else - in my case - I am happy now, finally after what I consider an unsuccessful experiment.

bronzeblood, 2024-10-28:
Yeah im over my SLZB-06M what a unstable piece of shit.

mmokrejs, 2024-10-28:
For me the latest firmware update 2.5.6 works:

```
Current core firmware version: v2.5.6

Release notes:

CHANGELOG
Fixed web server lockdown that occurred in some cases in v2.5.5
Added WG ping feature
Added 'DEV FIRMWARE' label for dev ZB firmware
Updated DE locale
Updated IT locale
Updated config helper locale
Changing header text 'ZHA/Z2M' when changing the mode to Thread
The device hostname is now displayed on every page
Fixed BT Proxy helper text
Fixed colons in CORE OTA dialog
Fixed radio mode change (This bug was in 2.5.4.dev)
Fixed accordions on OTA page
```

MilesTEG1, 2024-10-28:
Hello,
I own a SLZB-06M, and I had some troubles with it since I replaced my SkyConnect with it, keeping my Zigbee Network as is, and using the official zigbee firwmare and the Core firmware from SLZB.

But since I recreate my Zigbee Network from scratch, changing channel, and all other values I had with the SkyConnect, I don't have anymore the start problem of Z2M. I also flashed this Zigbee firmware : https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2

cyayon, 2024-11-02:
Hi,
I bought a SLZB-06M as a Zigbee coordinator.
i have upgraded to the last stable firmwares v2.5.6 and 20231030.
It is powered via POE (I tried different switches/cables).

i have some Ethernet connectivity issue and coordinator freeze/crashes.
After some days / hours the web interface is not more responding.
The ping sometimes always work but the web interface not after a few days/hours.

Disconnect / reconnect the Ethernet cable (power off/on), the web interface work again.
Tried to force the port speed (on the switch) to 100M full duplex, it works but there are CRC Ethernet errors. Return to auto speed, no more CRC.
Tried to change port switch and cable, same issue.
Of course, when the web interface fail, the tcp socket connection for zigbee2mqtt is also dead…
There are also lot of FCS Ethernet error on the switch port when the device is frozen.

I managed to dump logs via USB and https://www.anticyclone-systems.co.uk/aslwebterm.php
```
esp.emac: emac_esp32_transmit(229): insufficient TX buffer size E (62967041) esp.emac: emac_esp32_transmit(229): insufficient TX buffer size E (62967588) esp.emac: emac_esp32_transmit(229): insufficient TX buffer size E (62968588) esp.emac: emac_esp32_transmit(229): insufficient TX buffer size E (62969588) esp.emac: emac_esp32_transmit(229): insufficient TX buffer size E (62970588) esp.emac: emac_esp32_transmit(229): insufficient TX buffer size E (62974041) esp.emac: emac_esp32_transmit(229): insufficient TX buffer size E (62974588) esp.emac: emac_esp32_transmit(229): insufficient TX buffer size E (62975588) esp.emac: emac_esp32_transmit(229): insufficient TX buffer size E (62976588) esp.emac: emac_esp32_transmit(229): insufficient TX buffer size E (62977588) esp.emac: emac_esp32_transmit(229): insufficient TX buffer size E (62981041) esp.emac: emac_esp32_transmit(229): insufficient TX buffer size E (62981588) esp.emac: emac_esp32_transmit(229): insufficient TX buffer size E (62982588) esp.emac: emac_esp32_transmit(229): insufficient TX buffer size E (62983588) esp.emac: emac_esp32_trans
```

SMLIGHT support give the a 2.5.8 firmware to test but it was the same issue, after a few days, the adapter crash again.

Any idea ?
Thanks.

brugalibre, 2024-11-02:
Are there any news? Like an updated firmware or an official statement in which the issue is officialy acknowledged?
I'm switching back from the SLZB-06M to the SLZB-06 since the ember integration is causing to much pain for which I simply do not have the time.
Two times a perfectly working network suddenly stoped working after a restart since the adapter could not be started anymore

mrBobMusic, 2024-11-10:
Following

Rainking80, 2024-11-11:
Following - have the same issue

gokaycagri, 2024-11-27:
Following - have the same issue

clsergent, 2024-12-15:
Following - same issue...

BenjaminMichaelis, 2024-12-16:
Following - bummer there are no consistent work arounds currently it seems.

BenjaminMichaelis, 2024-12-16:
Following - bummer there are no consistent work arounds currently it seems.

Is there any information or help that can be provided to help resolve this @Nerivec? I'm not too familiar with this hardware so I'm not totally sure where to go next, but have debugged and tried what I can so far, but am happy to provide needed information.

Ra72xx, 2024-12-16:
I'm in the process of upgrading to an SLZB-07MG24, any chance that this more recent HW isn't suffering from this problems?

Nerivec, 2024-12-16:
@Ra72xx 07 isn't TCP-based, so no ESP side at all (no core firmware involved, just the serial driver on your machine).
I've been using the SLZB-07MG24 in testing for a while (8.0.2 firmware), no problem so far (as coordinator or as router). Granted, my test setup is more limited than a real network, but I do put it through far more than most users ever will in terms of reboots, edge cases, forced crashes, etc. 😉
_The MG24 series offers quite a lot more in terms of specs, most networks should benefit (but especially larger/crowded ones).
For examples, SMLight SLZB-07MG24 has 2x the RAM from (most) previous adapters (MG21), TubesZB MGM24 has 4x the RAM. This should allow them to handle tougher scenario._

Ra72xx, 2024-12-16:
Tbanks for the info. I'm using SkyConnect right now, but am suffering from various instabilities like the one here in this thread (which, however, seems to have gotten better recently), so I'm a bit hoping that the newer hardware will make my network more stable.

takbo3000, 2024-12-18:
Following - same issue...

cairnsie13, 2024-12-27:
Following - hopefully a update soon.

[18 more replies]

closed 2025-06-10 by github-actions[bot]