Koenkk/zigbee2mqtt #23121: Ember : NCP fatal error

smh51, 2024-06-21:
### What happened?

09:38:37] INFO: Preparing to start...
[09:38:38] INFO: Socat not enabled
[09:38:39] INFO: Starting Zigbee2MQTT...
[19-06-2024 09:38:54] error: 	z2m: Cannot find '0xa4c1385f5cd1e946/1' of group 'Lampadaire du salon'
[19-06-2024 09:38:54] error: 	z2m: Cannot find endpoint '1' of device '0xa4c1385f5cd1e946'
[19-06-2024 09:39:33] error: 	zh:ember:uart:ash: Received ERROR from NCP, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[19-06-2024 09:39:33] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[19-06-2024 09:39:33] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[19-06-2024 09:39:33] error: 	zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!
[21-06-2024 07:50:31] error: 	zh:ember:uart:ash: Received ERROR from NCP, with code=RESET_WATCHDOG.
[21-06-2024 07:50:31] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[21-06-2024 07:50:31] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[21-06-2024 07:50:31] error: 	zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!
[21-06-2024 12:55:30] error: 	zh:ember:uart:ash: Received ERROR from NCP, with code=RESET_WATCHDOG.
[21-06-2024 12:55:30] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[21-06-2024 12:55:30] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[21-06-2024 12:55:30] error: 	zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!

### What did you expect to happen?

no error

### How to reproduce it (minimal and precise)

Start the add-on z2m, look at logs

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

7.4.2 [GA]

### Adapter

ZBZ-1, Skyconnect

### Setup

add-on, HAOS on proxmox

Nerivec, 2024-06-22:
Can you please provide a `debug` log so I can see what triggers that timeout?

smh51, 2024-06-22:
[log.log](https://github.com/user-attachments/files/15936311/log.log)

Nerivec, 2024-06-22:
I don't see the error in those logs.

You can try a firmware without hardware flow control, it should provide cleaner startups:
https://github.com/darkxst/silabs-firmware-builder/tree/ember-nohw/firmware_builds/skyconnect
_PS: don't forget to set `rtscts: false` in [your config](https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#configuration-of-the-zigbee-adapter) if you do use one of these._

smh51, 2024-06-22:
i dont want to mess too much with the firmware as it is working.

You mean this error is nothing to worry about ?  

[22-06-2024 13:18:16] info: 	zh:ember:uart:ash: ======== ASH starting ========
[22-06-2024 13:18:16] debug: 	zh:ember:uart:ash: <--- [FRAME type=ERROR]
[22-06-2024 13:18:16] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=RESET_WATCHDOG.
[22-06-2024 13:18:16] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[22-06-2024 13:18:16] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[22-06-2024 13:18:16] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========

Nerivec, 2024-06-22:
Yes that error is linked to hardware flow control, but there is a retry mechanism in place to bypass it. It shouldn't prevent start.

smh51, 2024-06-22:
i use the firmeware from https://connectzbt1.home-assistant.io

I tried with rtscts true and false and i still have the same error. 

We can close the issue if you say there is nothing to worry about.

thx ;)

closed 2024-06-22 by Nerivec